lm-sensors/lm-sensors #144: Fan control on Asrock Fatal1ty AB350 Gaming-ITX/ac motherboard doesn't work under linux

flip111, 2018-11-21:
Hello,

I was referred here from upstream https://bugs.launchpad.net/ubuntu/+source/lm-sensors/+bug/1803795

Under windows the fancontrol (with asrock f-stream utility) works great, under linux i'm running into issues.

I'll upload the same file as in the upstream issue report [fancontrol bug report.txt](https://github.com/lm-sensors/lm-sensors/files/2604306/fancontrol.bug.report.txt)

olysonek, 2018-11-21:
So can you confirm that `cat /sys/class/hwmon/hwmon0/fan1_input` gives the error "invalid argument"? That seems odd...

In response to the comment from Andreas, lm-sensors.org has been down for a couple of years and is not going back. This github repo and https://hwmon.wiki.kernel.org/lm_sensors is the new upstream location.

flip111, 2018-11-21:
```
» cat /sys/class/hwmon/hwmon0/fan1_input
cat: /sys/class/hwmon/hwmon0/fan1_input: Invalid argument
» ls /sys/class/hwmon/hwmon0/fan1_input
/sys/class/hwmon/hwmon0/fan1_input
```

flip111, 2018-11-21:
Maybe applicable https://unix.stackexchange.com/a/319983

groeck, 2018-11-22:
I would be most interested to understand the conclusion that this is a nct6775 driver problem. From the context, it appears that hwmon0 is associated with amdgpu. Here is the output from the fancontrol bug report:

Found the following devices:
   hwmon0 is amdgpu
   hwmon1 is k10temp
   hwmon2 is iwlwifi

And:
-Sensors-
amdgpu/temp1	Temperature	32,00°C	
k10temp/temp1	Temperature	32,38°C	
k10temp/temp2	Temperature	32,38°C	
iwlwifi/temp1	Temperature	36,00°C	
thermal/thermal_zone0	Temperature	36,00°C	

I do not see "nct6775" in this list.

Note that the amdgpu hwmon driver is maintained by AMD and, to say it mildly, not well written (and without any input from hwmon maintainers). Reading its fan speed returns -EINVAL if the card is turned off (or maybe if the fan on the card is turned off, or if it doesn't have a fan to start with). This is of course not a valid or useful return value and does not match the definition of EINVAL, but there is really not much we can do about that at this point except maybe complain to the AMD DRM driver maintainer(s).

Nevertheless, getting back to the "nct6775 problem". If the person with the problem looks at the output of dmesg, it will most likely include some "ACPI resource conflict" messages. If so, _that_ is the problem.

flip111, 2018-11-23:
Hi @groeck thank you very much for your input.

In my computer case i have an APU with a fan. So since the GPU is in the APU i guess it could be possible that amdgpu driver is associated with this fan. But i'm not sure about that. If so reading your comment it looks like a driver issue with the amdgpu driver.

On the other hand may be due to ACPI conflicts the right driver is not loaded or in other ways not available.

When looking at dmesg a few messages stand out (these messages do not directly follow each other, i just copied relevant lines)

```
[    0.000000] ACPI BIOS Warning (bug): Optional FADT field Pm2ControlBlock has valid Length but zero Address: 0x0000000000000000/0x1 (20180531/tbfadt-624)
[    0.082934] ACPI BIOS Error (bug): Failure creating [\_SB.SMIC], AE_ALREADY_EXISTS (20180531/dswload2-316)
[    0.082941] ACPI Error: AE_ALREADY_EXISTS, During name lookup/catalog (20180531/psobject-221)
[    0.082945] ACPI BIOS Error (bug): Failure creating [\_SB.SMIB], AE_ALREADY_EXISTS (20180531/dsfield-594)
[    0.085109] ACPI: [Firmware Bug]: BIOS _OSI(Linux) query ignored
```

groeck, 2018-11-23:
The APU fan is managed by the system (with the nct6792). The amdgpu driver only reports fan speeds for fans on separate graphics cards. If you want to discuss the error further, I would suggest to contact the driver maintainers ("scripts/get_maintainer.pl -f drivers/gpu/drm/radeon/radeon_pm.c" in the Linux kernel source will tell you who they are). Feel free to copy me and linux-hwmon@vger.kernel.org when you do.

The ACPI messages point to problems in the DSDT, which looks like sloppy work by the ASRock BIOS team. If you have the latest BIOS version installed, it might make sense to open a case with ASRock support and see what they say. They might come back with "we don't support Linux", but it might be worth a try.

Having said that, the messages do not point to ACPI affecting with the nct6775 driver. That message would be different. Can you attach a complete dmesg, or make it available somewhere and point to it ? You can also send it to me directly - you'll find my email address in MAINTAINERS in the Linux kernel source (scripts/get_maintainer.pl -f drivers/hwmon/nct6775.c).

flip111, 2018-11-23:
I just put dmesg log here quickly (about to go out of the house). I briefly read your comment @groeck , i will read it more carefully later
[dmesg.txt](https://github.com/lm-sensors/lm-sensors/files/2610678/dmesg.txt)

groeck, 2018-11-23:
Hmm - there is no ACPI error message related to nct6775. What do you see when you run "sudo modprobe -r nct6775; sudo modprobe nct6775" ?

Note that you do have a problem with your system. Look at the output of dmesg, and the code tracebacks in it. You might want to open an Ubuntu error report on that.

flip111, 2018-11-27:
> if you want to discuss the error further, I would suggest to contact the driver maintainers

@groeck i'm not sure which error you mean with "the error". Because the paragraph is about amdgpu and amdgpu is for a separate graphics card (which i don't have). I guess it's not really relevant in my situation??

> but it might be worth a try.

Ok i will open a ticket with AsRock

> What do you see when you run "sudo modprobe -r nct6775; sudo modprobe nct6775" ?

Absolutely nothing. The prompt just returns immediately

> You might want to open an Ubuntu error report on that.

Ok i will do that as well

flip111, 2018-11-28:
@groeck ok i will send an e-mail to the maintainers of amdgpu. Not because i think it's related directly to the thing with nct6775 but i'll do it now that i'm on this issue anyway.

I added some logging to dmesg with printk function in the driver and i get the following messages in dmesg:
```
[ 7901.131954] function sensors_nct6775_init() executed
[ 7901.132054] nct6775: Found NCT6792D or compatible chip at 0x2e:0x290
[ 7901.132126] function nct6775_probe() executed
```

I performed the following steps to do this:
```bash
sudo apt install libssl1.1=1.1.1-1ubuntu2 libssl-dev libelf-dev

git clone --single-branch --depth=1 --branch v4.18 https://github.com/torvalds/linux.git linux-4.18

cd linux-4.18

# https://askubuntu.com/a/515408/231084

make oldconfig
make prepare
make scripts

sudo apt install linux-headers-$(uname -r)
cp -v /usr/src/linux-headers-$(uname -r)/Module.symvers .

sudo mv -v /lib/modules/$(uname -r)/kernel/drivers/hwmon/nct6775.ko /lib/modules/$(uname -r)/kernel/drivers/hwmon/nct6775.ko.backup

cd drivers/hwmon

# make changes to source code now

make -C /lib/modules/$(uname -r)/build M=$(pwd) modules
# has some errors here with ssl stuff, doesn't seem important https://github.com/umlaeute/v4l2loopback/issues/139
sudo make -C /lib/modules/$(uname -r)/build M=$(pwd) modules_install

# unload old driver
modprobe -r nct6775

# find new driver
sudo depmod

# load new driver
sudo modprobe -v nct6775
```

by the way while compiling the nct6775 drivers i get the some warnings. I don't think they are relevant to my issue but they stand out because the other drivers didn't give any warnings.

```
  CC [M]  /home/flip111/linux-4.18/drivers/hwmon/nct6775.o
/home/flip111/linux-4.18/drivers/hwmon/nct6775.c: In function ‘nct6775_probe’:
/home/flip111/linux-4.18/drivers/hwmon/nct6775.c:3594:2: warning: ISO C90 forbids mixed declarations and code [-Wdeclaration-after-statement]
  struct device *dev = &pdev->dev;
  ^~~~~~
/home/flip111/linux-4.18/drivers/hwmon/nct6775.c: In function ‘sensors_nct6775_init’:
/home/flip111/linux-4.18/drivers/hwmon/nct6775.c:4518:2: warning: ISO C90 forbids mixed declarations and code [-Wdeclaration-after-statement]
  int i, err;
  ^~~
```

groeck, 2018-11-28:
Can you send me (or attach) the source file (nct6775.c) ? The warnings are not ok. The first warning should be the first line in nct6775_probe(), and the second warning should be the first line in sensors_nct6775_init(). Both are most definitely not mixing declarations with code.

Also, can you print the return value from devm_hwmon_device_register_with_groups() ?

flip111, 2018-11-28:
> Or did you add those print statements at those locations ?

Yes pretty sure that was it. I'm not all too proficient with C so to me it wasn't immediately obvious from the compiler message that it was actually a problem with the previous line.

> Also, can you print the return value from devm_hwmon_device_register_with_groups() ?

I didn't find this as function declaration in the nct6775 file but i did see it being called once near the near of the nct6775_probe() function.

So my best guess is (i don't have fancy tools to inspect the source code) is that the return value is of this type https://github.com/torvalds/linux/blob/v4.18/include/linux/device.h#L926-L1006

I would either need to hook into the module at runtime with GDB and extract the current values at that point somehow (i never done this before).

Or i need to write a function that will print this struct nicely for me, in which case i rather write a program that converts a struct declaration into a print function. Also there is the case of macros in this structs which conditionally enable parts of the structs so i would need to run it through the preprocessor and possibly find out the right compiler flags before i do (i mean other than `-E`).

Both options would take me at least a week (if not more) to do. So i thought i first post back here before i run off and report back much later. Do you have any suggestions?

One thing that seems much easier to do is [dump the whole thing as hex](https://stackoverflow.com/a/5349944/1833322) but i doubt that's gonna be useful.

groeck, 2018-11-28:
All I was asking for is something like

        hwmon_dev = devm_hwmon_device_register_with_groups(dev, data->name,
                                                           data, data->groups);
        dev_info(dev, "hwmon_dev: %pK\n", hwmon_dev);
        return PTR_ERR_OR_ZERO(hwmon_dev);

flip111, 2018-11-28:
I don't know about the `dev_info()` function. Anyway when i add that line i get the following output in dmesg

```
[ 8958.767808] function sensors_nct6775_init() start
[ 8958.767893] nct6775: Found NCT6792D or compatible chip at 0x2e:0x290
[ 8958.767976] function nct6775_probe() start
[ 8958.769005] nct6775 nct6775.656: hwmon_dev: ffff884008e9d008
```

groeck, 2018-11-28:
This gets more and more confusing. The return value suggests that the device was correctly instantiated, meaning entries in /sys/class/hwmon/ should exist.

Can you run `grep . /sys/class/hwmon/*/*` ?

flip111, 2018-11-28:
```
grep: /sys/class/hwmon/hwmon0/device: Is a directory
grep: /sys/class/hwmon/hwmon0/fan1_input: Invalid argument
grep: /sys/class/hwmon/hwmon0/in0_input: Invalid argument
/sys/class/hwmon/hwmon0/in0_label:vddgfx
grep: /sys/class/hwmon/hwmon0/in1_input: Invalid argument
/sys/class/hwmon/hwmon0/in1_label:vddnb
/sys/class/hwmon/hwmon0/name:amdgpu
grep: /sys/class/hwmon/hwmon0/power: Is a directory
grep: /sys/class/hwmon/hwmon0/power1_average: Invalid argument
/sys/class/hwmon/hwmon0/pwm1:0
/sys/class/hwmon/hwmon0/pwm1_enable:0
/sys/class/hwmon/hwmon0/pwm1_max:255
/sys/class/hwmon/hwmon0/pwm1_min:0
grep: /sys/class/hwmon/hwmon0/subsystem: Is a directory
/sys/class/hwmon/hwmon0/temp1_crit:80000
/sys/class/hwmon/hwmon0/temp1_crit_hyst:0
/sys/class/hwmon/hwmon0/temp1_input:28000
/sys/class/hwmon/hwmon1/beep_enable:0
grep: /sys/class/hwmon/hwmon1/device: Is a directory
/sys/class/hwmon/hwmon1/fan1_alarm:0
/sys/class/hwmon/hwmon1/fan1_beep:0
/sys/class/hwmon/hwmon1/fan1_input:0
/sys/class/hwmon/hwmon1/fan1_min:0
/sys/class/hwmon/hwmon1/fan1_pulses:2
/sys/class/hwmon/hwmon1/fan1_target:0
/sys/class/hwmon/hwmon1/fan1_tolerance:337500
/sys/class/hwmon/hwmon1/fan2_alarm:0
/sys/class/hwmon/hwmon1/fan2_beep:0
/sys/class/hwmon/hwmon1/fan2_input:1660
/sys/class/hwmon/hwmon1/fan2_min:0
/sys/class/hwmon/hwmon1/fan2_pulses:2
/sys/class/hwmon/hwmon1/fan2_target:27000
/sys/class/hwmon/hwmon1/fan2_tolerance:1082
/sys/class/hwmon/hwmon1/fan3_alarm:0
/sys/class/hwmon/hwmon1/fan3_beep:0
/sys/class/hwmon/hwmon1/fan3_input:0
/sys/class/hwmon/hwmon1/fan3_min:0
/sys/class/hwmon/hwmon1/fan3_pulses:2
/sys/class/hwmon/hwmon1/fan3_target:0
/sys/class/hwmon/hwmon1/fan3_tolerance:337500
/sys/class/hwmon/hwmon1/fan4_alarm:0
/sys/class/hwmon/hwmon1/fan4_beep:0
/sys/class/hwmon/hwmon1/fan4_input:0
/sys/class/hwmon/hwmon1/fan4_min:0
/sys/class/hwmon/hwmon1/fan4_pulses:2
/sys/class/hwmon/hwmon1/fan4_target:0
/sys/class/hwmon/hwmon1/fan4_tolerance:0
/sys/class/hwmon/hwmon1/fan5_alarm:0
/sys/class/hwmon/hwmon1/fan5_beep:0
/sys/class/hwmon/hwmon1/fan5_input:0
/sys/class/hwmon/hwmon1/fan5_min:0
/sys/class/hwmon/hwmon1/fan5_pulses:2
/sys/class/hwmon/hwmon1/fan5_target:0
/sys/class/hwmon/hwmon1/fan5_tolerance:0
/sys/class/hwmon/hwmon1/in0_alarm:0
/sys/class/hwmon/hwmon1/in0_beep:0
/sys/class/hwmon/hwmon1/in0_input:376
/sys/class/hwmon/hwmon1/in0_max:1744
/sys/class/hwmon/hwmon1/in0_min:0
/sys/class/hwmon/hwmon1/in10_alarm:1
/sys/class/hwmon/hwmon1/in10_beep:0
/sys/class/hwmon/hwmon1/in10_input:112
/sys/class/hwmon/hwmon1/in10_max:0
/sys/class/hwmon/hwmon1/in10_min:0
/sys/class/hwmon/hwmon1/in11_alarm:1
/sys/class/hwmon/hwmon1/in11_beep:0
/sys/class/hwmon/hwmon1/in11_input:120
/sys/class/hwmon/hwmon1/in11_max:0
/sys/class/hwmon/hwmon1/in11_min:0
[... 214 lines cut ...]
/sys/class/hwmon/hwmon1/temp5_type:4
/sys/class/hwmon/hwmon1/temp6_input:-128000
/sys/class/hwmon/hwmon1/temp6_label:AUXTIN3
/sys/class/hwmon/hwmon1/temp6_offset:0
/sys/class/hwmon/hwmon1/temp6_type:4
/sys/class/hwmon/hwmon1/temp7_input:0
/sys/class/hwmon/hwmon1/temp7_label:PCH_CHIP_CPU_MAX_TEMP
/sys/class/hwmon/hwmon1/temp8_input:0
/sys/class/hwmon/hwmon1/temp8_label:PCH_CHIP_TEMP
/sys/class/hwmon/hwmon1/temp9_input:0
/sys/class/hwmon/hwmon1/temp9_label:PCH_CPU_TEMP
grep: /sys/class/hwmon/hwmon2/device: Is a directory
/sys/class/hwmon/hwmon2/name:k10temp
grep: /sys/class/hwmon/hwmon2/power: Is a directory
grep: /sys/class/hwmon/hwmon2/subsystem: Is a directory
/sys/class/hwmon/hwmon2/temp1_input:28000
/sys/class/hwmon/hwmon2/temp1_label:Tdie
/sys/class/hwmon/hwmon2/temp1_max:70000
/sys/class/hwmon/hwmon2/temp2_input:28000
/sys/class/hwmon/hwmon2/temp2_label:Tctl
/sys/class/hwmon/hwmon3/name:iwlwifi
grep: /sys/class/hwmon/hwmon3/power: Is a directory
grep: /sys/class/hwmon/hwmon3/subsystem: Is a directory
/sys/class/hwmon/hwmon3/temp1_input:32000
```

groeck, 2018-11-28:
That looks perfectly ok; nct6792 is instantiated as hwmon1. The "sensors" command should show the device.

flip111, 2018-11-28:
```
k10temp-pci-00c3
Adapter: PCI adapter
Tdie:         +38.9°C  (high = +70.0°C)
Tctl:         +38.9°C  

amdgpu-pci-3800
Adapter: PCI adapter
vddgfx:           N/A  
vddnb:            N/A  
fan1:             N/A
temp1:        +38.0°C  (crit = +80.0°C, hyst =  +0.0°C)
power1:           N/A  

iwlwifi-virtual-0
Adapter: Virtual device
temp1:        +33.0°C  

nct6792-isa-0290
Adapter: ISA adapter
in0:                    +0.34 V  (min =  +0.00 V, max =  +1.74 V)
in1:                    +0.00 V  (min =  +0.00 V, max =  +0.00 V)
in2:                    +3.34 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in3:                    +3.34 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in4:                    +0.00 V  (min =  +0.00 V, max =  +0.00 V)
in5:                    +0.00 V  (min =  +0.00 V, max =  +0.00 V)
in6:                    +0.91 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in7:                    +3.47 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in8:                    +3.30 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in9:                    +0.00 V  (min =  +0.00 V, max =  +0.00 V)
in10:                   +0.11 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in11:                   +0.12 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in12:                   +1.84 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in13:                   +1.70 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in14:                   +0.00 V  (min =  +0.00 V, max =  +0.00 V)
fan1:                     0 RPM  (min =    0 RPM)
fan2:                  1660 RPM  (min =    0 RPM)
fan3:                     0 RPM  (min =    0 RPM)
fan4:                     0 RPM  (min =    0 RPM)
fan5:                     0 RPM  (min =    0 RPM)
SYSTIN:                +122.0°C  (high =  +0.0°C, hyst =  +0.0°C)  sensor = thermistor
CPUTIN:                 +28.5°C  (high = +80.0°C, hyst = +75.0°C)  sensor = thermistor
AUXTIN0:                +31.0°C  (high =  +0.0°C, hyst =  +0.0°C)  ALARM  sensor = thermistor
AUXTIN1:               +111.0°C    sensor = thermistor
AUXTIN2:               +112.0°C    sensor = thermistor
AUXTIN3:               -128.0°C    sensor = thermistor
PCH_CHIP_CPU_MAX_TEMP:   +0.0°C  
PCH_CHIP_TEMP:           +0.0°C  
PCH_CPU_TEMP:            +0.0°C  
PCH_MCH_TEMP:            +0.0°C  
intrusion0:            OK
intrusion1:            ALARM
beep_enable:           disabled
```

groeck, 2018-11-28:
Everything looks perfect, except of course the amdgpu output, and we can't do anything about that. I think I am lost/confused. What problem are you seeing at this time ?

flip111, 2018-11-28:
Hhmm i guess there wasn't enough debug information to know before that `hwmon1` contained the right fan and sensors would work o_O

I will try to undo the steps in https://github.com/lm-sensors/lm-sensors/issues/144#issuecomment-442289069 by deleting the compiled modules in `/lib/modules/4.18.0-11-generic/extra` and also put back the backup back
`sudo mv -v /lib/modules/$(uname -r)/kernel/drivers/hwmon/nct6775.ko.backup /lib/modules/$(uname -r)/kernel/drivers/hwmon/nct6775.ko` and then restart

So my best guess at the moment about what happened is:

1. The amdgpu driver was loaded (which it shouldn't) under hwmon0
2. The amdgpu driver gave wrong readings
3. fancontrol choked on the previous 2 steps happening
4. The right fan in hwmon1 was never found and by fancontrol due to previous errors.

The problem which i was seeing is the original problem in this bug report https://bugs.launchpad.net/ubuntu/+source/lm-sensors/+bug/1803795 of which i copy the text below for conveniance

> When running sudo pwmconfig i get the following errors:
>
> /usr/sbin/pwmconfig: line 455: let: threshold=3*/4: syntax error: operand expected (error token is "/4")
>
> cat: hwmon0/fan1_input: Invalid argument
>
> Also my CPU fan did not stop but kept spinning (it was suppose to stop for 5 seconds). Please see the attached file for my terminal session and a report on my computer.

groeck, 2018-11-28:
I think the underlying problem is that pwmconfig does not handle the error from reading hwmon0/fan1_input.

mirh, 2020-06-10:
Still getting the same token error
```
# sensors-detect version 3.6.0
# System: MSI MS-7916 [1.0]
# Board: MSI Z97 GAMING 7 (MS-7916)
# Kernel: 5.6.15-1-MANJARO x86_64
# Processor: Intel(R) Core(TM) i7-4770K CPU @ 3.50GHz (6/60/3)
```

Not sure if having a water pump connected to the cpu fan header (meaning fixed speed you can't really control) couldn't have something to do with this
```
Found the following fan sensors:
cat: hwmon2/fan1_input: invalid argument
   hwmon2/fan1_input     current speed:  RPM
   hwmon3/fan1_input     current speed: 0 ... skipping!
   hwmon3/fan2_input     current speed: 1535 RPM
   ...
```

flip111, 2020-06-11:
@mirh check if your UEFI panel has settings for fan control, that might be easier than controlling them from software.

mirh, 2020-06-11:
If I had been fine already with that, I wouldn't have checked lm-sensors.

rittme, 2020-07-06:
For people that get to this issue because their ASRock B450i didn't get detected, the following fixed it for me:
```
sudo modprobe -r nct6775; sudo modprobe nct6775
```