lm-sensors/lm-sensors #148: SENSORS_API_VERSION is not useful anymore after 3.5.0

rpv-tomsk, 2018-12-02:
1)  Commit e8afbda10fba571c816abddcb5c8180afc435bba.

By your such decision and release you break all idea of SENSORS_API_VERSION.

First digit was intended to change only when API changed and code have to be _rewritten_ to work properly with new lm_sensors release.

2) Commits, which only adds new enums or flags have to not break ABI.

And dcf23676cc264927ad58ae7960f518689372741a is wrong about this rule.
You haven't break ABI only to keep lists of subfeatures "sorted".

That change is really minor change and only last, third digit really have to be changed (if add new subfeatures to end of list).

People have forgotten how to think ahead.
People have forgotten how to think.

Unfortunately, there are no other people on this planet.

olysonek, 2018-12-03:
Whatever the original lm_sensors author intended to do with the first digit, there really is only one correct way to approach it - bump the first digit, i.e. change the soname, whenever API *or* ABI changes. If you didn't bump the soname when ABI changes, dependent programs could silently start producing bad results if they weren't rebuilt. By changing a library's soname you make consumers of the library aware that at least a rebuild is necessary.

That being said, I did make a mistake when assessing the impact of commit dcf23676cc26 on the ABI. Even though new labels were inserted in the middle of a public enum in sensors.h, the ABI did not change, thanks to lines that assign specific values to certaing labels, such as `SENSORS_SUBFEATURE_POWER_AVERAGE_INTERVAL = (SENSORS_FEATURE_POWER << 8) | 0x80,`. If such lines weren't there, the commit would have caused the ABI to change. They are there though, I didn't notice that and I'm sorry about that.

Anyway, library soname bumps are a completely normal thing to do, they happen all the time. So keep calm, rebuild dependent packages and carry on.

closed 2018-12-03 by olysonek

rpv-tomsk, 2018-12-03:
>there really is only one correct way to approach it - bump the first digit, i.e. change the soname, whenever API or ABI changes. If you didn't bump the soname when ABI changes, dependent programs could silently start producing bad results if they weren't rebuilt. 

It looks like you going to bring ABI changes w/o API changes. Hmm.

Original authors was adequate to not bring incompatible ABI changes w/o API changes and to do development in backward-compatible way. So, their rule `first digit is the major version (changed when the API + ABI breaks)` was correct. 

Hope you will be adequate too, but the beginning showed poor results. 

>Anyway, library soname bumps are a completely normal thing to do, they happen all the time. 

That "completely normal thing" does not happen "all the time" for 10 years. 
And what fact does not prohibit growth.

>So keep calm, rebuild dependent packages and carry on.

It complicates things while this was not really needed.
This is irresponsible statement from your side, and irresponsible behaviour.

rpv-tomsk, 2018-12-03:
>So keep calm, rebuild dependent packages and carry on.

You forget to recommend to "Update your code, which checks SENSORS_API_VERSION, then ...".

olysonek, 2018-12-17:
Hi @rpv-tomsk, we're currently considering reverting the soname change. Would that be ok, or would it complicate things even more? Also, can you elaborate on what impact the change had for you, meaning, for example, are you a lm_sensors maintainer in a distro, or do you build lm_sensors for yourself?

Whissi, 2018-12-17:
Gentoo package maintainer speaking here.

Like you already noticed, updating SONAME in 3.5.0 was useless. There were zero ABI change:
```diff
--- ABI-4.dump	2018-12-17 14:33:57.931031288 +0100
+++ ABI-5.dump	2018-12-17 14:31:56.693306035 +0100
@@ -5,8 +5,8 @@
           'GccVersion' => '8.2.0',
           'Headers' => {},
           'Language' => 'C',
-          'LibraryName' => 'libsensors.so.4.4.0',
-          'LibraryVersion' => '4',
+          'LibraryName' => 'libsensors.so.5.0.0',
+          'LibraryVersion' => '5',
           'NameSpaces' => {},
           'Needed' => {
                         'libc.so.6' => 1,
@@ -278,7 +278,7 @@
                           },
           'SymbolVersion' => {},
           'Symbols' => {
-                         'libsensors.so.4.4.0' => {
+                         'libsensors.so.5.0.0' => {
                                                     'libsensors_version' => -8,
                                                     'sensors_cleanup' => 1,
                                                     'sensors_do_chip_sets' => 1,
@@ -1327,7 +1327,7 @@
                                    }
                         },
           'UndefinedSymbols' => {
-                                  'libsensors.so.4.4.0' => {
+                                  'libsensors.so.5.0.0' => {
                                                              '_ITM_deregisterTMCloneTable' => 0,
                                                              '_ITM_registerTMCloneTable' => 0,
                                                              '__cxa_finalize@GLIBC_2.2.5' => 0,
@@ -1361,6 +1361,7 @@
                                                              'readdir@GLIBC_2.2.5' => 0,
                                                              'readlink@GLIBC_2.2.5' => 0,
                                                              'realloc@GLIBC_2.2.5' => 0,
+                                                             'realpath@GLIBC_2.3' => 0,
                                                              'scandir@GLIBC_2.2.5' => 0,
                                                              'setlocale@GLIBC_2.2.5' => 0,
                                                              'sscanf@GLIBC_2.2.5' => 0,
```
(I used https://lvc.github.io/abi-compliance-checker/ to verify)

But you cannot revert such a change in an already released version. This will break setup for everyone who has already picked up the new version.

olysonek, 2018-12-17:
Yes, I'm trying to figure out who, if anybody, already picked it up. Do you know anybody?

I've received a request to revert the soname  change, so I'm trying to figure out if it's reasonable to do.
Here's the thread from linux-hwmon (spinics didn't pickup all of it yet though):
https://www.spinics.net/lists/linux-hwmon/msg04837.html

Whissi, 2018-12-17:
We, Gentoo, picked up the changes, see https://gitweb.gentoo.org/repo/gentoo.git/log/?qt=grep&q=lm_sensors

And we are not alone: https://repology.org/metapackage/lm-sensors/versions

olysonek, 2018-12-17:
Cool, thanks for that repology link! I didn't know about that.

olysonek, 2018-12-17:
@Whissi So would reverting the soname change mean a significant amount of work for you in Gentoo?

Whissi, 2018-12-17:
This would cause a rebuild of ~70 packages. We can handle it but it would be wrong.
Also: In theory you can never ever use SONAME 5 again... so if you go back to 4 you would have to jump to 6 if you ever need to increase SONAME if you want to do it right.

I would recommend to keep the SONAME change. You learned something and it will hopefully never happen again. But anything else would create a bigger mess.

olysonek, 2018-12-18:
Thanks. I completely agree, I just wanted to get input from other people.

rpv-tomsk, 2018-12-18:
>Hi @rpv-tomsk, we're currently considering reverting the soname change.

As already discussed here, time for this action was lost :-( 
Unfortunately, mine efforts was not enough for this to happen a time. 
I'm only one package maintainer, not whole distro )) and found no strong argues to my position. An nobody invited by me (from Debian) also said no any words. So, that had to go on like it goes.

Thanks for your work and efforts on resolving this.