lm-sensors/lm-sensors #152: Saving config in pwmconfig fails

alexanderbazhenoff, 2018-12-29:
Executing command `sudo pwmconfig` fails on Debian Stretch with:
```
mv: cannot create regular file 'y': Permission denied
chmod: cannot access 'y': No such file or directory
```
when you set default path (blank $FCCONFIG) or set to 'y' by mistake.

Fixed.

olysonek, 2019-01-08:
Well, it's not a bug, it's the user who made a mistake.

closed 2019-01-08 by olysonek

alexanderbazhenoff, 2019-01-09:
Yeah, it's not bug, just better usabilty. The default path is shown in brackets when it's prompted. I whought that this is default, pushed "enter" and got an error. I didn't get what the error and what's going on three times, until I looked at the code... LOL As in previous times you could also answer "y". With handling is more userfriendly, but as you wish.

olysonek, 2019-01-10:
> Yeah, it's not bug, just better usabilty.

I think treating "y" differently could  in fact hurt usability.

> The default path is shown in brackets when it's prompted. I whought that this is default, pushed "enter" and got an error.

It is the default. Just pressing enter should choose the defaut. It does work like that for me.