lm-sensors/lm-sensors #154: ITE IT8688E not detected

sbraz, 2018-12-30:
Hello,
I'm running version 3.5.0 and this chip is not detected. 
```
# System: Gigabyte Technology Co., Ltd. Z390 GAMING SLI [Default string]
# Board: Gigabyte Technology Co., Ltd. Z390 GAMING SLI-CF
…
Trying family `ITE'...                                      Yes
Found unknown chip with ID 0x8688
```
I see that you have been contributing to the it87 driver, would it be possible to add support for this chip in the kernel too?
I was able to make it work by adding `acpi_enforce_resources=lax` to my kernel parameters and forcing a different chip ID with `modprobe it87 force_id=0x8628` and `sensors` seems to be showing consistent info:
```
it8628-isa-0a40
Adapter: ISA adapter
in0:          +0.88 V  (min =  +0.00 V, max =  +3.06 V)
in1:          +2.04 V  (min =  +0.00 V, max =  +3.06 V)
in2:          +2.03 V  (min =  +0.00 V, max =  +3.06 V)
in3:          +2.05 V  (min =  +0.00 V, max =  +3.06 V)
in4:          +0.00 V  (min =  +0.00 V, max =  +3.06 V)
in5:          +1.34 V  (min =  +0.00 V, max =  +3.06 V)
in6:          +1.39 V  (min =  +0.00 V, max =  +3.06 V)
3VSB:         +3.43 V  (min =  +0.00 V, max =  +6.12 V)
Vbat:         +3.14 V
fan1:         742 RPM  (min =    0 RPM)
fan2:         229 RPM  (min =    0 RPM)
fan3:         469 RPM  (min =    0 RPM)
fan4:         674 RPM  (min =    0 RPM)
fan5:         483 RPM  (min =    0 RPM)
temp1:        +33.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp2:        +40.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp3:        +23.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = disabled
temp4:        +33.0°C
temp5:        +37.0°C
temp6:        +39.0°C
intrusion0:  OK
```

Please let me know if there is any way I can help.

groeck, 2019-01-02:
Sorry, I don't make changes to the it87 driver anymore. I'll be happy to accept patches, but those have to be based on datasheet information, since chip details are always different between the various chip variants from ITE - we can not just add chip IDs and assume the functionality is identical to older chips. Worst case, this can destabilize the system and result in random crashes.

sbraz, 2019-01-03:
I wouldn't have the time or knowledge to write those patches.
How would one go about finding the datasheet though?
I agree that it makes sense to not just assume that it's similar to another chip.

groeck, 2019-01-03:
Unfortunately I lost my Linux-friendly contact at Gigabyte, and no longer have the means to get ITE datasheets.  I stopped using Gigabyte boards myself because of that. Trying to get datasheets from ITE is all but hopeless. In other words, I have no idea how to get datasheets for chips used in Gigabyte boards.

djkerya, 2019-07-26:
Any updates on it?

groeck, 2019-07-26:
Not from me. I don't have a datasheet (nor the time to implement the code).

djkerya, 2019-07-27:
Gigabyte Z390 UD. Ubuntu 18.04 kernel 4.15.0, lm-sensors 3.4.0. My workaround :
I have used  acpi_enforce_resources=lax in /etc/default/grub
and after that sudo modprobe it87 force_id=0x8628
Sensors show both my connected cpu fan and sysfan 1.

eyaleb, 2019-12-20:
Waiting for progress, I contacted gigabyte support asking what the different voltages and temperatures are. Below is their response.
```
Hi Sir

Linux is not our officially support OS.

We don't have that kind of info to provide.
Please refer to the below user from your link to modify the it87 driver.
[copy of the above comment from @djkerya trimmed]

Best Regards,

GIGABYTE
```
So gigabyte do not support linux...

FYI

groeck, 2019-12-20:
eyaleb@: Correct. Gigabyte, like pretty much all major board vendors, does not support Linux. Gigabyte used to be friendly enough to make datasheets available when asked nicely, but they no longer do that.

LXXero, 2020-01-06:
I used to have this working on my gigabyte z390 designaire with the acpi lax and 8628, but as of the latest bios, which contained a major restructure (released october 2019), it87 no longer works at all. Prior to this bios update, it used to find two chips, one at 0xa40, and one at 0xfff8. Now, it only finds the latter.

jpsollie, 2020-01-08:
I requested gigabytes help for getting the documents, view screenshot for their reply.
Let's hope something pops out! 
[image]

prosac, 2020-05-02:
Ohh! That looks better!
 
(As you can imagine I am here because of the same issue)

marcosscriven, 2020-05-03:
@jpsollie - did you hear anything back?

jpsollie, 2020-05-03:
Yes, I'm sorry, tried a few times to pull it out of their noses, but I failed ... sorry "we don't provide this info to end-users".  I'd need to find a new job in an IT company ... but the substitute trick is easier, I guess

zeus86, 2020-06-28:
> I was able to make it work by adding `acpi_enforce_resources=lax` to my kernel parameters and forcing a different chip ID with `modprobe it87 force_id=0x8628` and `sensors` seems to be showing consistent info

Thank you, exactly this worked for me on an Ubuntu 20.04 with an `Gigabyte x570 Gaming X` equipped with an `AMD Ryzen 5 3700+`:
```
~#  sensors
k10temp-pci-00c3
Adapter: PCI adapter
Tdie:         +45.0°C  (high = +70.0°C)
Tctl:         +45.0°C  

acpitz-acpi-0
Adapter: ACPI interface
temp1:        +16.8°C  (crit = +20.8°C)
temp2:        +16.8°C  (crit = +20.8°C)

it8628-isa-0a40
Adapter: ISA adapter
in0:         912.00 mV (min =  +0.00 V, max =  +3.06 V)
in1:           2.04 V  (min =  +0.00 V, max =  +3.06 V)
in2:           2.03 V  (min =  +0.00 V, max =  +3.06 V)
in3:           2.02 V  (min =  +0.00 V, max =  +3.06 V)
in4:           1.09 V  (min =  +0.00 V, max =  +3.06 V)
in5:         888.00 mV (min =  +0.00 V, max =  +3.06 V)
in6:           1.37 V  (min =  +0.00 V, max =  +3.06 V)
3VSB:          3.29 V  (min =  +0.00 V, max =  +6.12 V)
Vbat:          3.22 V  
fan1:        2909 RPM  (min =    0 RPM)
fan2:        1112 RPM  (min =    0 RPM)
fan3:        1144 RPM  (min =    0 RPM)
fan4:        2149 RPM  (min =    0 RPM)
fan5:           0 RPM  (min =    0 RPM)
temp1:        +43.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp2:        +41.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp3:        +45.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = disabled
temp4:        +44.0°C  
temp5:        +40.0°C  
temp6:        +59.0°C  
intrusion0:  ALARM
```

foxxx0, 2020-06-28:
Is there a way to patch the sources a bit (maybe add this id 0x8628 to the list of supported device IDs) to remove the need for acpi_enforce_resources=lax boot option?

ChrisCheney, 2020-07-28:
Its not the datasheet but the hackintosh guys appear to have support for the IT8688E in their VirtualSMC kext, so it might be useful to see how they get data from it.

https://github.com/acidanthera/VirtualSMC

This was when it was added to the repo, there has been some refactoring since this commit.

https://github.com/acidanthera/VirtualSMC/pull/14

aaronjensen, 2020-07-28:
I believe there's a path to creating an unofficial, non-datasheet backed kernel patch to get us there, then a motherboard definition here to give everything names. It'd require a manual mapping process to figure out what maps to what, which takes some doing. 

This is what would need to be patched, I believe: https://elixir.bootlin.com/linux/v5.7.8/source/drivers/hwmon/it87.c

ChrisCheney, 2020-07-28:
Yea, about half the it87 devices supported in the kernel have no datasheet according to the it87 kernel documentation. So they were written by trial and error.

aaronjensen, 2020-07-28:
It looks like @gamanakis (thanks!) has a patch:

https://github.com/gamanakis/it87/tree/it8688E
I was able to get it by cloning, checking out the `it8688E` branch and then:

```
make
sudo make install
sudo modprobe it87 ignore_resource_conflict=1
```

dkms should work, but when I installed with dkms (`sudo make dkms`) it did not pick up the new driver.

chironjit, 2020-08-24:
> It looks like @gamanakis (thanks!) has a patch:
> 
> https://github.com/gamanakis/it87/tree/it8688E
> I was able to get it by cloning, checking out the `it8688E` branch and then:
> 
> ```
> make
> sudo make install
> sudo modprobe it87 ignore_resource_conflict=1
> ```

This worked for me, thank you :)

ClodoaldoPinto, 2020-10-07:
> I was able to make it work by adding `acpi_enforce_resources=lax` to my kernel parameters and forcing a different chip ID with `modprobe it87 force_id=0x8628` and `sensors` seems to be showing consistent info:
> 
It works for me too but I get this warning at boot:
```
out 07 09:18:29 d3.localdomain kernel: it87: Found IT8628E chip at 0xa40, revision 1
out 07 09:18:29 d3.localdomain kernel: it87: Beeping is supported
out 07 09:18:29 d3.localdomain kernel: ACPI Warning: SystemIO range 0x0000000000000A45-0x0000000000000A46 conflicts with OpRegion 0x0000000000000A45-0x0000000000000A46 (\GSA1.SIO1) (20200528/utaddress-204)
out 07 09:18:29 d3.localdomain kernel: ACPI: This conflict may cause random problems and system instability
out 07 09:18:29 d3.localdomain kernel: ACPI: If an ACPI driver is available for this device, you should use it instead of the native driver
out 07 09:18:29 d3.localdomain kernel: it87: Found IT8628E chip at 0xfff8, revision 15
out 07 09:18:29 d3.localdomain kernel: it87: Beeping is supported
out 07 09:18:29 d3.localdomain kernel: fuse: init (API version 7.31)
out 07 09:18:29 d3.localdomain systemd[1]: Finished Load Kernel Modules.
```

lcpc78, 2020-11-11:
I was able to get this working on my X570 I AORUS PRO WIFI

sudo make dkms
sudo modprobe it87 ignore_resource_conflict=1

I did not have to to put the acpi_encore_resources=lax kernel parameter.

MajorGonzo, 2020-11-15:
> It looks like @gamanakis (thanks!) has a patch:
> 
> https://github.com/gamanakis/it87/tree/it8688E
> I was able to get it by cloning, checking out the `it8688E` branch and then:
> 
> ```
> make
> sudo make install
> sudo modprobe it87 ignore_resource_conflict=1
> ```
> 
> dkms should work, but when I installed with dkms (`sudo make dkms`) it did not pick up the new driver.
> 
> If that works, I believe creating a `/etc/modprobe.d/it87.conf` with:
> 
> ```
> options it87 ignore_resource_conflict=1
> ```
> 
> should make it permanent.
> 
> One note is that it only appears to report 5 fan speeds, I'm missing sysfan4 or sysfan2, not sure which yet.

I used this method, and it works on my Gigabyte x570 UD.  Also configured it with files suggested by tonyfinn:
https://forum.level1techs.com/t/i-patched-the-it87-temperature-monitoring-driver-need-some-help-testing-it/132675/28
and it looks nice:
```
k10temp-pci-00c3
Adapter: PCI adapter
Tdie:         +43.8°C  (high = +70.0°C)
Tctl:         +43.8°C  

amdgpu-pci-0800
Adapter: PCI adapter
vddgfx:       +0.95 V  
fan1:        1472 RPM  (min = 1000 RPM, max = 3800 RPM)
edge:         +40.0°C  (crit = +94.0°C, hyst = -273.1°C)
power1:       32.20 W  (cap = 145.00 W)

it8688-isa-0a40
Adapter: ISA adapter
in0:           +0.92 V  (min =  +0.00 V, max =  +3.06 V)
in1:           +2.04 V  (min =  +0.00 V, max =  +3.06 V)
in2:           +2.06 V  (min =  +0.00 V, max =  +3.06 V)
in3:           +2.03 V  (min =  +0.00 V, max =  +3.06 V)
in4:           +1.02 V  (min =  +0.00 V, max =  +3.06 V)
in5:           +0.90 V  (min =  +0.00 V, max =  +3.06 V)
in6:           +1.22 V  (min =  +0.00 V, max =  +3.06 V)
3VSB:          +3.26 V  (min =  +0.00 V, max =  +6.12 V)
Vbat:          +3.24 V  
CPU_FAN:      1409 RPM  (min =   10 RPM)
SYS_FAN1:     1078 RPM  (min =   10 RPM)
SYS_FAN2:      968 RPM  (min =   10 RPM)
Chipset fan:  2351 RPM  (min =    0 RPM)
CPU_OPT:         0 RPM  (min =    0 RPM)
SYS1 (rear):   +35.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
SYS2 (front):  +33.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
CPU:           +43.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = AMD AMDSI
PCIe:          +37.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
VRM:           +40.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
Chipset:       +48.0°C  (low  = -16.0°C, high = +90.0°C)  sensor = thermistor
intrusion0:   ALARM
```

t-8ch, 2021-04-05:
For those with Gigabyte mainboards you could try a driver I have written for those boards: 
https://github.com/t-8ch/linux-gigabyte-wmi-driver

It uses the proper (albeit limited) ACPI APIs to get the temperature data.
Please report success/failures at https://github.com/t-8ch/linux-gigabyte-wmi-driver/issues/1

daredeviloff, 2021-08-24:
> It looks like @gamanakis (thanks!) has a patch:
>
> https://github.com/gamanakis/it87/tree/it8688E
> I was able to get it by cloning, checking out the `it8688E` branch and then:
>
> ```
> make
> sudo make [...]

Non-dkms method was working fine until I've decided to try dkms. Now, both installation methods don't work.
My problem is exactly same for both methods:
`modprobe: ERROR: could not insert 'it87': Exec format error`

I've tried to clean all the related files before installing again. I'd appreciate if direct me for a solution.

ohhai, 2021-08-25:
@daredeviloff 
I would check dmesg, it might show more details for this error.

hobbes1069, 2021-11-06:
Just a me too on a Gigabyte B550 Aorus Pro-P on Fedora running bleeding edge kernel:
```
$ uname -r
5.16.0-0.rc0.20211103gitdcd68326d29b.2.fc36.x86_64
```

miquels, 2021-11-24:
I have a _Gigabyte Aorus Elite v2_ motherboard, and booting with `acpi_enforce_resources=lax`, then loading the module with `modprobe it87 force_id=0x8628` works for me as well.

The only weird output from `sensors` is:

```
it8628-isa-0a40
Adapter: ISA adapter
CPU Vcore:     888.00 mV (min =  +0.00 V, max =  +6.12 V)  ALARM <--- ALARM?
[....]
CPU Socket:     +30.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = disabled <-- disabled ?
```

etienne-k, 2022-01-22:
**Gigabyte Aorus Pro x570 Rev 1.0** here (the one without Wifi, but afaik they're pretty much the same), running on 5.16.2-zen1.

Couldn't get anything from `gigabyte_wmi`: The module did not detect anything, even when loaded using `force_load=1`.

Using the module from https://github.com/hannesha/it87 _did not_ work for me; maybe I oversaw something.

However, building the module from https://github.com/gamanakis/it87/tree/it8688E (the it8688E branch!) and loading it with DKMS and the parameter `ignore_resource_conflict=1` _did_ work!
`echo options it87 ignore_resource_conflict=1 | sudo tee /etc/modprobe.d/it87 && make && sudo make dkms`

I also still have the kernel parameter `acpi_enforce_resources=lax`, not sure if it makes a difference.

I'm using the following config in `/etc/sensors.d/gigabyte`, in case it helps anyone: https://gist.github.com/etienne-k/8c5039dbf693b951f7936c8f65589e67

demetera, 2022-01-25:
Method, provided by comment author also working for chip IT8689 (motherboard gigabyte B560M DS3H V2). I did totally the same.
```Found unknown chip with ID 0x8688```
Thanks a lot

PaulJohnson, 2022-02-17:
I've got a B550 AORUS PRO V2 motherboard. `sensors-detect` seems to find two chips:

> Probing for Super-I/O at 0x2e/0x2f
> [...]
> Trying family ITE...                                      Yes
> Found unknown chip with ID 0x8688
> Probing for Super-I/O at 0x4e/0x4f
> [...]
> Trying family ITE...                                      Yes
> Found ITE IT8792E Super IO Sensors                        Success!
>     (address 0xa60, driver `it87')`

So it looks like I've got two different IT87 chips, one at 0x2e/0x2f and one at 0x4e/0x4f, and only the second one gets recognised.

This is what I get from `sensors`:
> [root@eiffel lib]# sensors
it8792-isa-0a60
Adapter: ISA adapter
in0:           1.78 V  (min =  +0.00 V, max =  +2.78 V)
in1:         665.00 mV (min =  +0.00 V, max =  +2.78 V)
in2:           1.05 V  (min =  +0.00 V, max =  +2.78 V)
+3.3V:         1.68 V  (min =  +0.00 V, max =  +2.78 V)
in4:           1.78 V  (min =  +0.00 V, max =  +2.78 V)
in5:           1.50 V  (min =  +0.00 V, max =  +2.78 V)
in6:           2.78 V  (min =  +0.00 V, max =  +2.78 V)  ALARM
3VSB:          1.67 V  (min =  +0.00 V, max =  +2.78 V)
Vbat:          1.64 V  
fan1:           0 RPM  (min =    0 RPM)
fan2:           0 RPM  (min =    0 RPM)
fan3:           0 RPM  (min =    0 RPM)
temp1:        +34.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp2:        -55.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp3:        +34.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
intrusion0:  ALARM
> 
> radeon-pci-0400
Adapter: PCI adapter
temp1:        +43.0°C  (crit = +120.0°C, hyst = +90.0°C)
> 
> k10temp-pci-00c3
Adapter: PCI adapter
Tctl:         +47.9°C  
Tccd1:        +49.2°C`

The board has a fan on CPU_FAN, a water pump on CPU_OPT, and two case fans connected to SYS_FAN2 and SYS_FAN3. All are spinning, but they don't show up in `sensors`. However I've been able to get the following to work by replugging the case fans:

> SYS_FAN4 on the board seems to be fan3 on `sensors`.
> SYS_FAN5 seems to be fan1
> SYS_FAN6_PUMP seems to be fan2.

I can't plug anything in to SYS_FAN1 because my radiator is in the way.

PaulJohnson, 2022-02-17:
Update: I've now got the @gamanakis driver working. `dmesg` has an ACPI address conflict warning, so you need the `ignore_resource_conflict` argument. Sensors config with fans and tentative temperature and voltage names:

         chip "it8688-*"
             label in0 "CPU Vcore"
             label in4 "VCORE SOC"
             label in5 "CPU VDDP"
             label in6 "DRAM A/B"
             label fan1 "CPU_FAN"
             label fan2 "SYS_FAN1"
             label fan3 "SYS_FAN2"
             label fan4 "SYS_FAN3"
             label fan5 "CPU_OPT"
             label temp3 "AMDSI"
             ignore intrusion0
       chip "it8792-*"
             label in0 "CPU VDD18"
             label in1 "DDRVtt A-B"
             label in2 "CHIPSET Core"
             label in4 "CPU VDD18"
             label in6 "PM_2V5"
             label fan3 "SYS_FAN4"
             label fan1 "SYS_FAN5"
             label fan2 "SYS_FAN6"
             label temp1 "MB: PCIe x8"
             ignore temp2
             label temp3 "MB: SYSTEM2"
             ignore intrusion0
        chip "k10temp-*"
             label temp1 "CPU: Tctl"
             label temp3 "CPU: Tccd1"
             label temp4 "CPU: Tccd2"
        chip "radeon-pci-*"
             label temp1 "Video Temp"`

Chris2000SP, 2022-07-25:
I have the Gigabyte X570S AORUS PRO AX and with `modprobe it87 ignore_resource_conflict=1` i got this:

`~/ sensors`

```
it8689-isa-0a40
Adapter: ISA adapter
in0:         936.00 mV (min =  +0.00 V, max =  +3.06 V)
in1:           2.02 V  (min =  +0.00 V, max =  +3.06 V)
in2:           2.03 V  (min =  +0.00 V, max =  +3.06 V)
in3:           2.02 V  (min =  +0.00 V, max =  +3.06 V)
in4:           1.01 V  (min =  +0.00 V, max =  +3.06 V)
in5:         912.00 mV (min =  +0.00 V, max =  +3.06 V)
in6:           1.37 V  (min =  +0.00 V, max =  +3.06 V)
3VSB:          3.36 V  (min =  +0.00 V, max =  +6.12 V)
Vbat:          3.14 V
fan1:         425 RPM  (min =   10 RPM)
fan2:           0 RPM  (min =    0 RPM)
fan3:         828 RPM  (min =    0 RPM)
fan4:         708 RPM  (min =    0 RPM)
fan5:           0 RPM  (min =    0 RPM)
temp1:        +35.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp2:        -55.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp3:        +40.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = AMD AMDSI
temp4:        +47.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp5:        +38.0°C  (low  =  +0.0°C, high = -126.0°C)  sensor = thermistor
temp6:        +45.0°C  (low  = -16.0°C, high = +95.0°C)  sensor = thermistor
intrusion0:  ALARM

zenpower-pci-00c3
Adapter: PCI adapter
SVI2_Core:     1.46 V
SVI2_SoC:    994.00 mV
Tdie:         +40.0°C  (high = +95.0°C)
Tctl:         +40.0°C
SVI2_P_Core:   3.82 W
SVI2_P_SoC:    2.63 W
SVI2_C_Core:   2.63 A
SVI2_C_SoC:    2.65 A

acpitz-acpi-0
Adapter: ACPI interface
temp1:        +16.8°C  (crit = +20.8°C)
temp2:        +16.8°C  (crit = +20.8°C)

it8695-isa-0a60
Adapter: ISA adapter
in0:           1.80 V  (min =  +0.00 V, max =  +2.78 V)
in1:           2.78 V  (min =  +0.00 V, max =  +2.78 V)  ALARM
in2:         937.00 mV (min =  +0.00 V, max =  +2.78 V)
+3.3V:         4.05 V  (min =  +0.00 V, max =  +5.56 V)
in4:           1.80 V  (min =  +0.00 V, max =  +2.78 V)
in5:           1.19 V  (min =  +0.00 V, max =  +2.78 V)
in6:           1.40 V  (min =  +0.00 V, max =  +2.78 V)
3VSB:          3.36 V  (min =  +0.00 V, max =  +5.56 V)
Vbat:          3.12 V
fan1:           0 RPM  (min =    0 RPM)
fan2:         757 RPM  (min =    0 RPM)
fan3:         798 RPM  (min =    0 RPM)
temp1:        +42.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp2:        -55.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp3:        +34.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
intrusion0:  ALARM

amdgpu-pci-0300
Adapter: PCI adapter
vddgfx:      856.00 mV
fan1:           0 RPM  (min =    0 RPM, max = 3000 RPM)
edge:         +48.0°C  (crit = +100.0°C, hyst = -273.1°C)
                       (emerg = +105.0°C)
junction:     +50.0°C  (crit = +110.0°C, hyst = -273.1°C)
                       (emerg = +115.0°C)
mem:          +56.0°C  (crit = +100.0°C, hyst = -273.1°C)
                       (emerg = +105.0°C)
PPT:          12.00 W  (cap = 255.00 W)

nvme-pci-0c00
Adapter: PCI adapter
Composite:    +51.9°C  (low  = -273.1°C, high = +84.8°C)
                       (crit = +84.8°C)
Sensor 1:     +51.9°C  (low  = -273.1°C, high = +65261.8°C)
Sensor 2:     +52.9°C  (low  = -273.1°C, high = +65261.8°C)
```
("zenpower" is a separate package to install)

zeljkostjepanovic, 2022-08-10:
Ubuntu 20.04
Kernel 5.15 via linux-generic-hwe-20.04
MB: Gigabyte B550 Aorus Elite V2

This worked for me:

```
git clone git@github.com:gamanakis/it87.git
cd it87
git switch it8688E
make
sudo make install
sudo modprobe it87 mmio=1
sudo sensors-detect
sensors
sudo touch /etc/modprobe.d/it87.conf
echo "options it87 mmio=1" | sudo tee -a  /etc/modprobe.d/it87.conf
echo "it87" | sudo tee -a /etc/modprobe
sudo update-initramfs -u
reboot
```

Didn't need to add any boot parameters in grub config.

```
gigabyte_wmi-virtual-0
Adapter: Virtual device
temp1:        +30.0°C  
temp2:        +32.0°C  
temp3:        +37.0°C  
temp4:        +34.0°C  
temp5:        +35.0°C  
temp6:        +37.0°C  

it8688-isa-0a40
Adapter: ISA adapter
in0:         540.00 mV (min =  +0.00 V, max =  +3.06 V)
in1:           2.00 V  (min =  +0.00 V, max =  +3.06 V)
in2:           2.02 V  (min =  +0.00 V, max =  +3.06 V)
in3:           2.00 V  (min =  +0.00 V, max =  +3.06 V)
in4:           1.19 V  (min =  +0.00 V, max =  +3.06 V)
in5:         888.00 mV (min =  +0.00 V, max =  +3.06 V)
in6:           1.37 V  (min =  +0.00 V, max =  +3.06 V)
3VSB:          3.26 V  (min =  +0.00 V, max =  +6.12 V)
Vbat:          3.02 V  
fan1:         919 RPM  (min =   10 RPM)
fan2:           0 RPM  (min =    0 RPM)
fan3:         793 RPM  (min =    0 RPM)
fan4:           0 RPM  (min =    0 RPM)
fan5:           0 RPM  (min =    0 RPM)
temp1:        +30.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp2:        +32.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp3:        +37.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = AMD AMDSI
temp4:        +34.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp5:        +35.0°C  (low  =  +0.0°C, high = -125.0°C)  sensor = thermistor
temp6:        +37.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
intrusion0:  OK

acpitz-acpi-0
Adapter: ACPI interface
temp1:        +16.8°C  (crit = +20.8°C)
temp2:        +16.8°C  (crit = +20.8°C)

k10temp-pci-00c3
Adapter: PCI adapter
Tctl:         +37.4°C  
Tccd1:        +36.8°C  
Tccd2:        +33.0°C  

amdgpu-pci-0500
Adapter: PCI adapter
fan1:             N/A
edge:         +39.0°C  (crit = +120.0°C, hyst = +90.0°C)

```

jimklimov, 2022-11-12:
Thanks, wanted to confirm that this latest comment https://github.com/lm-sensors/lm-sensors/issues/154#issuecomment-1210022521 was best detailed and worked for my "Gigabyte X570S UD" too. Kudos to all who laid this path above :)

The OS is Proxmox 7.x (based on Debian, kernel currently 5.15.64-pve) and needed `pve-headers` instead of `linux-headers` to build the custom module. 

The `acpi_enforce_resources=lax` did not seem accepted by this kernel - never appeared in `/proc/cmdline` and had no effect for enabling the load of stock driver module, at least.

cuihaoleo, 2022-12-07:
Gigabyte B650M AORUS ELITE AX + Arch Linux 6.0.11-zen1-1-zen kernel

I am able to read fanspeed and other sensors as well as get pwm controls under `/sys/class/hwmon` using the workaround here, i.e.:
- Add `acpi_enforce_resources=lax` to the kernel parameter.
- Load it87 module with `force_id` parameter: `modprobe it87 force_id=0x8628` (`ignore_resource_conflict` is no longer a valid parameter)

Mr-Figs, 2022-12-30:
Fedora 36, kernel 6.0.15, Gigabyte Z390 AORUS PRO WIFI

Thanks to everyone here for the great work. I was able to get `sensors` working on my system with the procedure in #154, after a few modifications. My method:

```
# Build module and test
git clone git@github.com:gamanakis/it87.git
cd it87
git switch it8688E
sudo make dkms
sudo modprobe it87 mmio=1
sensors  # make sure it8688 shows up

# Auto-load it87 module
echo "options it87 mmio=1" | sudo tee -a  /etc/modprobe.d/it87.conf
echo "it87" | sudo tee -a /etc/modules-load.d/it87.conf
```

I had to install the `dkms` package in order for the `make dkms` recipe to work (just `sudo dnf install dkms` on Fedora).

I also had to add `acpi_enforce_resources=lax` to my linux cmdline. Otherwise, it's impossible to insert the `it87` module.

I did not find it necessary to add `force_id=8628` or `ignore_resource_conflict=1` on my system, but those could be added to `/etc/modprobe.d/it87.conf` if needed. I decided to leave `mmio=1` enabled, although the module still works without it.

I don't think that `/etc/modprobe` is the correct file for autoloading modules. On distros with `systemd`, the correct place is `/etc/modules-load.d/foo.conf`. Might also be `/etc/modules` on some distros. I also don't see a reason to touch the initramfs image, since the module doesn't need to be loaded so early.

Even with this all set up, `sensors-detect` still doesn't seem to work properly:
```
$ sudo sensors-detect
# --cut--
Some Super I/O chips contain embedded sensors. We have to write to
standard I/O ports to probe them. This is usually safe.
Do you want to scan for Super I/O sensors? (YES/no): 
Probing for Super-I/O at 0x2e/0x2f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               No
Trying family `ITE'...                                      Yes
Found unknown chip with ID 0x8688
Probing for Super-I/O at 0x4e/0x4f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               No
Trying family `ITE'...                                      Yes
Found `ITE IT8792E Super IO Sensors'                        Success!
    (address 0xa60, driver `it87')
# --cut--
```

However, even without running `sensors-detect`, I see the it8688 right away after inserting the module and running `sensors`:
```
$ sensors
# --cut--
it8688-isa-0a40
Adapter: ISA adapter
CPU Vcore:       1.36 V  (min =  +0.35 V, max =  +1.45 V)
+3.3V:           3.35 V  (min =  +3.21 V, max =  +3.41 V)
+12V:           12.17 V  (min = +11.66 V, max = +12.38 V)
+5V:             5.04 V  (min =  +4.86 V, max =  +5.16 V)
Vcore SoC:       0.00 V  (min =  +0.90 V, max =  +1.26 V)  ALARM
CPU Vddp:        1.30 V  (min =  +0.85 V, max =  +0.95 V)  ALARM
DRAM A/B:        1.37 V  (min =  +1.10 V, max =  +1.60 V)
3 Vsb:           3.41 V  (min =  +0.00 V, max =  +6.12 V)
Vbat:            3.22 V  
CPU fan:        486 RPM  (min =  300 RPM)
Sys1_Fan:       236 RPM  (min =  300 RPM)  ALARM
Sys2_Fan:         0 RPM  (min =  300 RPM)  ALARM
Sys3_PCH_Fan:     0 RPM  (min =  300 RPM)  ALARM
CPU_Opt_Fan:      0 RPM  (min =  300 RPM)  ALARM
Sys1 Temp:      +30.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
Ambient:        +42.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
CPU Temp:       +26.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = Intel PECI
PCIe_x16 Temp:  +43.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
VRM_MOS Temp:   +36.0°C  (low  =  +0.0°C, high = -123.0°C)  sensor = thermistor
PCH Temp:       -55.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
# --cut--
```

I found a decent `lm-sensors` mapping here: https://linustechtips.com/topic/1095169-x570-aorus-pro-wifi-with-3900x-and-32gig-of-ram/?do=findComment&comment=13843717. Not guaranteed to be 100% accurate on all systems, but otherwise all the voltages are just `in0`, `in1`, etc. It would take some legwork, but it might be smart to boot up Windows and grab the mapping from HWiNFO for my exact board.

bniels707, 2023-03-31:
Gigabyte X570S Aorus Elite, booting with `acpi_enforce_resources=lax`, followed by `modprobe it87 force_id=0x8628`:

```
sensors it8628-isa-0a40
it8628-isa-0a40
Adapter: ISA adapter
in0:         768.00 mV (min =  +0.00 V, max =  +3.06 V)
in1:           2.04 V  (min =  +0.00 V, max =  +3.06 V)
in2:           2.03 V  (min =  +0.00 V, max =  +3.06 V)
in3:           2.03 V  (min =  +0.00 V, max =  +3.06 V)
in4:         996.00 mV (min =  +0.00 V, max =  +3.06 V)
in5:         912.00 mV (min =  +0.00 V, max =  +3.06 V)
in6:           1.22 V  (min =  +0.00 V, max =  +3.06 V)
3VSB:          3.34 V  (min =  +0.00 V, max =  +6.12 V)
Vbat:          3.10 V  
fan1:         656 RPM  (min =    0 RPM)
fan2:         728 RPM  (min =    0 RPM)
fan3:           0 RPM  (min =    0 RPM)
fan4:         482 RPM  (min =    0 RPM)
fan5:         680 RPM  (min =    0 RPM)
fan6:           0 RPM  (min =    0 RPM)
temp1:        +27.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp2:        +30.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp3:        +27.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = disabled
temp4:        +33.0°C  
temp5:        +30.0°C  
temp6:        +43.0°C  
intrusion0:  ALARM
```

mirh, 2023-04-28:
For the records, the latest work is happening here
https://github.com/frankcrawford/it87

coffnix, 2023-11-12:
> Method, provided by issue author also working for chip IT8689 (motherboard gigabyte B560M DS3H V2). I did totally the same except the message I got when detecting the sensors. `Found unknown chip with ID 0x8689` Thanks a lot

Working for chip IT8689 (motherboard gigabyte Z790D). I did totally the same except the message I got when detecting the sensors. Found unknown chip with ID 0x8689

`options it87 force_id=0x8628 ignore_resource_conflict=1
`
or

`modprobe it87 force_id=0x8628 ignore_resource_conflict=1`

Thanks a lot

howels, 2024-01-05:
On a Gigabyte X570S UD I also had `Found unknown chip with ID 0x8689` and no fan speed with kernel 5.19 or 6.2.

Adding kernel option `acpi_enforce_resources=lax` and `modprobe it87 force_id=0x862` allows fan speed to be reported.

MichalLebeda, 2024-01-08:
> On a Gigabyte X570S UD I also had `Found unknown chip with ID 0x8689` and no fan speed with kernel 5.19 or 6.2.
> 
> Adding kernel option `acpi_enforce_resources=lax` and `modprobe it87 force_id=0x862` allows fan speed to be reported.

Hi, is `force_id=0x862` the correct id? I think the ids should be 4 digits and this one does not work

howels, 2024-01-08:
> Hi, is `force_id=0x862` the correct id? I think the ids should be 4 digits and this one does not work

typo error - I meant `force_id=0x8628`

MajorGonzo, 2024-01-08:
I have a UD and I'm using 8688

 

&mdash;

MichalLebeda, 2024-01-09:
Which FW version do you have? I have X570S US dont know if it makes a difference

misaligar, 2024-01-18:
Are there any guides to make this work for Gigabyte B650 Aorus Elite EX using Debian 12 Stable?

briha, 2024-09-09:
For anyone looking to get this ITE chip (or others, like IT8689 on Gigabyte's X670 Aorus Elite, for example) working, @frankcrawford 's [it87 fork](https://github.com/frankcrawford/it87) contains support for a lot more chips :)

It won't be detected in `sensors-detect`, but it should show up in `sensors` nevertheless.

alexspurling, 2024-09-18:
The following worked for me:

Nobara Linux 40 (6.10.7-200.fsync.fc40.x86_64)
Gigabyte B550i Auorus Pro AX

```
git clone git@github.com:gamanakis/it87.git
cd it87
git switch it8688E
make
sudo make install
sudo modprobe it87 ignore_resource_conflict=true
sudo sensors-detect
sensors
```

At this point you should see the previously missing voltage, temperature and fan sensors under a new heading:

```
it8688-isa-0a40
Adapter: ISA adapter
in0:           1.34 V  (min =  +0.00 V, max =  +3.06 V)
in1:           2.00 V  (min =  +0.00 V, max =  +3.06 V)
in2:           2.03 V  (min =  +0.00 V, max =  +3.06 V)
in3:           2.03 V  (min =  +0.00 V, max =  +3.06 V)
in4:           1.20 V  (min =  +0.00 V, max =  +3.06 V)
in5:         900.00 mV (min =  +0.00 V, max =  +3.06 V)
in6:           1.25 V  (min =  +0.00 V, max =  +3.06 V)
3VSB:          3.29 V  (min =  +0.00 V, max =  +6.12 V)
Vbat:          3.19 V  
fan1:         643 RPM  (min =    0 RPM)
fan2:        2789 RPM  (min =    0 RPM)
fan3:           0 RPM  (min =    0 RPM)
temp1:        +49.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp2:        +52.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp3:        +56.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = AMD AMDSI
temp4:        +21.0°C  (low  =  +0.0°C, high = +127.0°C)  sensor = thermistor
temp5:        +49.0°C  (low  =  +0.0°C, high = -124.0°C)  sensor = thermistor
temp6:        +55.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
intrusion0:  ALARM
```

To ensure that this module is loaded on boot:

```
echo "options it87 ignore_resource_conflict=true" | sudo tee -a  /etc/modprobe.d/it87.conf
echo "it87" | sudo tee -a /etc/modules-load.d/it87.conf
```

Unlike @future-figs, I found that the final command write the `it87.conf` file was necessary for the module to be loaded on boot.

From here, I used CoolerControl to set up a custom CPU fan curve profile to keep the fans quiet at idle temps.

alexspurling, 2024-11-27:
Here's an update. I found that updating the Linux kernel removes the custom modprobe module and it must be reinstalled. After updating to Nobara 6.11.8-200 I had to re-run the following steps to get the it87 module installed again:

```
sudo make install
sudo modprobe it87 ignore_resource_conflict=true
sudo sensors-detect
```

I'd also like to make a note here how to use these sensors to control your CPU fan speed using Cooler Control (so this can help me or anyone else in the future):

* With the above steps completed, you should see a new sensor group called it8688 which contains temperature and fan speeds.
* Go to Profiles & Functions and click New Profile.
* Rename the profile to something like CPU Fan Curve and change the type to Graph
* Select the temp source
* Adjust the points on the graph to match the fan duty cycle that you want for your CPU fan. Click Apply
* Find the fan that you would like to adjust on the left and select it.
* Click on Profiles and then under the Profile drop-down select the new profile you just created.
* Click Apply and that's it!

[image]

[image]

If at some point, the sensors become unavailable (maybe because if a kernel update), be sure to restart the Cooler Control daemon in order to refresh the sensors that you see on the left.

TheHairforce, 2025-01-21:
@alexspurling 
Thank you for the detailed Instruction! - With that I finally got the Sensors to show up on my  Gigabyte B550i as well.

Though sadly, after that the Issue for me got weirder...
Even with the added conf files under modprobe and module-load on boot the sensors won't show up on CoolerControl, lm-sensors or anywhere Else. Even when I run sensors-detect again lm-sensors isn't able to 'find' them.

With every reboot (Using Fedora KDE 41), I have to go through the 'Whole Process'(!) again before they show up once more. - After I reboot -> Gone again. Any Idea why that could be?

Greetings and thanks!

hxii, 2025-05-12:
I hate to be the one bumping this, but after adding the module to autoload and rebooting it fails to load due to:

```
hxii@reincarna ~/m/it87 (master)> sudo modprobe -v it87 force_id=0x8689 ignore_resource_conflict=1
insmod /lib/modules/6.14.6-200.nobara.fc42.x86_64/kernel/drivers/hwmon/it87.ko.zst force_id=0x8689 ignore_resource_conflict=1 force_id=0x8689 ignore_resource_conflict=1
modprobe: ERROR: could not insert 'it87': No such device
hxii@reincarna ~/m/it87 (master) [1]>
```

While sensors-detect does find it:
```
hxii@reincarna ~/m/it87 (master)> sudo sensors-detect --auto | grep "8689" -C 3
...
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               No
Trying family `ITE'...                                      Yes
Found unknown chip with ID 0x8689
Probing for Super-I/O at 0x4e/0x4f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
```

Any ideas?

coffnix, 2025-05-12:
> sensors-detect --auto | grep "8689" -C 3

works here on my openSUSE 15.6:

[image]

frankcrawford, 2025-05-13:
@hxii firstly, what it87 module are you loading, the one shipped with the mainline kernel, or the one built from an external repo?  Note that the mainline kernel does not currently handle IT8689E chipset.

If from an external repo, how are you building and installing it, with dkms, or manually?

Note that `sensors-detect` does not detect chips that are not in the mainline kernel.  You will see that `sensors-detect` lists it as an `unknown chip` and hence not handled by the mainline kernel.

hxii, 2025-05-22:
> [@hxii](https://github.com/hxii) firstly, what it87 module are you loading, the one shipped with the mainline kernel, or the one built from an external repo? Note that the mainline kernel does not [...]

I'm loading it87 built from `https://github.com/frankcrawford/it87`, which I've built via `sudo make`, and then loaded via `sudo modprobe it87 ignore_resource_conflict=true force_id=0x8689`.
This would result in a `No Such Device` message, but not all the time which is the weird part.

I just tried via `sudo make dkms` and loaded it again, and it seemed to have worked as verified by `dmsg`.

frankcrawford, 2025-05-24:
@hxii first off, if it is a IT8689 chip then you don't need the `force_id=0x8689` option, and it would be best to leave it off, as it can have unexpected consequences on boards with multiple chips.  It also bypasses some tests related to the chip.

And secondly, some recent chips will lock up with `sensors-detect`, so if you load the `it87` after running it, it will not be detected until after a reboot.

frankcrawford, 2025-05-30:
What options are you using to get `unknown parameter `mmio'`?  That option does not exist in the mainline version.

However, just as important as it already identifies it as an IT8628E, you should not need to use `force_id=0x8628`.  That option is mostly needed for an unknown chip to be mapped to a known chip.  Especially if you have 2 chips on your motherboard using `force_id` can cause issues detecting the second of the chips.

dElogics, 2025-06-08:
Using https://github.com/frankcrawford/it87, it's working for IT8689 which is in Z790 UD motherboard. Fan speed control is also working.
load the module using parameters -- 
modprobe it87.ko ignore_resource_conflict=1 force_id=0x8689

frankcrawford, 2025-06-28:
@GittyMoober I'm not sure why you are using that repo, which has not been updated in over 6 years, but you probably want to use my repo, [https://github.com/frankcrawford/it87](https://github.com/frankcrawford/it87), which is pretty much continuously update with fixes for kernel changes and new info from the community.

Especially, the IT8688E is standard, not a separate branch, and the `fall through` error has been long fixed.

GittyMoober, 2025-06-29:
@frankcrawford Well the main reason was because that was the only thing I could find with step by step instructions of how to do it. I did actually see your repo but I couldn't figure out how to use the information provided there to actually install something. I even tried tried switching the repo in Alex's post with yours to see if it would work but it didn't. The fallthrough is indeed gone but I'm still getting the vmlinux error and modprobe does nothing (not even getting an error like before).

frankcrawford, 2025-06-29:
@GittyMoober  in the middle of the README it has `Building & Installing`, which is very simple:

```
* make clean
* make
* sudo make install
*  sudo modprobe it87 ignore_resource_conflict=true
* sensors
```

Do not use `sensors-detect` as it is extremely broken for any chips it does not currently know about.  In fact it unloads modules so will undo your modprobe.

However, if you are having issues, go to the GIT repo and log an issue with the errors you are seeing and we can look at it further.

However, I assume you will want it to work after a reboot, you will also need to do something like:

```
echo "options it87 ignore_resource_conflict=true" | sudo tee -a  /etc/modprobe.d/it87.conf
```

BTW `Skipping BTF generation for it87.ko due to unavailability of vmlinux` is only a warning, so can be ignored.  It is also almost totally unrelated, as it is a distribution item that is not in the right place.

Have you checked what the running kernel actually says, i.e. `sudo dmesg | grep it87` to see what the module status is?

And finally, if you have set it up as above and are not seeing anything, reboot, and then see if it loads properly.

[2 more replies]