lm-sensors/lm-sensors #155: Support for Asus Z9PA-D8

twsI, 2019-01-07:
Running `sensors-detect` tells me
```
Driver `nct6775':
  * ISA bus, address 0x290
    Chip `Nuvoton NCT5573D/NCT5577D/NCT6776F Super IO Sensors' (confidence: 9)
```
But this driver only has support for 3 fans which the `sensors` commands whows without RPM while my motherboard has a total of 8 fan headers. What can I do to support this board? I currently don't even know, where to start.

twsI, 2019-05-20:
https://media.digikey.com/pdf/Data%20Sheets/Nuvoton%20PDFs/NCT6776F,D.pdf

RonnySvedman, 2020-11-14:
what is the full output of sensors-detect? I suspect that board may have an ITE it87 derivative chip for fan controls in addidtion to the NCT? Many Asus boards use the it87-one-or-the-other and it has been a pain for linux users since before 2010 until this day. IFF you find an unknown ITE chip reported in the output you could first try modprobe it87, or compile a newer version than the really old one in the kernel from unofficial sources, for example [https://github.com/a1wong/it87]

twsl, 2020-11-15:
You can physically locate the chip on the motherboard. But unfortunately I am not using the board anymore, so I cannot provide any full logs right now.