lm-sensors/lm-sensors #160: Support for AMD Opteron X3216 APU

somethingGoneWrong, 2019-01-19:
Running sensors-detect output:
Sorry, no sensors were detected.
```
...
Probing for `FSC Syleus'...                                 No
Client found at address 0x77
Probing for `Asus Mozart-2'...                              No

Either your system has no sensors, or they are not supported, or
they are connected to an I2C or SMBus adapter that is not
supported. If you find out what chips are on your board, check
http://www.lm-sensors.org/wiki/Devices for driver status.
```
but if i run sensor it can get 3 value from motherboard:
```
root@pve:~# sensors
fam15h_power-pci-00c4
Adapter: PCI adapter
power1:       26.41 mW (avg =   2.60 mW, interval =   0.01 s)
                       (crit =  15.00 W)

amdgpu-pci-0008
Adapter: PCI adapter
fan1:             N/A
temp1:        +28.0°C  (crit =  +0.0°C, hyst =  +0.0°C)

k10temp-pci-00c3
Adapter: PCI adapter
temp1:        +28.2°C  (high = +70.0°C)
                       (crit = +100.0°C, hyst = +99.0°C)
```
I do not understand if it found all the sensors or not. thx

olysonek, 2019-01-23:
`sensors-detect` tries to identify IDs of PCI devices and Super IO devices that are present on the system. `sensors` reads values from devices that are supported by the kernel, using `/sys/class/hwmon`. They work differently. `sensors-detect` can detect sensors that are not displayed by `sensors` and vice versa.

Some device IDs are currently missing in `sensors-detect`, I'll add them.

olysonek, 2019-01-23:
Can you try running the current sensors-detect? It should now detect the thermal and power sensors of the CPU. I don't know how to detect the GPU sensors.
https://github.com/lm-sensors/lm-sensors/blob/master/prog/detect/sensors-detect

somethingGoneWrong, 2019-01-23:
> Can you try running the current sensors-detect? 

```
BLA BLA BLA
Client found at address 0x77
Probing for `Asus Mozart-2'...                              No

Now follows a summary of the probes I have just done.
Just press ENTER to continue: 

Driver `k10temp' (autoloaded):
  * Chip `AMD Family 15h thermal sensors' (confidence: 9)

Driver `fam15h_power' (autoloaded):
  * Chip `AMD Family 15h power sensors' (confidence: 9)

No modules to load, skipping modules configuration.
```

> I don't know how to detect the GPU sensors.

from what I know, there is no driver to use the graphics unit integrate in the APU.

groeck, 2019-04-25:
drivers/gpu/drm/amd/amdgpu/amdgpu_drv.c
drivers/gpu/drm/amd/amdgpu/amdgpu_pm.c