lm-sensors/lm-sensors #163: Adding support for USB HID hwmon devices.

jameshilliard, 2019-02-03:
What would be the best approach for adding a kernel driver for sensors that use a custom USB HID based protocol? There's a number of sensor/fan devices on the market that seem to use USB HID for communications such as CPU Coolers and power supplies.

Would the approach used [here](https://github.com/jameshilliard/CorsairLink/blob/f155421a632756a3df8c3ef7cd7e955b87a743a6/h80/clink.c) and [here](https://github.com/jameshilliard/CorsairLink/blob/f155421a632756a3df8c3ef7cd7e955b87a743a6/h80/h80i.c) work? I have a similar CPU cooler to the ones supported there, the protocol is different but it's also USB HID based.

jameshilliard, 2019-02-03:
So seems that the Asetek variant of the corsair water cooler I have is not actually HID based, only the Coolit variant seems to be. It seems to be a custom binary USB protocol.

groeck, 2019-02-03:
I'll be happy to accept patches for the hwmon subsystem in the upstream kernel. The drivers pointed to would need some rework, but they are a good start. Please send patches or patch series to linux-hwmon@vger.kernel.org.