lm-sensors/lm-sensors #164: ASUS laptop GL553 no fan speed

nhhuayt, 2019-02-07:
I run sensors-detect but no fan control. And low fan speed make my laptop too hot.
Please help!

`sudo sensors
coretemp-isa-0000
Adapter: ISA adapter
Physical id 0:  +60.0°C  (high = +100.0°C, crit = +100.0°C)
Core 0:         +57.0°C  (high = +100.0°C, crit = +100.0°C)
Core 1:         +57.0°C  (high = +100.0°C, crit = +100.0°C)
Core 2:         +56.0°C  (high = +100.0°C, crit = +100.0°C)
Core 3:         +57.0°C  (high = +100.0°C, crit = +100.0°C)

acpitz-virtual-0
Adapter: Virtual device
temp1:        +60.0°C  (crit = +98.0°C)

asus-isa-0000
Adapter: ISA adapter
cpu_fan:       -1 RPM
`

`sudo ./find-driver asus-isa-0000
Driver: asus-nb-wmi
Module: asus_wmi
`

glats, 2019-10-25:
same problem here.

jkladola, 2019-11-13:
Did you found any solution ? Same here with Asus Strix GL702ZC.

Env:
Debian 10
Linux Kernel 4.19.0-6-amd64
lm-sensors/stable,now 1:3.5.0-3 amd64

acpitz-acpi-0
Adapter: ACPI interface
Temp:         +73.0°C  (crit = +98.0°C)

amdgpu-pci-0c00
Adapter: PCI adapter
vddgfx:      750.00 mV 
GPU Temp:     +69.0°C  (crit = +94.0°C, hyst = -273.1°C)
GPU Watt:      7.17 W  (cap =  68.00 W)

k10temp-pci-00c3
Adapter: PCI adapter
Core0 Temp:   +73.2°C  (high = +70.0°C)
Core0 Temp:   +73.2°C

azamet90, 2019-11-13:
same problem with G731GV

jkladola, 2019-11-13:
> same problem with G731GV

Hey azamet90, Until someone come to help us Let us know about your Environment.

Which Distro, Kernel Version and lm-sensors version you are using?

azamet90, 2019-11-13:
> > same problem with G731GV
> 
> Hey azamet90, Until someone come to help us Let us know about your Environment.
> 
> Which Distro, Kernel Version and lm-sensors version you are using?

im using the latest version of lm-sensors from github source installed 
Linux localhost 5.3.0-kali1-amd64 #1 SMP Debian 5.3.7-1kali2 (2019-11-04) x86_64 GNU/Linux

azamet90, 2019-11-16:
> > same problem with G731GV
> 
> Hey azamet90, Until someone come to help us Let us know about your Environment.
> 
> Which Distro, Kernel Version and lm-sensors version you are using?

im not so good on linux if you need some outputs please sayme the commands i need to put to terminal then i can sent you the outputs too

jkladola, 2019-11-18:
I my self is stuck here. I can't think of a way to get around it. We have to wait until some one from Lm-Sensors team help us.

RonnySvedman, 2020-11-14:
is there by chance some unknown it87 derivative in your sensors-config output? If so you can try to get unofficial sources for it87 module from for example [https://github.com/a1wong/it87] or search for whoever is developing them now. the version in the kernel is really old