lm-sensors/lm-sensors #167: ITE IT8613E

kajzersoze, 2019-02-19:
Support for this driver is not in the kernel.
However, there is a support for it [here](https://github.com/a1wong/it87)
I've built that driver in the kernel.
dmesg reports it's loaded
```
$ dmesg | grep 8613                                                                                                                                                                                              
[    1.065628] it87: Found IT8613E chip at 0xa30, revision 1
```

`sensors` doesn't get it though, sensors-detect finds it but thinks that driver isn't written 
```
Found `ITE IT8613E Super IO Sensors'                        Success!
    (address 0xa30, driver `to-be-written')
```

```
Driver `to-be-written':
  * ISA bus, address 0xa30
    Chip `ITE IT8613E Super IO Sensors' (confidence: 9)
```

How can I make this work?

groeck, 2019-04-25:
Either use the out-of-tree driver, or submit a series of patches to align the in-kernel driver with the out-of-tree driver.

kajzersoze, 2019-04-25:
Out-of-tree driver isn't recognized by `sensors` , that's the problem.

Anyway, I get it, thanks for taking the time.

closed 2019-04-25 by kajzersoze

groeck, 2019-04-25:
From the sensors-detect command 's perspective, out-of-tree drivers don't exist. "to-be-written' is therefore appropriate. I don't know what you mean with "sensors doesn't get it though". The sensors command _should_ show the sensors. sensors-detect has nothing to do with the sensors command (or with kernel drivers).

kajzersoze, 2019-04-25:
I meant it shows nothing and driver is loaded.
sensors-detect was bonus info.

groeck, 2019-04-25:
Question then is if there was some other message in the kernel log, such a s an IO address conflict.

kajzersoze, 2019-04-25:
Can't really remember, it was a while ago.
It's not that important to me anymore, in fact I forgot about it.
Basically the only thing I can't get is fan RPM, which that driver provides while kernel one doesn't.

Thing is, under Ubuntu it worked, but under Gentoo it didn't, and I use Gentoo, Ubuntu was just for the testing at that time.
Really weird, didn't try it much, if and when I try it again I'll provide more info.

groeck, 2019-04-25:
Wild guess, but that very much sounds like a resource conflict.

kajzersoze, 2019-04-25:
Right, don't know how to solve that though, Gentoo doesn't have DKMS
I tried it again, same story, here's the dmesg
```
[  +0.253996] it87: loading out-of-tree module taints kernel.
[  +0.000740] it87: it87 driver version v1.0-48-g40bec4b
[  +0.000071] it87: Found IT8613E chip at 0xa30, revision 1
[  +0.000026] it87: Beeping is supported
[  +0.000009] ACPI Warning: SystemIO range 0x0000000000000A35-0x0000000000000A36 conflicts with OpRegion 0x0000000000000A35-0x0000000000000A36 (\SENP) (20181213/utaddress-213)
[  +0.000008] ACPI: If an ACPI driver is available for this device, you should use it instead of the native driver

```

groeck, 2019-04-26:
acpi_enforce_resources=lax boot parameter, or ignore_resource_conflict module parameter. FWIW, that is explained in the README file of the out of tree driver.

kajzersoze, 2019-04-26:
That works, but I figured it's not worth it to risk system crashes and whatnots because of this.

I think changing the kernel module with pieces from that driver is the way to go, when I find some time maybe I'll play with that.

Thanks for all the help!

groeck, 2019-04-26:
Using pieces from the out-of-tree driver won't make a difference. You would still get the conflict.

kajzersoze, 2019-04-26:
You know, I guess there's a good reason why there's no support for this chip in the kernel.
I think I'll leave it be.

groeck, 2019-04-26:
Uuh ... no. The out-of-tree driver was originally written by me. Problem is that I never had time to clean it up and send it upstream. Various people got more and more demanding, expecting me to do it for them, without any desire or willingness to help. This went on until I got so annoyed that I pulled the driver from github, so now all that remains is various clones of it. And still no one is willing to step up and go through the process of upstreaming it. I am sure you'll find the story on the internet if you search for it, including the claims that I only threatened to pull the driver from github to get attention.

kajzersoze, 2019-04-26:
I see, thanks for the explanations.
I'm willing to help, but my knowledge regarding this is very limited.
What I can do is make the out-of-tree code dealing with my chip work inside the kernel code.
I already looked at it and did maybe 90%, there are some differences between the two sources but nothing that can't be done.
However, this conflict problem is making me think of just letting this go.
I know it's solvable by ignoring it but I don't like that.

I have some errors from BIOS in dmesg and I guess it could be related to this conflict
```
[    0.000000] ACPI BIOS Warning (bug): Optional FADT field Pm2ControlBlock has valid Length but zero Address: 0x0000000000000000/0x1 (20181213/tbfadt-624)
[    0.058114] ACPI BIOS Error (bug): Failure creating [\_SB.SMIC], AE_ALREADY_EXISTS (20181213/dswload2-324)
[    0.058126] ACPI BIOS Error (bug): Failure creating [\_SB.SMIB], AE_ALREADY_EXISTS (20181213/dsfield-626)
[    0.059250] ACPI: [Firmware Bug]: BIOS _OSI(Linux) query ignored
[    0.580921] [drm] BIOS signature incorrect 0 0
```

kajzersoze, 2020-01-18:
> Besides, it is not a chip problem. It is the board vendor's BIOS causing the conflict. If you know how to decode the DSDT, you'll find all the information there.

Sorry to bring this old issue after some time.
Can you please tell me what to look for in the decoded DSDT ?

groeck, 2020-01-18:
> > Besides, it is not a chip problem. It is the board vendor's BIOS causing the conflict. If you know how to decode the DSDT, you'll find all the information there.
> 
> Sorry to bring this old issue after some time.
> Can you please tell me what to look for in the decoded DSDT ?

Not really; DSDT content is not standardized to the point where generic advice like this would be possible. All you can do is to find the reserved memory range in DSDT and references to it, and try to understand why it is reserved. Either case, the ACPI error messages above suggest a BIOS bug. The very first step should be to get an updated BIOS from the board vendor.

kajzersoze, 2020-01-18:
BIOS is updated, it's two months old.

Well, I understand, it is what it is.
I tried using out-of-tree driver, with ignore_resource_conflict option and it worked fine for maybe 12 hours or something, then machine powered off out of the blue, on idle, and that never happened when that driver wasn't used.
I don't know is it because resource conflict or something in that driver not done quite right.
Anyway I stopped using it but I'm curious to make it work, if possible.

In case you're interested to take a quick look [here](https://github.com/kajzersoze/.config/blob/master/temp/dsdt.dsl) is the DSDT

groeck, 2020-01-18:
If the out-of-tree driver doesn't work, you are pretty much out of luck. That means that the BIOS (or possibly a second Super-IO chip or embedded controller in the system) _does_ access the chip in parallel, and it does it in a way that conflicts with the driver. Without support from the board vendor and from ITE this is all but impossible to resolve. The out-of-tree driver includes some workarounds trying to handle the situation, but I always had the suspicion that it wasn't reliable. Your observation pretty much confirms that this is the case.

kajzersoze, 2020-01-18:
It was working great, all the readings were right, I thought that was it and started using it.
Everything was really fine.
It seems like fans stopped working for some reason, looking at the logs....
```
Jan 18 12:05:04 kernel: thermal thermal_zone0: critical temperature reached (127 C), shutting down
Jan 18 12:05:04 shutdown[3128]: shutting down for system halt
```

You've done more than enough, thanks for all your work!

groeck, 2020-01-18:
If you look into the DSDT you'll notice that it accesses the chip to get and set thermal limits. That access is unpreotected, so there is no means for the driver to synchronize with the BIOS. So the crash isn't entirely surprising.

kajzersoze, 2020-01-18:
Well yeah, it doesn't report limits right (low high), k10temp does it right though.
temp3 is cpu , temp1 is motherboard.

btw I don't know if it matters for that crash but while I was using it87 I turned off k10temp in the kernel.

```
k10temp-pci-00c3
Adapter: PCI adapter
Tdie:         +33.9°C  (high = +70.0°C)
Tctl:         +33.9°C 

it8613-isa-0a30
Adapter: ISA adapter
in0:         880.00 mV (min =  +0.00 V, max =  +2.81 V)
in1:           1.20 V  (min =  +0.00 V, max =  +2.81 V)
in2:           2.05 V  (min =  +0.00 V, max =  +2.81 V)
in4:           2.06 V  (min =  +0.00 V, max =  +2.81 V)
in5:         968.00 mV (min =  +0.00 V, max =  +2.81 V)
3VSB:          3.30 V  (min =  +0.00 V, max =  +5.61 V)
Vbat:          3.15 V  
+3.3V:         3.21 V  
fan2:        1901 RPM  (min =   13 RPM)
fan3:           0 RPM  (min =   10 RPM)  ALARM
fan5:           0 RPM  (min =   -1 RPM)  ALARM
temp1:        +35.0°C  (low  = +88.0°C, high = +93.0°C)  sensor = thermistor
temp2:        +33.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp3:        +33.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermal diode
intrusion0:  ALARM
```

groeck, 2020-01-18:
Turning off k10temp while using it87 is about as relevant as turning off the lights in your house while using it. So, no, that doesn't matter. Also, k10temp only reports a static maximum temperature of 70 degrees C, which has nothing to do with reality on recent AMD CPUs (and is going to be removed shortly for that very reason).

kajzersoze, 2020-01-18:
I figured but I was thinking saying it wouldn't hurt :)
Running Gentoo, always compiling new kernels, it was nothing to turn it off, figured if it87 works there's no need for k10temp to run as well.

I guess this is it then, very close, but no cigar, still great work!

kajzersoze, 2020-01-19:
btw do you know maybe what is this (\x5cSENP)  which is causing this resource conflict ?

```
kernel: it87: loading out-of-tree module taints kernel.
kernel: it87: it87 driver version v1.0-48-g40bec4b
kernel: it87: Found IT8613E chip at 0xa30, revision 1
kernel: it87: Beeping is supported
kernel: ACPI Warning: SystemIO range 0x0000000000000A35-0x0000000000000A36 conflicts with OpRegion 0x0000000000000A35-0x0000000000000A36 (\x5cSENP) (20190816/utaddress-204)
kernel: ACPI: If an ACPI driver is available for this device, you should use it instead of the native driver
```

groeck, 2020-01-19:
From the DSDT:
Name (SSBS, 0x0A35)
...
OperationRegion (SENP, SystemIO, SSBS, 0x02)
...
    Method (SBYT, 2, NotSerialized)

then RTMP and CTMP use it. Those are in turn called from Scope (_TZ) and ThermalZone (THRM), meaning they are used for thermal monitoring to report the current and critical temperature.

kajzersoze, 2020-01-19:
Thanks, I'll try to play with DSDT and then boot with modified one.

kajzersoze, 2020-01-19:
btw those errors you mentioned earlier are not present anymore.
There are no errors except this bug line :
`ACPI: [Firmware Bug]: BIOS _OSI(Linux) query ignored`

tried to solve it by acpi_osi=Linux , that changed that line to :
`ACPI: [Firmware Bug]: BIOS _OSI(Linux) query honored via cmdline`

Also tried to turn off various cpu and thermal options in BIOS but resources are still used for it87

Playing with DSDT will take some more time.

groeck, 2020-01-20:
It won't help. The point of that code is to prevent driver access if the BIOS ever accesses the region. Once that happens, the driver can no longer access it at all. It is not preventing driver access "until BIOS finishes", it is preventing access forever.

kajzersoze, 2020-01-20:
I see, thanks for clarifying.

kajzersoze, 2020-08-27:
Just to update this, maybe someone in future will find this useful.

Managed to avoid resource conflict with custom DSDT, patched mainline it87 driver and it's working fine.
The only caveat is that I had to return fixed values for thermal zone in DSDT, so if fans ever stop working for some reason the system wouldn't shutdown when the cpu temperature gets very high.

kajzersoze, 2020-08-28:
btw  I don't know about other sensors from out-of-tree driver, but I don't see why IT8613E wouldn't be added in the kernel driver, there's nothing special to it, I've added `lsb` to be 110, that's how it was in out-of-tree driver, didn't try would it work with 120, I guess it would.

```
--- a/drivers/hwmon/it87.c	2020-08-26 14:43:49.315721428 +0200
+++ b/drivers/hwmon/it87.c	2020-08-27 17:16:49.184790191 +0200
@@ -63,7 +63,7 @@
 
 enum chips { it87, it8712, it8716, it8718, it8720, it8721, it8728, it8732,
 	     it8771, it8772, it8781, it8782, it8783, it8786, it8790,
-	     it8792, it8603, it8620, it8622, it8628 };
+	     it8792, it8603, it8613, it8620, it8622, it8628 };
 
 static unsigned short force_id;
 module_param(force_id, ushort, 0);
@@ -153,6 +153,7 @@
 #define IT8786E_DEVID 0x8786
 #define IT8790E_DEVID 0x8790
 #define IT8603E_DEVID 0x8603
+#define IT8613E_DEVID 0x8613
 #define IT8620E_DEVID 0x8620
 #define IT8622E_DEVID 0x8622
 #define IT8623E_DEVID 0x8623
@@ -433,6 +434,15 @@
 		  | FEAT_AVCC3 | FEAT_PWM_FREQ2,
 		.peci_mask = 0x07,
 	},
+	[it8613] = {
+		.name = "it8613",
+		.suffix = "E",
+		.features = FEAT_NEWER_AUTOPWM | FEAT_12MV_ADC | FEAT_16BIT_FANS
+		  | FEAT_TEMP_PECI | FEAT_FIVE_FANS
+		  | FEAT_FIVE_PWM | FEAT_IN7_INTERNAL | FEAT_PWM_FREQ2
+		  | FEAT_AVCC3,
+		.peci_mask = 0x07,
+	},
 	[it8620] = {
 		.name = "it8620",
 		.suffix = "E",
@@ -573,6 +583,8 @@
 		lsb = 109;
 	else
 		lsb = 160;
+	if (data->type == it8613)
+		lsb = 110;
 	if (data->in_scaled & BIT(nr))
 		lsb <<= 1;
 	return lsb;
@@ -2449,6 +2461,9 @@
 	case IT8790E_DEVID:
 		sio_data->type = it8790;
 		break;
+	case IT8613E_DEVID:
+		sio_data->type = it8613;
+		break;	
 	case IT8603E_DEVID:
 	case IT8623E_DEVID:
 		sio_data->type = it8603;
@@ -2610,6 +2625,43 @@
 
 		sio_data->beep_pin = superio_inb(sioaddr,
 						 IT87_SIO_BEEP_PIN_REG) & 0x3f;
+	} else if (sio_data->type == it8613) {
+		int reg27, reg29, reg2a;
+
+		superio_select(sioaddr, GPIO);
+
+		/* Check for pwm3, fan3, pwm5, fan5 */
+		reg27 = superio_inb(sioaddr, IT87_SIO_GPIO3_REG);
+		if (reg27 & BIT(1))
+			sio_data->skip_fan |= BIT(4);
+		if (reg27 & BIT(3))
+			sio_data->skip_pwm |= BIT(4);
+		if (reg27 & BIT(6))
+			sio_data->skip_pwm |= BIT(2);
+		if (reg27 & BIT(7))
+			sio_data->skip_fan |= BIT(2);
+
+		/* Check for pwm2, fan2 */
+		reg29 = superio_inb(sioaddr, IT87_SIO_GPIO5_REG);
+		if (reg29 & BIT(1))
+			sio_data->skip_pwm |= BIT(1);
+		if (reg29 & BIT(2))
+			sio_data->skip_fan |= BIT(1);
+
+		/* Check for pwm4, fan4 */
+		reg2a = superio_inb(sioaddr, IT87_SIO_PINX1_REG);
+		if (!(reg2a & BIT(0)) || (reg29 & BIT(7))) {
+			sio_data->skip_fan |= BIT(3);
+			sio_data->skip_pwm |= BIT(3);
+		}
+
+		sio_data->skip_pwm |= BIT(0); /* No pwm1 */
+		sio_data->skip_fan |= BIT(0); /* No fan1 */
+		sio_data->skip_in |= BIT(3);  /* No VIN3 */
+		sio_data->skip_in |= BIT(6);  /* No VIN6 */
+
+		sio_data->beep_pin = superio_inb(sioaddr,
+						 IT87_SIO_BEEP_PIN_REG) & 0x3f;					 
 	} else if (sio_data->type == it8620 || sio_data->type == it8628) {
 		int reg;
 
```

gcoimbra, 2022-11-09:
@TheMainMMan Hi, thanks a lot about your work. I've tried applying your patch above with the following error:

```
patching file it87.c
Hunk #1 FAILED at 63.
Hunk #2 FAILED at 153.
Hunk #3 succeeded at 647 with fuzz 1 (offset 214 lines).
Hunk #4 succeeded at 896 with fuzz 1 (offset 314 lines).
Hunk #5 succeeded at 3093 with fuzz 2 (offset 633 lines).
Hunk #6 FAILED at 2624.
3 out of 6 hunks FAILED -- saving rejects to file it87.c.rej
```

Would you be so kind to say which it87.c this patch applies to? Thanks!

richwalm, 2022-12-09:
@gcoimbra It applies to 'drivers/hwmon/it87.c'. I can confirm that it applies cleanly to kernel version 5.19.17. Do you have any other patches you're using?

HanM23, 2023-02-06:
Hi,
I am following this issue as I also have an IT8613E chip on my motherboard, and tonight, I was running pwmconfig in order to get and regulate my cpu and case fans...I am discovering I can't do it!
My motherboard is new, that's disappointed.

My current problem is that i do not understand one word (almost) about what you are talking about in order to sort this out, although i would like to help...

What could i do ?
Thank you for your help

gcoimbra, 2023-02-06:
Hi @HanM23 , we are discussing the patch @TheMainMMan made to it87 module to make it work for that kind of chip.
To make things easier for me, I made a repository https://github.com/gcoimbra/it87 with the patch included, it should work. You need to just to compile and load the module.  Like this, as root:
# cd it87/
# make
# insmod it87.ko

If your distro doesn't used signed modules, this alone should work.
But I must warn you, you need to verify this kind of code for malware, as it will be executing inside your kernel.

w-flo, 2023-02-09:
I also have a new mainboard (a cheap one, Biostar A320MH 2.0) using the IT8613E, and I've looked at its DSDT. It appears to use the same / a similar thermal management code (SENP, RTMP, CTMP etc.) as the one discussed above in early 2020. Basically, when the kernel wants to query the ACPI Thermal Zone temperature, it will execute that AML bytecode provided by the mainboard firmware because that's how ACPI works. And the AML bytecode will access the IT8613E registers without any kind of synchronization because the vendor (Biostar, or whoever sold them that code) didn't put any kind of synchronization in there.

As I understand it, this means we can't really use the it87 Linux driver. As I said, ACPI/AML will regulary access the IT8613E environment controller without any kind of synchronization when querying the ACPI Thermal Zone temperature. So the kernel shouldn't try to access these two registers, because it might try to access them at exactly the same time as the ACPI/AML code, leading to all kinds of issues.

For example, I used the "ignore acpi conflict option" and the well-known it87 fork supporting the IT8613E yesterday. I used it with the "Vitals" sensors monitoring gnome extension, which regularly queries the ACPI Thermal Zone temperature at roughly the same time as the it87 temperatures / fan data. I would regularly get incorrect readings, like a current temperature of 127°C – which is actually the critical temperature, so there was probably interleaved access to the IT8613E registers by the AML code and the it87 kernel driver, so the it87 driver read the data register only after the AML code had already changed the index register to read the critical temperature instead of the current temperature.

Things went downhill from there – after some time, The IT8613E environment controller was apparently convinced that temperature of one sensor had risen to 103°C (which is very hot), massively throttling my system. Rebooting into BIOS had fans spinning at max and the BIOS hardware monitor still claimed a constant 103°C. So I had to shut down and restart to fix this.

So if you have the "\SENP" OpRegion conflicting with SystemIO range 0xA35-0xA36 ACPI warning, it might be better to **take it seriously** because the "SENP" code actually accesses the registers in a way that appears to be incompatible with the it87 kernel driver.

I'm not sure if this can be solved in the kernel. Maybe there is a way to have the IT8613E driver lock some mutex in the ACPI subsystem before querying the environment controller to work around the issue. But generally speaking, I'd say it's a bug / design issue with the ACPI AML code provided by the mainboard vendor. @TheMainMMan worked around it by modifying the AML code, probably to remove the IT8613E environment controller access. But that probably means the ACPI Thermal Zone no longer works and is generally not something I want to do.

graysky2, 2023-12-23:
@gcoimbra - I was able to compile this on Arch Linux but unable to insert the module.  Any advice?
```
% make -j4
  CC [M]  /tmp/it87-main/it87.o
/tmp/it87-main/it87.c: In function ‘get_temp_type’:
/tmp/it87-main/it87.c:1481:28: warning: this statement may fall through [-Wimplicit-fallthrough=]
 1481 |                         if (index < 3)
      |                            ^
/tmp/it87-main/it87.c:1483:17: note: here
 1483 |                 case it8655:
      |                 ^~~~
  MODPOST /tmp/it87-main/Module.symvers
  CC [M]  /tmp/it87-main/it87.mod.o
  LD [M]  /tmp/it87-main/it87.ko
  BTF [M] /tmp/it87-main/it87.ko

% sudo insmod it87.ko
insmod: ERROR: could not insert module it87.ko: Unknown symbol in module
```

gcoimbra, 2023-12-28:
@graysky2 which kernel and gcc version are you using? I have not tried with the msot recent ones. but I how give a go now

graysky2, 2023-12-28:
@gcoimbra - `gcc (GCC) 13.2.1 20230801`

I ended up building just fine using a modified version of [this](https://aur.archlinux.org/packages/it87-dkms-git) package.  I took the suggestions of Hanabishi in the comments.

Now I am looking to create a lm_sensors conf for my specific hardware.  Just started searching.  Are you aware of any?
```
% sensors
coretemp-isa-0000
Adapter: ISA adapter
Package id 0:  +54.0°C  (high = +105.0°C, crit = +105.0°C)
Core 0:        +52.0°C  (high = +105.0°C, crit = +105.0°C)
Core 1:        +52.0°C  (high = +105.0°C, crit = +105.0°C)
Core 2:        +52.0°C  (high = +105.0°C, crit = +105.0°C)
Core 3:        +52.0°C  (high = +105.0°C, crit = +105.0°C)

acpitz-acpi-0
Adapter: ACPI interface
temp1:        +27.8°C  

it8613-isa-0a30
Adapter: ISA adapter
in0:         671.00 mV (min =  +0.00 V, max =  +2.81 V)
in1:           1.16 V  (min =  +0.00 V, max =  +2.81 V)
in2:           1.96 V  (min =  +0.00 V, max =  +2.81 V)
in4:           1.96 V  (min =  +0.00 V, max =  +2.81 V)
in5:           1.96 V  (min =  +0.00 V, max =  +2.81 V)
3VSB:          4.18 V  (min =  +0.00 V, max =  +5.61 V)
Vbat:          3.01 V  
+3.3V:         3.23 V  
fan2:        2947 RPM  (min =    0 RPM)
fan3:         748 RPM  (min =    0 RPM)
temp1:        +48.0°C  (low  = -128.0°C, high = +127.0°C)  sensor = thermistor
temp2:        +50.0°C  (low  = -128.0°C, high = +127.0°C)  sensor = thermistor
temp3:        +50.0°C  (low  = -128.0°C, high = +127.0°C)
intrusion0:  ALARM

nvme-pci-0300
Adapter: PCI adapter
Composite:    +50.9°C  (low  = -273.1°C, high = +81.8°C)
                       (crit = +84.8°C)
Sensor 1:     +50.9°C  (low  = -273.1°C, high = +65261.8°C)
Sensor 2:     +62.9°C  (low  = -273.1°C, high = +65261.8°C)
```

gcoimbra, 2024-01-29:
@graysky2 sorry for the lateness. use sensors-detect works·?