lm-sensors/lm-sensors #169: Issues with Intel i9 9900K

dmouritsen, 2019-03-12:
I got a i9-9900K running on a Gigabyte Z390-AORUS-PRO , which is misbehaving on a 5.0 kernel. 

Basically, it seems to be reporting CPU temps roughly 20% below the readings I get under Windows when Idle, versus under full load (simulated with the same prime generator on both platforms). Also, it's interesting to add that the readings I get under windows when idle, are consistent with my idle readings in the bios.

Also, I got an alert configured in the BIOS, to trigger when it reaches 90 degrees cels. However, during the Linux stress test, I go north of 93. Sometimes in peaks, even up to 98. But, the bios alert isn't triggering. 

So, long story short. I think something is iffy with the 9900K support temp. support.

justdanyul, 2019-03-17:
Update, I checked on another motherboard, and it was the same. However, Mprime actually fails, so, unfortunately, I think the sensor reading are accurate

nemoinis, 2019-05-09:
Hi Daniel,
Do you have a lm-sensors conf file you can share for the Gigabyte Z390-AORUS-PRO?
I'm setting up a system with the same motherboard and it'd be wonderful not to have to re-invent the wheel! cheers.