lm-sensors/lm-sensors #172: Fancontrol doesn't work after waking from sleep

Sporif, 2019-04-06:
On my system at least fancontrol (if it was running) doesn't work after waking up from sleep. This is because upon wakeup `/sys/class/hwmon/hwmon2/pwm*_enable` is not set to `1` anymore, but rather `5`.  Restarting fancontrol.service or setting `pwm*_enable` back to `1` are workarounds. I think the script should check the value of `pwm*_enable` every time it updates fan speeds and set it to 1 if it is not.

olysonek, 2019-04-08:
It makes sense, we can do that. However I wonder whether it should be considered a kernel bug.

groeck, 2019-04-25:
This is really a kernel bug; the kernel should store configuration and reapply it on resume. Unfortunately, most drivers don't implement that. While I would be happy to accept respective patches for kernel drivers, I consider it quite low priority: For safety reasons, systems should in general stick with automatic fan control, configured in the BIOS. This works even if the kernel misbehaves, and does not depend on active code in the system.

maxbla, 2019-07-25:
I had this same problem and was in the middle of writing an issue for it when I found this thread. Included is my attempt at debugging. But first, I'm a tad disappointed that the response is "its a kernel bug". If it is a kernel bug, shouldn't we report it? and if it isn't, is some investigation in order? (I am happy to do that investigation).
Furthermore, regardless of if it is a kernel bug or not, I think we can work around it with systemd. As a quick, hacky fix, I added `Restart=always` to my fancontrol.service, and it restarts after failing related to suspends (though that's not a great default behavior and I don't think it should be shipped that way). Also it seems to me like fancontrol could be a systemd timer service rather than a systemd simple service like [this](https://stackoverflow.com/questions/50219999/run-timer-exactly-every-second-with-systemd). Such a change has potential to fix this whole situation, as systemd is involved in the process of suspending, and might magically wait the right amount of time before letting fancontrol run.
# Original Issue Report
## OS / Environment
Arch Linux, kernel `5.2.0`
Systemd `systemd 242 (242.32-3-arch)`
Bash `5.0.7`

## Description of problem
fancontrol works, but crashes when the computer is suspended. The problem is reproducible by starting fancontrol (`systemctl start fancontrol.service`) then suspending my computer. After crashing, fans behave the way they are set in UEFI/BIOS, not the way they were set with fancontrol (determined by audio inspection). Checking the logs for fancontrol (`journalctl -u fancontrol.service`) I find this message
```
...
/usr/sbin/fancontrol: line 610: echo: write error: Invalid argument
Error writing PWM value to /sys/class/hwmon/hwmon1/pwm1
Aborting, restoring fans...
...
fancontrol.service: Main process exited, code=exited, status=1/FAILURE
fancontrol.service: Failed with result 'exit-code'.
```
(I enabled debug mode in fancontrol i.e. `DEBUG=1` on line 47 of fancontrol)

## What is actually causing the problem?
TL;DR there is an error writing to the file the kernel creates for controlling fan speed, but that file exists and bash reports it as writeable just before writing to it.
I guessed the problem had something to do with the kernel not making the file handles for the fans writable soon enough after resume from suspend. I tried running the `CheckFIles` function immediately before the line that crashed. I was expecting it to tell me something like the file handle is not writeable. Turns out `CheckFiles` doesn't even check for writeability, only readability I added some code to check for writeability and `CheckFiles` still succeeds in its checks. Then I suspended for long enough to tell the difference between timestamps before and after suspend, and found that systemd reports the errors happening after resume from suspend, not before suspend. But maybe the errors are happening before suspend and systemd isn't reporting them until after. I'm not really sure what is causing the issue. It could either be that the last write to fans before suspend is causes an error that isn't logged until after suspend or it could be that the first write to fans after suspend is broken.

further testing: create a systemd unit file that runs fancontrol right before suspend and one that tries right after suspend and see which if any crash

olysonek, 2019-07-25:
> But first, I'm a tad disappointed that the response is "its a kernel bug". If it is a kernel bug, shouldn't we report it?

@maxbla, groeck is the kernel maintainer of the hwmon subystem. Please read his comment again. I consider this a known issue.

> /usr/sbin/fancontrol: line 610: echo: write error: Invalid argument
> Error writing PWM value to /sys/class/hwmon/hwmon1/pwm1
...
> What is actually causing the problem?

The problem is that `/sys/class/hwmon/hwmon1/pwm1` accepts different values depending on the value of `/sys/class/hwmon/hwmon1/pwm1_enable`.

groeck, 2019-07-25:
Kind of hilarious that no one bothers to mention the hardware monitoring chip, which would enable us to at least identify the driver in question. It is also interesting that the user doesn't simply set fan control in the BIOS to manual. I mean, if you don't want the BIOS / Super IO chip to handle fan speed control, why not tell the BIOS to start with ? However, again, I strongly advise against using fancontrol in a system supporting automatic fan control in the Super IO chip.

DocSneider, 2020-02-12:
Sometimes the BIOS fan control is just not as good, as making your own rules with fancontrol in software. For example, my BIOS isn´t able to stop (single) fans when the temperature is below  a defined temperature.
In my case, when using the bios fan control, the fans are running at higher speeds than necessary and therefore they are louder than necessary. It´s a workstation I am talking about, so the noise of the fans is a important point.
Also because I am using a standalone water cooling, I am able to run the fans at lower minimum speeds than the bios fan control would me allow to, without getting to high temperatures.

maxbla, 2020-03-25:
> groeck is the kernel maintainer of the hwmon subystem

Whoops, didn't realize. He said he would accept patches, but I didn't know where they were being accepted to (could have been a patch set included in this repository for all I know). Good to know that the kernel people know.

> It is also interesting that the user doesn't simply set fan control in the BIOS to manual

There is no option called manual in my BIOS. I'm not sure if you mean set a fan curve in BIOS or somehow disable fan control by the Super IO chip. If you mean the later, I don't think that is possible.
I have a fan curve set in the BIOS (which I use for the CPU fan), but it has a few shortcomings. Primarily, it doesn't control the fans on my discrete graphics card, but also it doesn't have access to the temperature of that graphics card for controlling case fans (fancontrol, of course does). My computer's case is small, with two separate chambers for CPU and GPU, and fans for each chamber. It only makes sense to control the fan next to the GPU with the GPU temperature.

> Kind of hilarious that no one bothers to mention the hardware monitoring chip, which would enable us to at least identify the driver in question.

I have little idea what is relevant information. I was expecting you to ask for relevant information I missed instead of deriding me for not including it. Here is a command I ran and its output. Maybe it can help.

```
$ cat /sys/class/hwmon/hwmon1/name
k10temp

```
Also here is a link to my motherboard's manual, if that helps.
https://download.msi.com/archive/mnu_exe/mb/E7A40v2.2.pdf

 If this isn't what you were after, could you tell me how to get such information?

vedgy, 2020-07-24:
I have the same issue with my graphics card _MSI Radeon RX 570_. I use fancontrol because otherwise the graphics card's fans frequently alternate between two speeds and are loud enough to be annoying, even though the card's temperature is close to the room temperature and very far from 60 °C (the point below which the fans are supposed to be stopped according to [*TWIN FROZR VI*](https://www.msi.com/blog/secrets-twin-frozr-vi-revealed) description). With my fancontrol configuration the fans spin quietly and the card's temperature is normally below 40 °C.

```bash
$ cat /sys/class/hwmon/hwmon2/name
amdgpu
```

I've implemented a workaround for this issue in the form of a systemd service and a script: [try-restart-fancontrol-on-wake.zip](https://github.com/lm-sensors/lm-sensors/files/4972655/try-restart-fancontrol-on-wake.zip).

protist, 2020-08-16:
I can confirm this for my system too. For me, it's `/sys/class/hwmon/hwmon1/pwm2_enable` that changes from `1` to `5`.

Notably in my case, `fancontrol` never actually exits, but just fails silently. Hence, this issue isn't fixed for me by adding `Restart=always` to `fancontrol.service`, nor by using the script/service proposed [above](https://github.com/lm-sensors/lm-sensors/issues/172#issuecomment-663580357). Instead, I modified the proposed service to just always restart after a suspend as below, but this might not suit all users.

```
[Unit]
Description=Restart of fancontrol.service after suspend
After=hibernate.target suspend.target

[Service]
Type=simple
ExecStart=systemctl restart fancontrol.service

[Install]
WantedBy=hibernate.target suspend.target
```

paride, 2020-08-20:
Also tracked in Debian (https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=664141) and Ubuntu (https://bugs.launchpad.net/lmsensors/+bug/1882272).

I think that a decent workaround implemented at the downstream packaging level would be adding a script to `/usr/lib/systemd/system-sleep/` stopping the service pre-sleep and starting it post-sleep (see systemd-sleep(8)).

RonnySvedman, 2020-11-14:
Same on rx580. Fan goes to full power on resume. systemd service fancontrol restart solves problem. Without fancontrol on cold boot my gfx card runs fan on low rpm until it overheats (+80C or thereabout) then goes abruptly to full power (card bios emergency measure I suppose), and abruptly back down to really slow when it cools off again, rinse and repeat. Also, the pwmconfig gets fooled by the fact that the rx580 simply stops updating the fan speed when it is asked for 0 pwm, so pwmconfig never sees the rpm drop even though the fan is physically obedient-ish (by going to low rpm but not completely stopping). I had to edit /etc/fancontrol by hand to correct the associations. I speculate that the gpu bios won't allow a complete stop of the fan for safety reasons.

outusuke, 2020-11-18:
> I can confirm this for my system too. For me, it's `/sys/class/hwmon/hwmon1/pwm2_enable` that changes from `1` to `5`.
>
> Notably in my case, `fancontrol` never actually exits, but just fails [...]

where should i put this script?

outusuke, 2020-11-18:
> I can confirm this for my system too. For me, it's `/sys/class/hwmon/hwmon1/pwm2_enable` that changes from `1` to `5`.
>
> Notably in my case, `fancontrol` never actually exits, but just fails [...]

where do i put this script?

outusuke, 2020-11-18:
> [Unit]
> Description=Restart of fancontrol.service after suspend
> After=hibernate.target suspend.target
> 
> [Service]
> Type=simple
> ExecStart=systemctl restart fancontrol.service

it worked after sudo chmod a+x /lib/systemd/system/fancontrol-fix.service

outusuke, 2020-11-18:
I added this line in /lib/systemd/system/fancontrol.service and it worked better for me this script [above](https://github.com/lm-sensors/lm-sensors/issues/172#issuecomment-674509161) fails for me after suspending more than 2 times
 

[Service]
Restart=on-failure
RestartSec=5s

acjbrouwer, 2021-06-09:
Systemd supports adding/overriding settings in an override file. To do so enter
```
[sudo ]systemctl edit fancontrol
```
and in the editor enter the three lines mentioned in the previous comment. That way, service files, which are typically owned by some package, need not be modified.

Sha1rath, 2021-07-02:
> I added this line in /lib/systemd/system/fancontrol.service and it worked better for me this script [above](https://github.com/lm-sensors/lm-sensors/issues/172#issuecomment-674509161) fails for me after suspending more than 2 times
> 
> [Service]
> Restart=on-failure
> RestartSec=1s

This does not work for an obvious reason: Facontrol does not terminate or fail, it keeps running after suspend but just is not able to adjust fans anymore. So there is no "on-failure" event that triggers the restart.

FeralBytes, 2021-11-18:
The fix by @protist solved my problem by ensuring that the fancontrol daemon was restarted. I also use Corectrl to control the max frequency for my RX5700XT and the 2 together prevent the GPU from crashing automatically for me.

EnricoBara, 2022-01-28:
Sorry maybe I'm OT. I have the same problem, but in windows. Is there a way to do something like @protist proposed?

justdanyul, 2023-03-12:
> Kind of hilarious that no one bothers to mention the hardware monitoring chip, which would enable us to at least identify the driver in question. It is also interesting that the user doesn't simply [...]

Since you are the kernel maintainer, and you seem to be perplexed why people just use manual fan curves in the BIOS, I just thought I'd give you some examples of scenarios where it is troublesome to use the BIOS.

It's largely related to water-cooling.

- It's not unusual that motherboard manufacturers don't allow you change what temperature sensor you want to use in bios. ASUS have intentionally removed this feature from their BIOSes, and you can now only do manual fan curves based on CPU temperature. Under windows, you have various MB provides tools that will allow you to create curves based on other things, but on linux, we rely largely on fancontrol
- Some folks doing water-cooling want to base the fan curves for their radiator fans and their pump on water temperature as opposed to cpu temperature
- If you do work, in a water-cooled system, which predominately utilises the GPU, with very little CPU utilisation, (which is not uncommon in workflows relying on CUDA, say, ML and other types of scientific computing) , you can't rely on curves based on CPU temps alone.

Hope that helps to clarify why people would look outside of their BIOSes.

cut1less, 2023-07-13:
> I have the same issue with my graphics card _MSI Radeon RX 570_. I use fancontrol because otherwise the graphics card's fans frequently alternate between two speeds and are loud enough to be annoying

> Same on rx580. Fan goes to full power on resume. systemd service fancontrol restart solves problem. Without fancontrol on cold boot my gfx card runs fan on low rpm until it overheats (+80C or [...]

Same on my RX570. Curiously, though, this used to not happen on older kernels -- in fact, using an old LTS kernel eliminates the need for fancontrol. 5.10 is the newest that was working for me, and I was using it until I wanted features in the newer kernels.

groeck, 2023-07-13:
The Radeon graphics kernel driver implements its hardware monitoring functionality as part of its power management code. The code has been written without any input from a hardware monitoring maintainer, and I know nothing about it. The driver is maintained by AMD. I would suggest to raise any concerns/problems regarding any Radeon graphics cards with the amd-gfx@lists.freedesktop.org mailing list.

On a higher level, the various hardware monitoring attributes are always controlled by the driver. The actual implementation can and will be different for each driver. It is quite common, for example, that the underlying hardware restores to its default values after a suspend/resume cycle, and that the driver does not handle this properly. This can and should be addressed, but it needs to be addressed on a per driver basis. This is especially important for graphics card drivers, which are typically implemented and controlled by graphics card driver developers. Hardware monitoring kernel maintainers have no control and (if any) little influence into the development or maintenance of those drivers.

magist3r, 2023-07-13:
> Same on my RX570. Curiously, though, this used to not happen on older kernels -- in fact, using an old LTS kernel eliminates the need for fancontrol. 5.10 is the newest that was working for me, and [...]

I guess it's the same issue: https://gitlab.freedesktop.org/drm/amd/-/issues/1917

UbikNoir, 2023-12-16:
The same thing happens on Windows - is there any way to fix it there?

scraprats, 2024-06-05:
I am leaving some more information, fancontrol service fails after waking up from suspend.

> Jun 05 12:22:54 fedora fancontrol[7916]: /usr/sbin/fancontrol: line 638: echo: write error: Device or resource busy
> Jun 05 12:22:54 fedora fancontrol[7916]: Error writing PWM value to /sys/class/hwmon/hwmon2/pwm1
> Jun 05 12:22:54 fedora fancontrol[7916]: Aborting, restoring fans...
> Jun 05 12:22:54 fedora fancontrol[7916]: Verify fans have returned to full speed
> Jun 05 12:22:54 fedora systemd[1]: fancontrol.service: Main process exited, code=exited, status=1/FAILURE
> Jun 05 12:22:54 fedora systemd[1]: fancontrol.service: Failed with result 'exit-code'. 

The device hwmon2 is Nuvoton NCT6792D Super IO.

Forcing the service to [restart on failure](https://github.com/lm-sensors/lm-sensors/issues/172#issuecomment-729331853) solved the issue for me.