lm-sensors/lm-sensors #175: Asus Z170 PRO GAMING reports 0 fan speed

ArielRosenfeld, 2019-05-05:
Asus Z170 PRO GAMING reports 0 fan speed

Archlinux using kernel 5.0.10-arch1-1-ARCH

sensors-detects finds Nuvoton NCT6793D with the driver nct6775

driver is loaded and dmesg reports
[    8.612897] nct6775: Enabling hardware monitor logical device mappings.
[    8.612924] nct6775: Found NCT6793D or compatible chip at 0x2e:0x290

modinfo nct6775
filename:       /lib/modules/5.0.10-arch1-1-ARCH/kernel/drivers/hwmon/nct6775.ko.xz
license:        GPL
description:    Driver for NCT6775F and compatible chips
author:         Guenter Roeck <linux@roeck-us.net>
srcversion:     BA0D3026B895274B2377369
depends:        hwmon-vid
retpoline:      Y
intree:         Y
name:           nct6775
vermagic:       5.0.10-arch1-1-ARCH SMP preempt mod_unload 
sig_id:         PKCS#7
signer:         Build time autogenerated kernel key
sig_key:        74:8C:6D:92:BE:90:1A:3D:B3:09:71:89:AB:DE:2D:34:AB:93:3D:88
sig_hashalgo:   sha512
signature:      12:3C:23:FD:0E:DB:1D:A6:62:6D:0B:FA:7F:24:90:C8:FC:02:B3:60:
                03:95:B5:3C:59:04:C9:1E:55:D0:A4:48:C4:ED:85:F5:E2:4D:34:9B:
                DC:75:6B:4F:EF:AD:A7:6F:73:B8:E4:81:42:3C:E0:CF:2E:16:54:A8:
                A1:E6:A3:0F:DE:47:AE:69:6E:3F:6A:05:C5:F4:4F:E1:F7:A3:73:B1:
                E0:FD:E0:D3:E8:C4:7A:8B:C5:8D:32:11:9D:36:FB:1C:27:B8:ED:D6:
                CE:52:8E:79:2E:1C:9A:05:96:ED:C1:8E:18:A3:C4:FF:A9:4A:C9:02:
                19:BB:7E:88:5B:C5:2D:6F:54:AD:DE:56:D1:CC:A1:D6:46:A2:BE:12:
                16:47:12:F9:AF:EB:DB:96:9A:64:CC:52:84:26:25:1F:C7:55:A8:E5:
                AD:89:70:D5:EB:29:ED:73:48:23:A7:75:C6:FA:57:D6:91:BE:60:5A:
                5A:5A:C9:63:87:93:23:CB:3B:5D:30:5D:95:19:18:2E:30:37:89:11:
                1B:6A:69:CD:59:32:79:F1:3E:A1:C2:E3:5E:32:B9:82:B6:92:28:48:
                75:7B:BA:BC:79:7B:3B:FB:C3:A7:C0:C8:69:4A:38:7F:F7:B0:90:C4:
                09:46:4A:7C:6E:4D:BD:8A:1C:B1:48:AB:E9:55:BF:6D:A7:E3:DA:C1:
                33:1A:BF:76:2F:46:3F:48:0B:D0:F3:83:04:77:07:33:28:EB:44:7C:
                D2:C9:AE:18:C2:95:68:F8:23:81:E7:4C:1A:79:03:59:32:E8:F6:25:
                87:60:B6:2A:0C:F1:9F:BA:F8:3F:9D:62:D6:48:79:D6:C5:ED:B8:5C:
                88:CA:8A:D8:9F:DE:AF:A3:38:52:07:AD:8E:E9:E6:F0:80:63:C4:E2:
                69:87:20:BE:3B:E8:42:03:FA:80:35:1C:20:8A:CC:70:03:E1:F0:9B:
                41:D6:0D:3A:04:4D:AD:0C:69:C4:B5:0D:C1:E1:14:19:9C:FC:6A:5A:
                6F:54:E6:3B:A0:23:83:C6:15:78:1F:47:E9:9E:4B:C8:B1:8A:FF:05:
                0B:A5:2D:AC:6C:08:38:93:F5:31:87:30:1B:C1:61:49:A8:92:AA:44:
                9B:F9:49:55:59:6B:88:4C:BC:8B:A7:45:BE:9D:2D:BE:46:AA:E6:A9:
                58:AC:9B:4D:9E:B1:DD:01:7D:E3:D3:0E:13:34:12:34:CC:91:16:75:
                5B:EE:0F:ED:06:96:D4:8D:2C:B0:01:D4:A6:16:A1:52:69:47:87:03:
                6B:EC:14:E6:03:D7:50:3A:5C:6A:E7:0F:0A:5C:40:64:08:7C:B1:8A:
                AF:EA:89:DD:53:4D:E2:E5:D4:64:AD:E6
parm:           force_id:Override the detected device ID (ushort)
parm:           fan_debounce:Enable debouncing for fan RPM signal (ushort)

sensors

coretemp-isa-0000  
Adapter: ISA adapter 
Package id 0:  +34.0°C  (high = +84.0°C, crit = +100.0°C)                
Core 0:        +33.0°C  (high = +84.0°C, crit = +100.0°C)                
Core 1:        +34.0°C  (high = +84.0°C, crit = +100.0°C)                
Core 2:        +31.0°C  (high = +84.0°C, crit = +100.0°C)                
Core 3:        +32.0°C  (high = +84.0°C, crit = +100.0°C)                
                                    
acpitz-acpi-0        
Adapter: ACPI interface
temp1:        +27.8°C  (crit = +119.0°C)                                 
temp2:        +29.8°C  (crit = +119.0°C)                                 

asus-isa-0000
Adapter: ISA adapter
cpu_fan:        0 RPM

dmidecode > sensors.txt
[sensors.txt](https://github.com/lm-sensors/lm-sensors/files/3145482/sensors.txt)

louitrilobite, 2019-05-07:
Hello,

Asus TUF z270
Ubuntu 19.04 using kernel 5.1.0-050100-generic

I was getting pretty much the same problem, sensors was reading nothing from nct6775 Editing /etc/default/grub and adding... 'acpi_enforce_resources=lax' on the end of 'RUB_CMDLINE_LINUX_DEFAULT, so it looks like this (for me anyway)

GRUB_CMDLINE_LINUX_DEFAULT="quiet splash acpi_enforce_resources=lax"

solves the issue.

Please note though during my search for and answer to this issue I found that adding the above could cause system instability, so there is that. seems to be working for me though.

ArielRosenfeld, 2019-05-09:
Thank you louitrilobite,  seems to be the correct solution for the problem on the lm-sensors side, ACPI still not reporting correctly so i guess its a kernel driver bug

closed 2019-05-09 by ArielRosenfeld