lm-sensors/lm-sensors #176: Support for RK3399

dougcooper, 2019-05-05:
Being playing with the RockPro64. Running lm-sensors using the vendor recommended kernel effectively locks the system up. Is this on the roadmap? Is it possible to help contribute upstream on an effort like this?

olysonek, 2019-07-15:
What do you mean by running lm_sensors? Do you run `sensors` or `sensors-detect`? Which causes the problem? If `sensors`, then this is probably a kernel issue. If it's `sensors-detect`, I'm afraid it can cause problems like that - it even warns about it.

dougcooper, 2019-07-15:
I believe ‘sensors-detect’. I guess my concern is that I cannot see an obvious way for the available implementation to correctly identify the hardware for monitoring without locking up the system. If there is a way it would be beneficial to know and document it.

olysonek, 2019-07-15:
`sensors-detect` needs to do some probing that may be unsafe on some systems. This should however only be those probes that `sensors-detect` explicitly asks for permission to do. If you run `sensors-detect --auto`, or run `sensors-detect` and accept the default answers, it should be safe. If you however explicitly tell `sensors-detect` to do some of the probes that it says may be unsafe, problems may occur. I don't think there's a way to detect if a certain probe operation is safe or not in general.

If some of the probe operations made by `sensors-detect --auto` results in the lockup, please let me know and I'll try to fix it. If some of the probes marked as potentially unsafe is the cause, I'm afraid there's not much I can do.

olysonek, 2019-07-16:
Thinking some more about this, whichever probe operation causes the lockup, I could start a list of hardware where a given probe operation causes problems. I would need you to identify the hardware you're running (mainboard model number/laptop model number/...) and identify the exact probe that causes the problem.

olysonek, 2019-08-01:
> ... and identify the exact probe that causes the problem.

I'd still need an answer to this. Can you provide the output of sensors-detect up till the point where the system locks up?

dougcooper, 2019-08-03:
```bash
root@rockpro64:/# sensors-detect
# sensors-detect revision 6284 (2015-05-31 14:00:33 +0200)

Message from syslogd@rockpro64 at Aug  3 08:22:40 ...
 kernel:[2463954.645620] Internal error: Oops - bad mode: 0 [#1] SMP

Message from syslogd@rockpro64 at Aug  3 08:22:40 ...
 kernel:[2463954.694902] Process dmidecode (pid: 6505, stack limit = 0xffffffc0a7368000)

Message from syslogd@rockpro64 at Aug  3 08:22:40 ...
 kernel:[2463954.707107] Internal error: Oops - bad mode: 0 [#2] SMP

Message from syslogd@rockpro64 at Aug  3 08:22:40 ...
 kernel:[2463954.758031] Process dmidecode (pid: 6507, stack limit = 0xffffffc0eb548000)

Message from syslogd@rockpro64 at Aug  3 08:22:40 ...
 kernel:[2463954.767002] Internal error: Oops - bad mode: 0 [#3] SMP

Message from syslogd@rockpro64 at Aug  3 08:22:40 ...
 kernel:[2463954.817807] Process dmidecode (pid: 6509, stack limit = 0xffffffc0eb54c000)

Message from syslogd@rockpro64 at Aug  3 08:22:40 ...
 kernel:[2463954.826922] Internal error: Oops - bad mode: 0 [#4] SMP

Message from syslogd@rockpro64 at Aug  3 08:22:40 ...
 kernel:[2463954.877727] Process dmidecode (pid: 6511, stack limit = 0xffffffc02ea80000)

Message from syslogd@rockpro64 at Aug  3 08:22:40 ...
 kernel:[2463954.886988] Internal error: Oops - bad mode: 0 [#5] SMP

Message from syslogd@rockpro64 at Aug  3 08:22:40 ...
 kernel:[2463954.937721] Process dmidecode (pid: 6513, stack limit = 0xffffffc0a73e0000)

Message from syslogd@rockpro64 at Aug  3 08:22:40 ...
 kernel:[2463954.946816] Internal error: Oops - bad mode: 0 [#6] SMP

Message from syslogd@rockpro64 at Aug  3 08:22:40 ...
 kernel:[2463954.997477] Process dmidecode (pid: 6515, stack limit = 0xffffffc0e7dec000)

Message from syslogd@rockpro64 at Aug  3 08:22:40 ...
 kernel:[2463955.006419] Internal error: Oops - bad mode: 0 [#7] SMP

Message from syslogd@rockpro64 at Aug  3 08:22:40 ...
 kernel:[2463955.057245] Process dmidecode (pid: 6517, stack limit = 0xffffffc0a7368000)
# Kernel: 4.4.154-1124-rockchip-ayufan-ged3ce4d15ec1 aarch64
# Processor:  (//)

This program will help you determine which kernel modules you need
to load to use lm_sensors most effectively. It is generally safe
and recommended to accept the default answers to all questions,
unless you know what you're doing.

Some south bridges, CPUs or memory controllers contain embedded sensors.
Do you want to scan for them? This is totally safe. (YES/no): 
modprobe: FATAL: Module cpuid not found in directory /lib/modules/4.4.154-1124-rockchip-ayufan-ged3ce4d15ec1
Failed to load module cpuid.
Silicon Integrated Systems SIS5595...                       No
VIA VT82C686 Integrated Sensors...                          No
VIA VT8231 Integrated Sensors...                            No
AMD K8 thermal sensors...                                   No
AMD Family 10h thermal sensors...                           No
AMD Family 11h thermal sensors...                           No
AMD Family 12h and 14h thermal sensors...                   No
AMD Family 15h thermal sensors...                           No
AMD Family 16h thermal sensors...                           No
AMD Family 15h power sensors...                             No
AMD Family 16h power sensors...                             No
Intel digital thermal sensor...                             No
Intel AMB FB-DIMM thermal sensor...                         No
Intel 5500/5520/X58 thermal sensor...                       No
VIA C7 thermal sensor...                                    No
VIA Nano thermal sensor...                                  No

Lastly, we can probe the I2C/SMBus adapters for connected hardware
monitoring devices. This is the most risky part, and while it works
reasonably well on most systems, it has been reported to cause trouble
on some systems.
Do you want to probe the I2C/SMBus adapters now? (YES/no): 
Sorry, no supported PCI bus adapters found.

Next adapter: rk3x-i2c (i2c-0)
Do you want to scan it? (YES/no/selectively): 
Client found at address 0x1b
Handled by driver `rk808' (built-in), chip type `rk808'
    (note: this is probably NOT a sensor chip!)

Next adapter: rk3x-i2c (i2c-1)
Do you want to scan it? (YES/no/selectively): 

Next adapter: rk3x-i2c (i2c-4)
Do you want to scan it? (YES/no/selectively): 

Next adapter: DesignWare HDMI (i2c-9)
Do you want to scan it? (YES/no/selectively): 

Next adapter: DP-AUX (i2c-10)
Do you want to scan it? (YES/no/selectively): 
```

At this point the system becomes unresponsive and needs to be hard reset.

dlr-3, 2019-08-11:
I can confirm that I get the same issue on my rockpro64.

`[root@rockpro64 etc]# sensors-detect
# sensors-detect revision 3.4.0-6 (2016-06-01)
sh: line 1: 28363 Segmentation fault      dmidecode -s chassis-type 2> /dev/null

Message from syslogd@rockpro64 at Aug 11 14:01:19 ...
 kernel:Internal error: Oops - bad mode: 0 [#16] SMP

Message from syslogd@rockpro64 at Aug 11 14:01:19 ...
 kernel:Process dmidecode (pid: 28363, stack limit = 0xffffffc0e79e4000)

Message from syslogd@rockpro64 at Aug 11 14:01:19 ...
 kernel:Internal error: Oops - bad mode: 0 [#17] SMP
sh: line 1: 28365 Segmentation fault      dmidecode -s baseboard-manufacturer 2> /dev/null

Message from syslogd@rockpro64 at Aug 11 14:01:19 ...
 kernel:Process dmidecode (pid: 28365, stack limit = 0xffffffc0eff44000)
sh: line 1: 28368 Bus error               dmidecode -s system-manufacturer 2> /dev/null
sh: line 1: 28370 Segmentation fault      dmidecode -s baseboard-product-name 2> /dev/null

Message from syslogd@rockpro64 at Aug 11 14:01:19 ...
 kernel:Internal error: Oops - bad mode: 0 [#18] SMP

Message from syslogd@rockpro64 at Aug 11 14:01:19 ...
 kernel:Process dmidecode (pid: 28370, stack limit = 0xffffffc0efad4000)

Message from syslogd@rockpro64 at Aug 11 14:01:19 ...
 kernel:Internal error: Oops - bad mode: 0 [#19] SMP
sh: line 1: 28372 Segmentation fault      dmidecode -s baseboard-product-name 2> /dev/null

Message from syslogd@rockpro64 at Aug 11 14:01:19 ...
 kernel:Process dmidecode (pid: 28372, stack limit = 0xffffffc0e09a0000)

Message from syslogd@rockpro64 at Aug 11 14:01:20 ...
 kernel:Internal error: Oops - bad mode: 0 [#20] SMP
sh: line 1: 28374 Segmentation fault      dmidecode -s system-version 2> /dev/null

Message from syslogd@rockpro64 at Aug 11 14:01:20 ...
 kernel:Process dmidecode (pid: 28374, stack limit = 0xffffffc0f0eb0000)

Message from syslogd@rockpro64 at Aug 11 14:01:20 ...
 kernel:Internal error: Oops - bad mode: 0 [#21] SMP
sh: line 1: 28376 Segmentation fault      dmidecode -s system-product-name 2> /dev/null
# Kernel: 4.4.166-rockchip64 aarch64
**********
Your processor's architecture is not yet supported. Beware that sensors-detect might not work properly!
**********

This program will help you determine which kernel modules you need
to load to use lm_sensors most effectively. It is generally safe
and recommended to accept the default answers to all questions,
unless you know what you're doing.

Some south bridges, CPUs or memory controllers contain embedded sensors.
Do you want to scan for them? This is totally safe. (YES/no):
Message from syslogd@rockpro64 at Aug 11 14:01:20 ...
 kernel:Process dmidecode (pid: 28376, stack limit = 0xffffffc0efad4000)`

therevoman, 2020-06-13:
I ran into a similar error with my RockPro64.  However, I'm running a CentOS 7 image.