lm-sensors/lm-sensors #185: Any plans to make new release?

kloczek, 2019-06-30:
It is already quite long time since latest release.
I think that it would be good to make new release to publish new code base :)

PS. If may I ask this time to use version tag without underscores because it makes automatic packaging difficult.

olysonek, 2019-07-01:
I'm a bit short on time these days, however I plan to make a new release soon.

> PS. If may I ask this time to use version tag without underscores because it makes automatic packaging difficult.

Yeah, it's a pain, but all the version tags so far have dashes in them and I'm afraid that by changing the tag format I'd bring more work to people who have packaging already set up. I'll consider it though.

laichiaheng, 2019-08-07:
For the Ryzen3000 users, a new release is necessary.

Nekotje, 2019-08-07:
it would be great to get it to work on Ryzen 3000.
Trying family `VIA/Winbond/Nuvoton/Fintek'...               Yes
Found unknown chip with ID 0xd42a
    (logical device B has address 0x290, could be sensors)
maybe this is it i dont know.

olysonek, 2019-08-08:
> For the Ryzen3000 users, a new release is necessary.

Well, it's not necessary to get output from the the CPU sensors. It's only necessary for sensors-detect to detect the sensors, and that is really not that important, as the k10temp driver is autoloaded. The problem is that there has not yet been a kernel release that supports the new CPUs in the driver.

Anyway, a new release is still on my short-term TODO.

olysonek, 2019-10-17:
Tomorrow I'll create a 3.6.0 release from the current master branch.

olysonek, 2019-10-18:
It's done.