lm-sensors/lm-sensors #186: ITE 8625E Support

andrebadaia, 2019-07-03:
Hi! I have a Ryzen 7 1700 system, with a Biostar X370 GT5 motherboard that uses the ITE 8625E super IO chip. Do you have any plans to support it? Thanks!

Myrddin-Wyllt, 2019-08-23:
<pre>Found `ITE IT8625E Super IO Sensors'                        Success!
    (address 0xa20, driver `to-be-written')
</pre>

According to the output, it would seem that there are plans for support. I've a Biostar X370GT7 & am eager to learn how to help write this driver. I'm also available for testing; I hope a developer here takes me up on this offer.

RonnySvedman, 2020-11-14:
ITE superio chips are a thorn in the butt of linux users since 15 years back. So much so that the it87 module maintainer quit in 2018, and now the sources are forked in several places, and the in-kernel it87 module is very old. try compiling [https://github.com/a1wong/it87](https://github.com/a1wong/it87), and modprobe it87. It worked for me yesterday on my asus z97p (old IT8606 chip, so YMMV).

We as a community should nag ITE and/or ASUS for specs so we could do proper drivers, but folklore says it's been "nope" so far.

andrebadaia, 2020-11-14:
> ITE superio chips are a thorn in the butt of linux users since 15 years back. So much so that the it87 module maintainer quit in 2018, and now the sources are forked in several places, and the in-kernel it87 module is very old. try compiling https://github.com/a1wong/it87, and modprobe it87. It worked for me yesterday on my asus z97p (old IT8606 chip, so YMMV).
> 
> We as a community should nag ITE and/or motherboard vendors for specs so we could do proper drivers, but folklore says it's been "nope" so far.

Ok, I'll try!!! Thanks a lot!