lm-sensors/lm-sensors #187: Can't display frequency and others of Ryzen7 3700X.

laichiaheng, 2019-07-21:
I can read the frequency by command `grep "cpu MHz" /proc/cpuinfo`, why can't lm-sensors read it?
None of the temperatures and fan speed works, either.

CPU: Ryzen7 3700X
RAM: Micron LT 3200MHz 16G*2
MB: ASUS X470 Crosshair VII Hero
Kernel: 5.2.1-1-MANJARO
Distro: Manjaro

olysonek, 2019-07-22:
> I can read the frequency by command `grep "cpu MHz" /proc/cpuinfo`, why can't lm-sensors read it?

I assume you mean sensors-detect. Why would it read it? sensors-detect only prints information necessary to identify the CPU. The current CPU frequency does not fall in that category.

>  None of the temperatures and fan speed works, either.

You're going to have to be more specific there. Anyway, have you run sensors-detect to completion?

dllu, 2019-07-22:
`sensors-detect` is unable to read temperatures of my Ryzen 9 3900x either.

```
# Processor: AMD Ryzen 9 3900X 12-Core Processor (23/113/0)

This program will help you determine which kernel modules you need
to load to use lm_sensors most effectively. It is generally safe
and recommended to accept the default answers to all questions,
unless you know what you're doing.

Some south bridges, CPUs or memory controllers contain embedded sensors.
Do you want to scan for them? This is totally safe. (YES/no): 
Module cpuid loaded successfully.
Silicon Integrated Systems SIS5595...                       No
VIA VT82C686 Integrated Sensors...                          No
VIA VT8231 Integrated Sensors...                            No
AMD K8 thermal sensors...                                   No
AMD Family 10h thermal sensors...                           No
AMD Family 11h thermal sensors...                           No
AMD Family 12h and 14h thermal sensors...                   No
AMD Family 15h thermal sensors...                           No
AMD Family 16h thermal sensors...                           No
AMD Family 17h thermal sensors...                           No
AMD Family 15h power sensors...                             No
AMD Family 16h power sensors...                             No
Hygon Family 18h thermal sensors...                         No
Intel digital thermal sensor...                             No
Intel AMB FB-DIMM thermal sensor...                         No
Intel 5500/5520/X58 thermal sensor...                       No
VIA C7 thermal sensor...                                    No
VIA Nano thermal sensor...                                  No
```

This is very odd to me because the `sensors-detect` I am running is bleeding edge and definitely supports `AMD Family 17h thermal sensors` with `amd_pci_detect(..., '1493')` which corresponds to `PCI_DEVICE_ID_AMD_17H_M30H_DF_F3`, which should supposedly include my processor. AFAIK family 17h m30h has been sitting in the kernel since November 2018 and I am running Arch Linux with `linux-amd-staging-drm-next-git 5.3.839842.40cc64619a25-1`.

`sensors-detect` did detect my gpu and it also detected  `Nuvoton NCT6796D Super IO Sensors` and loaded `nct6775`. After running `sensors-detect` and pressing enter for all the prompts, and then running `sensors`, here's what I get:

```
dllu » sensors
asus-isa-0000
Adapter: ISA adapter
cpu_fan:        0 RPM

amdgpu-pci-0b00
Adapter: PCI adapter
vddgfx:       +0.72 V  
fan1:             N/A  (min =    0 RPM, max = 4950 RPM)
edge:         +38.0°C  (crit = +80.0°C, hyst =  +0.0°C)
                       (emerg = +80.0°C)
junction:     +38.0°C  (crit = +80.0°C, hyst =  +0.0°C)
                       (emerg = +80.0°C)
mem:          +38.0°C  (crit = +80.0°C, hyst =  +0.0°C)
                       (emerg = +80.0°C)
power1:        9.00 W  (cap = 180.00 W)

```

Here's my motherboard for reference:

```
sudo dmidecode -t 2
# dmidecode 3.2
Getting SMBIOS data from sysfs.
SMBIOS 3.2.0 present.

Handle 0x0002, DMI type 2, 15 bytes
Base Board Information
	Manufacturer: ASUSTeK COMPUTER INC.
	Product Name: Pro WS X570-ACE
	Version: Rev X.0x
	Serial Number: (redacted)
	Asset Tag: Default string
	Features:
		Board is a hosting board
		Board is replaceable
	Location In Chassis: Default string
	Chassis Handle: 0x0003
	Type: Motherboard
	Contained Object Handles: 0
```

I will continue to do some digging to find out what's going on...

Monitoring the temperature of my CPU is important to me since I'm struggling with some [mysterious problems](https://community.amd.com/thread/241543).

laichiaheng, 2019-07-22:
@dllu https://github.com/electrified/asus-wmi-sensors

olysonek, 2019-07-22:
From what I understand, Ryzen 3700X is family 17h, model 71h. Support for that is currently being added to the kernel:
https://www.spinics.net/lists/linux-hwmon/msg06154.html

Regarding Ryzen 9 3900x, given that `sensors` doesn't show temperatures from the k10temp driver, it would seem that CPU is not supported - sensors-detect will not help here; the driver should autoload. Where do you get the information that it's model 30h? Does a device with PCI ID 1493 exist on your system? Perhaps this CPU is also model 71h. Does a device with PCI ID 0x1443 [1] exist on your system?
[1] https://www.spinics.net/lists/linux-hwmon/msg06153.html

dllu, 2019-07-23:
Oh I think you may be right. I got too excited after reading [phoronix on zen2 temperature monitoring](https://www.phoronix.com/scan.php?page=news_item&px=AMD-Zen-2-Linux-Temp-Driver) but I forgot that there are different types of zen 2 processors and that mine is probably not model 30h.

olysonek, 2019-07-23:
I've updated sensors-detect to detect family 17h model 70h. Can you try it out?
https://github.com/lm-sensors/lm-sensors/blob/00e7d37b35f6264fc3c121db47fbcc1e90010d89/prog/detect/sensors-detect

I guess kernel support will land in the 5.4 release.

closed 2019-07-23 by olysonek

dllu, 2019-07-23:
Cool, the Ryzen 3900x is detected now:

```
# Processor: AMD Ryzen 9 3900X 12-Core Processor (23/113/0)

Running in automatic mode, default answers to all questions
are assumed.

Some south bridges, CPUs or memory controllers contain embedded sensors.
Do you want to scan for them? This is totally safe. (YES/no): 
Module cpuid loaded successfully.
Silicon Integrated Systems SIS5595...                       No
VIA VT82C686 Integrated Sensors...                          No
VIA VT8231 Integrated Sensors...                            No
AMD K8 thermal sensors...                                   No
AMD Family 10h thermal sensors...                           No
AMD Family 11h thermal sensors...                           No
AMD Family 12h and 14h thermal sensors...                   No
AMD Family 15h thermal sensors...                           No
AMD Family 16h thermal sensors...                           No
AMD Family 17h thermal sensors...                           Success!
    (driver `k10temp')
AMD Family 15h power sensors...                             No
AMD Family 16h power sensors...                             No
Hygon Family 18h thermal sensors...                         No
Intel digital thermal sensor...                             No
Intel AMB FB-DIMM thermal sensor...                         No
Intel 5500/5520/X58 thermal sensor...                       No
VIA C7 thermal sensor...                                    No
VIA Nano thermal sensor...                                  No
```

Thanks!!!!!!!

Tintest, 2019-08-09:
Hello, I'm stuck with you guys :)

I'm not familiar with Linux kernel releases. Do you have any approximate idea about how much time we will have to wait ? 

Thanks

laichiaheng, 2019-08-10:
@Tintest Some guys said linux5.4 will fully support the Ryzen3000 CPU, but nobody knows.

Redsandro, 2019-11-18:
Same.

CPU: Ryzen7 3700X
MB: Aorus X570 I Pro Wifi

I guess we need to wait for Ubuntu 20.04 LTS and derivatives for clean Linux 5.4 support.

SCdF, 2019-11-21:
Clean 5.4 support will probably require waiting for 20.04, but for not so clean 5.4 support, 5.4 releases in general on Sunday (2019-11-24), and I presume [ubuntu's generic kernel builds](https://kernel.ubuntu.com/~kernel-ppa/mainline/) will follow soon after that. I've generally not had luck with them though, and will be happy to wait for 20.04

sanosay, 2019-11-26:
I just installed kernel 5.4 and compiled from source 3.6 on Ubuntu 18.04, yet while the CPU "AMD Family 17h thermal sensors" is showing as "SUCCESS", the temps are still not showing :(

pgnd, 2019-12-24:
@olysonek 

Is this correct?

> I've updated sensors-detect to detect family 17h model 70h. Can you try it out?

Isn't Ryzen7 3700X a Zen2 arch, and family 23h?

ref:
  https://en.wikipedia.org/wiki/List_of_AMD_CPU_microarchitectures
  https://en.wikipedia.org/wiki/Zen_2

Dunno it that might be a/the reason I'm seeing limited sensors info, or if 17h is an umbrella that's to include the Zen* families?

Or are we still waiting on kernel support? At https://cdn.kernel.org/pub/linux/kernel/v5.x/ChangeLog-5.4, I _do_ see some work by Marcel Bocu referencing the 3700X directly -- not clear to me what the current state of affairs re: this issue is.

I'm running
```
Asus X570-Pro mobo + Ryzen7 3700X cpu
Linux kernel 5.4.6-24.ge5f8301-default
sensors-3.6.0-131.1.x86_64
```

exec of `sensors-detect` generates

<details><summary>sensors-detect</summary>

```
# sensors-detect version 3.6.0
# Board: ASUSTeK COMPUTER INC. PRIME X570-PRO
# Kernel: 5.4.6-24.ge5f8301-default x86_64
# Processor: AMD Ryzen 7 3700X 8-Core Processor (23/113/0)

This program will help you determine which kernel modules you need
to load to use lm_sensors most effectively. It is generally safe
and recommended to accept the default answers to all questions,
unless you know what you're doing.

Some south bridges, CPUs or memory controllers contain embedded sensors.
Do you want to scan for them? This is totally safe. (YES/no): 
Module cpuid loaded successfully.
Silicon Integrated Systems SIS5595...                       No
VIA VT82C686 Integrated Sensors...                          No
VIA VT8231 Integrated Sensors...                            No
AMD K8 thermal sensors...                                   No
AMD Family 10h thermal sensors...                           No
AMD Family 11h thermal sensors...                           No
AMD Family 12h and 14h thermal sensors...                   No
AMD Family 15h thermal sensors...                           No
AMD Family 16h thermal sensors...                           No
AMD Family 17h thermal sensors...                           Success!
    (driver `k10temp')
AMD Family 15h power sensors...                             No
AMD Family 16h power sensors...                             No
Hygon Family 18h thermal sensors...                         No
Intel digital thermal sensor...                             No
Intel AMB FB-DIMM thermal sensor...                         No
Intel 5500/5520/X58 thermal sensor...                       No
VIA C7 thermal sensor...                                    No
VIA Nano thermal sensor...                                  No

Some Super I/O chips contain embedded sensors. We have to write to
standard I/O ports to probe them. This is usually safe.
Do you want to scan for Super I/O sensors? (YES/no): YES
Probing for Super-I/O at 0x2e/0x2f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               Yes
Found `Nuvoton NCT6798D Super IO Sensors'                   Success!
    (address 0x290, driver `nct6775')
[... 72 lines cut ...]
	# Generated by sensors-detect on Mon Dec 23 18:01:00 2019
	## Path: Hardware/Sensors
	## Description:         Defines the modules to used
	## Type:                string
	## ServiceRestart:      lm_sensors
	## Default:             ""
	#
	# This file is sourced by /etc/init.d/lm_sensors and defines the modules to
	# be loaded/unloaded.
	#
	# The format of this file is a shell script that simply defines variables:
	# HWMON_MODULES for hardware monitoring driver modules, and optionally
	# BUS_MODULES for any required bus driver module (for example for I2C or SPI).

	HWMON_MODULES="k10temp nct6775"
```

and `sensors` returns only the CPU temp
```
sensors
	k10temp-pci-00c3
	Adapter: PCI adapter
	Tdie:         +40.8°C  (high = +70.0°C)
	Tctl:         +40.8°C
```

olysonek, 2020-02-10:
> Isn't Ryzen7 3700X a Zen2 arch, and family 23h?
> 
> ref:
> https://en.wikipedia.org/wiki/List_of_AMD_CPU_microarchitectures
> https://en.wikipedia.org/wiki/Zen_2

@pgnd It seems you're confusing '23' with '23h' and '17' with '17h'. There's no such thing as AMD family 23h AFAIK.

> and `sensors` returns only the CPU temp

That's correct - from what I've seen, that's all the `k10temp` driver exports.

However according to `sensors-detect`, there is a Nuvoton Super IO chip on your board, supported by the `nct6775` kernel module. Do you have that module loaded? If yes, `sensors` should output readings from that.

pgnd, 2020-02-10:
@olysonek 

> you're confusing '23' with '23h' and '17' with '17h'

ugh. yup.

> there is a Nuvoton Super IO chip on your board, supported by the nct6775 kernel module. Do you have that module loaded?

yes
```
lsmod | grep 6775
  nct6775                73728  0
  hwmon_vid              16384  1 nct6775
```

> If yes, sensors should output readings from that.

fyi,

not sufficient, atm.  requires also
```
acpi_enforce_resources=lax 
```

cref:

Bug 204807 - Hardware monitoring sensor nct6798d doesn't work unless acpi_enforce_resources=lax is enabled 
  https://bugzilla.kernel.org/show_bug.cgi?id=204807

with that,
```
sensors
k10temp-pci-00c3
Adapter: PCI adapter
Tdie:         +46.0°C  (high = +70.0°C)
Tctl:         +46.0°C

nct6798-isa-0290
Adapter: ISA adapter
in0:                   952.00 mV (min =  +0.00 V, max =  +1.74 V)
in1:                     1.02 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in2:                     3.34 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in3:                     3.34 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in4:                   1000.00 mV (min =  +0.00 V, max =  +0.00 V)  ALARM
in5:                   792.00 mV (min =  +0.00 V, max =  +0.00 V)  ALARM
in6:                     1.02 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in7:                     3.34 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in8:                     3.26 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in9:                   896.00 mV (min =  +0.00 V, max =  +0.00 V)  ALARM
in10:                  392.00 mV (min =  +0.00 V, max =  +0.00 V)  ALARM
in11:                  544.00 mV (min =  +0.00 V, max =  +0.00 V)  ALARM
in12:                    1.02 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in13:                  1000.00 mV (min =  +0.00 V, max =  +0.00 V)  ALARM
in14:                  1000.00 mV (min =  +0.00 V, max =  +0.00 V)  ALARM
fan1:                  1105 RPM  (min =    0 RPM)
fan2:                  1362 RPM  (min =    0 RPM)
fan3:                   727 RPM  (min =    0 RPM)
fan4:                  1080 RPM  (min =    0 RPM)
fan5:                     0 RPM  (min =    0 RPM)
fan6:                     0 RPM  (min =    0 RPM)
fan7:                     0 RPM  (min =    0 RPM)
SYSTIN:                 +37.0°C  (high = +80.0°C, hyst = +75.0°C)  sensor = thermistor
CPUTIN:                 +30.0°C  (high = +80.0°C, hyst = +75.0°C)  sensor = thermistor
AUXTIN0:                +25.0°C    sensor = thermistor
AUXTIN1:                +37.0°C    sensor = thermistor
AUXTIN2:                +21.0°C    sensor = thermistor
AUXTIN3:                +26.0°C    sensor = thermistor
PCH_CHIP_CPU_MAX_TEMP:   +0.0°C
PCH_CHIP_TEMP:           +0.0°C
PCH_CPU_TEMP:            +0.0°C
PCH_MCH_TEMP:            +0.0°C
intrusion0:            ALARM
intrusion1:            ALARM
beep_enable:           disabled
```

looks like 'k10temp' fixes for Ryzen are pending for kernel 5.6,

https://www.phoronix.com/scan.php?page=news_item&px=K10temp-Could-Improve-5.6

avemilia, 2020-11-02:
I have added the boot parameter and the module, with
```
❯ sudo sensors-detect --auto
# sensors-detect version 3.6.0
# System: Micro-Star International Co., Ltd. MS-7C37 [3.0]
# Board: Micro-Star International Co., Ltd. X570-A PRO (MS-7C37)
# Kernel: 5.8.16-2-MANJARO x86_64
# Processor: AMD Ryzen 5 3600 6-Core Processor (23/113/0)

Running in automatic mode, default answers to all questions
are assumed.

Some south bridges, CPUs or memory controllers contain embedded sensors.
Do you want to scan for them? This is totally safe. (YES/no): 
Module cpuid loaded successfully.
Silicon Integrated Systems SIS5595...                       No
VIA VT82C686 Integrated Sensors...                          No
VIA VT8231 Integrated Sensors...                            No
AMD K8 thermal sensors...                                   No
AMD Family 10h thermal sensors...                           No
AMD Family 11h thermal sensors...                           No
AMD Family 12h and 14h thermal sensors...                   No
AMD Family 15h thermal sensors...                           No
AMD Family 16h thermal sensors...                           No
AMD Family 17h thermal sensors...                           Success!
    (driver `k10temp')
AMD Family 15h power sensors...                             No
AMD Family 16h power sensors...                             No
Hygon Family 18h thermal sensors...                         No
Intel digital thermal sensor...                             No
Intel AMB FB-DIMM thermal sensor...                         No
Intel 5500/5520/X58 thermal sensor...                       No
VIA C7 thermal sensor...                                    No
VIA Nano thermal sensor...                                  No

Some Super I/O chips contain embedded sensors. We have to write to
standard I/O ports to probe them. This is usually safe.
Do you want to scan for Super I/O sensors? (YES/no): 
Probing for Super-I/O at 0x2e/0x2f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               No
Trying family `ITE'...                                      No
Probing for Super-I/O at 0x4e/0x4f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               Yes
Found `Nuvoton NCT6797D Super IO Sensors'                   Success!
    (address 0xa20, driver `nct6775')

Some systems (mainly servers) implement IPMI, a set of common interfaces
through which system health data may be retrieved, amongst other things.
We first try to get the information from SMBIOS. If we don't find it
there, we have to read from arbitrary I/O ports to probe for such
interfaces. This is normally safe. Do you want to scan for IPMI
interfaces? (YES/no): 
Probing for `IPMI BMC KCS' at 0xca0...                      No
Probing for `IPMI BMC SMIC' at 0xca8...                     No

Some hardware monitoring chips are accessible through the ISA I/O ports.
We have to write to arbitrary I/O ports to probe them. This is usually
safe though. Yes, you do have ISA I/O ports even if you do not have any
ISA slots! Do you want to scan the ISA I/O ports? (yes/NO): 

Lastly, we can probe the I2C/SMBus adapters for connected hardware
monitoring devices. This is the most risky part, and while it works
reasonably well on most systems, it has been reported to cause trouble
on some systems.
Do you want to probe the I2C/SMBus adapters now? (YES/no): 
Using driver `i2c-piix4' for device 0000:00:14.0: AMD KERNCZ SMBus
[... 75 lines cut ...]
in13:                  688.00 mV (min =  +0.00 V, max =  +0.00 V)  ALARM
in14:                    1.54 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
fan1:                   956 RPM  (min =    0 RPM)
fan2:                  1243 RPM  (min =    0 RPM)
fan3:                   697 RPM  (min =    0 RPM)
fan4:                     0 RPM  (min =    0 RPM)
fan5:                     0 RPM  (min =    0 RPM)
fan6:                   710 RPM  (min =    0 RPM)
fan7:                     0 RPM  (min =    0 RPM)
SYSTIN:                 +40.0°C  (high = +80.0°C, hyst = +75.0°C)  sensor = CPU diode
CPUTIN:                 +41.0°C  (high = +115.0°C, hyst = +90.0°C)  sensor = thermistor
AUXTIN0:                +47.0°C  (high = +115.0°C, hyst = +90.0°C)  sensor = thermistor
AUXTIN1:               -128.0°C    sensor = thermistor
AUXTIN2:                +59.0°C    sensor = thermistor
AUXTIN3:                 -3.0°C    sensor = thermistor
SMBUSMASTER 0:          +44.5°C  
PCH_CHIP_CPU_MAX_TEMP:   +0.0°C  
PCH_CHIP_TEMP:           +0.0°C  
PCH_CPU_TEMP:            +0.0°C  
intrusion0:            ALARM
intrusion1:            ALARM
beep_enable:           disabled
```

but can somebody explain how to interpret the output? E.g. what is SYSTIN, what is CPUTIN, what is AUXTIN0-3 etc., what is high and hyst. I've also noticed that for other CPUs there's usually per-core temperature. Is it not available for this CPU?

commit 00e7d37b (2019-07-23): sensors-detect: Add detection of AMD family 17h, model 70h

k10temp support was added in groeck/hwmon-next commit
22cac1016a4831922d2.

Fixes #187