lm-sensors/lm-sensors #190: ARM no longer building with glibc-2.30

MilhouseVH, 2019-08-03:
`glibc-2.30` removes [sys/io.h](https://sourceware.org/glibc/wiki/Release/2.30#A.3Csys.2BAC8-io.h.3E_removed_on_32-bit_Arm) for ARM, and `lm_sensors-3.5.0` (for arch `arm`) no longer builds.

ARM build log: http://ix.io/1Qq2

```
sed -e 's@^\(.*\)\.o:@lib/conf-parse.ld lib/conf-parse.lo: Makefile '`dirname lib/conf-parse.ld`/Module.mk' @' > lib/conf-parse.ld
/home/ubuntu/projects/LibreELEC.tv/build.LibreELEC-RPi.arm-9.80-devel/toolchain/bin/armv6zk-libreelec-linux-gnueabi-gcc -M -MG -DETCDIR="\"/etc\"" -I.  -Wall -O2 -Wstrict-prototypes -Wshadow -Wpointer-arith -Wcast-qual -Wcast-align -Wwrite-strings -Wnested-externs -Winline -W -Wmissing-prototypes -Wundef -march=armv6zk -mtune=arm1176jzf-s -mabi=aapcs-linux -Wno-psabi -Wa,-mno-warn-deprecated -mfloat-abi=hard -mfpu=vfp -fomit-frame-pointer -Wall -pipe  -O2 lib/conf-lex.c | \
sed -e 's@^\(.*\)\.o:@lib/conf-lex.ad lib/conf-lex.ao: Makefile '`dirname lib/conf-lex.ad`/Module.mk' @' > lib/conf-lex.ad
/home/ubuntu/projects/LibreELEC.tv/build.LibreELEC-RPi.arm-9.80-devel/toolchain/bin/armv6zk-libreelec-linux-gnueabi-gcc -M -MG -DETCDIR="\"/etc\"" -I.  -fpic -D_REENTRANT -Wall -O2 -Wstrict-prototypes -Wshadow -Wpointer-arith -Wcast-qual -Wcast-align -Wwrite-strings -Wnested-externs -Winline -W -Wmissing-prototypes -Wundef -march=armv6zk -mtune=arm1176jzf-s -mabi=aapcs-linux -Wno-psabi -Wa,-mno-warn-deprecated -mfloat-abi=hard -mfpu=vfp -fomit-frame-pointer -Wall -pipe  -O2 lib/conf-lex.c | \
sed -e 's@^\(.*\)\.o:@lib/conf-lex.ld lib/conf-lex.lo: Makefile '`dirname lib/conf-lex.ld`/Module.mk' @' > lib/conf-lex.ld
make: *** No rule to make target 'sys/io.h', needed by 'prog/dump/superio.rd'.  Stop.
```

olysonek, 2019-08-05:
Thanks for the report! I'll look into this a bit later.

lucasrangit, 2019-08-07:
Hi @olysonek and @MilhouseVH ,

I was dealing with cross-compiling the ISA programs that used I/O port calls on ARM in the same way that PPC was handled, by not compiling them :wink: . Since the glibc-2.30 release notes mention how these methods didn't actually work on ARM and ARM64, I think this is a valid fix.

However, a more portable alternative would be to `open/lseek/read/write` `/dev/port`. Let me know if you want me to try that.

MilhouseVH, 2019-08-07:
Thanks #192 seems to be working fine - I'm able to build `lm_sensors` for x86_64 and ARM with `glibc-2.30`.

lucasrangit, 2019-08-08:
> Thanks #192 seems to be working fine - I'm able to build `lm_sensors` for x86_64 and ARM with `glibc-2.30`.

Can you include a link to the build log for completeness?

olysonek, 2019-08-08:
Oh, so only cross-compilation doesn't work? I didn't realize that.

lucasrangit, 2019-08-08:
Cross-compilation works; the binaries produced with `make CC=armv6zk-libreelec-linux-gnueabi-gcc AR=armv6zk-libreelec-linux-gnueabi-ar` are compatible. However, the `isadump` and `isaset` tools would not work on non-x86 (run-time issue).

The issue is that the `Makefile` test uses `uname -m` to exclude certain binaries for PPC (non-Intel), which assumes the build host and target are the same.

The issue is now a build-time issue because of the changes in `glibc` 2.30.

I will explain this better in the commit log message in the PR.

MilhouseVH, 2019-08-08:
> Can you include a link to the build log for completeness?

Building for RPi (armv6) with this PR:
```
[1;36mUNPACK[0m      lm_sensors
    [1;32mAPPLY PATCH [1;37m(common)[0m[0m      packages/addons/addon-depends/system-tools-depends/lm_sensors/patches/lm_sensors-01_link-static.patch
patching file prog/sensors/Module.mk
    [1;32mAPPLY PATCH [1;37m(common)[0m[0m      packages/addons/addon-depends/system-tools-depends/lm_sensors/patches/lm_sensors-02-PR192.patch
patching file Makefile
[1;33mBUILD[0m      lm_sensors [1;37m(target)[0m
    [1;35mTOOLCHAIN[0m      make (auto-detect)
Executing (target): make PREFIX=/usr CC=/home/ubuntu/projects/LibreELEC.tv/build.LibreELEC-RPi.arm-9.80-devel/toolchain/bin/armv6zk-libreelec-linux-gnueabi-gcc AR=/home/ubuntu/projects/LibreELEC.tv/build.LibreELEC-RPi.arm-9.80-devel/toolchain/bin/armv6zk-libreelec-linux-gnueabi-ar
/home/ubuntu/projects/LibreELEC.tv/build.LibreELEC-RPi.arm-9.80-devel/toolchain/bin/armv6zk-libreelec-linux-gnueabi-gcc -M -MG -DETCDIR="\"/etc\"" -I.  -Wall -O2 -Wstrict-prototypes -Wshadow -Wpointer-arith -Wcast-qual -Wcast-align -Wwrite-strings -Wnested-externs -Winline -W -Wmissing-prototypes -Wundef -march=armv6zk -mtune=arm1176jzf-s -mabi=aapcs-linux -Wno-psabi -Wa,-mno-warn-deprecated -mfloat-abi=hard -mfpu=vfp -fomit-frame-pointer -Wall -pipe  -O2 prog/sensors/chips.c | \
sed -e 's@^\(.*\)\.o:@prog/sensors/chips.rd prog/sensors/chips.ro: Makefile '`dirname prog/sensors/chips.rd`/Module.mk' @' > prog/sensors/chips.rd
/home/ubuntu/projects/LibreELEC.tv/build.LibreELEC-RPi.arm-9.80-devel/toolchain/bin/armv6zk-libreelec-linux-gnueabi-gcc -M -MG -DETCDIR="\"/etc\"" -I.  -Wall -O2 -Wstrict-prototypes -Wshadow -Wpointer-arith -Wcast-qual -Wcast-align -Wwrite-strings -Wnested-externs -Winline -W -Wmissing-prototypes -Wundef -march=armv6zk -mtune=arm1176jzf-s -mabi=aapcs-linux -Wno-psabi -Wa,-mno-warn-deprecated -mfloat-abi=hard -mfpu=vfp -fomit-frame-pointer -Wall -pipe  -O2 prog/sensors/main.c | \
sed -e 's@^\(.*\)\.o:@prog/sensors/main.rd prog/sensors/main.ro: Makefile '`dirname prog/sensors/main.rd`/Module.mk' @' > prog/sensors/main.rd
/home/ubuntu/projects/LibreELEC.tv/build.LibreELEC-RPi.arm-9.80-devel/toolchain/bin/armv6zk-libreelec-linux-gnueabi-gcc -M -MG -DETCDIR="\"/etc\"" -I.  -Wall -O2 -Wstrict-prototypes -Wshadow -Wpointer-arith -Wcast-qual -Wcast-align -Wwrite-strings -Wnested-externs -Winline -W -Wmissing-prototypes -Wundef -march=armv6zk -mtune=arm1176jzf-s -mabi=aapcs-linux -Wno-psabi -Wa,-mno-warn-deprecated -mfloat-abi=hard -mfpu=vfp -fomit-frame-pointer -Wall -pipe  -O2 lib/sysfs.c | \
sed -e 's@^\(.*\)\.o:@lib/sysfs.ad lib/sysfs.ao: Makefile '`dirname lib/sysfs.ad`/Module.mk' @' > lib/sysfs.ad
/home/ubuntu/projects/LibreELEC.tv/build.LibreELEC-RPi.arm-9.80-devel/toolchain/bin/armv6zk-libreelec-linux-gnueabi-gcc -M -MG -DETCDIR="\"/etc\"" -I.  -Wall -O2 -Wstrict-prototypes -Wshadow -Wpointer-arith -Wcast-qual -Wcast-align -Wwrite-strings -Wnested-externs -Winline -W -Wmissing-prototypes -Wundef -march=armv6zk -mtune=arm1176jzf-s -mabi=aapcs-linux -Wno-psabi -Wa,-mno-warn-deprecated -mfloat-abi=hard -mfpu=vfp -fomit-frame-pointer -Wall -pipe  -O2 lib/init.c | \
sed -e 's@^\(.*\)\.o:@lib/init.ad lib/init.ao: Makefile '`dirname lib/init.ad`/Module.mk' @' > lib/init.ad
/home/ubuntu/projects/LibreELEC.tv/build.LibreELEC-RPi.arm-9.80-devel/toolchain/bin/bison
bison -p sensors_yy -d lib/conf-parse.y -o lib/conf-parse.c
/home/ubuntu/projects/LibreELEC.tv/build.LibreELEC-RPi.arm-9.80-devel/toolchain/bin/armv6zk-libreelec-linux-gnueabi-gcc -M -MG -DETCDIR="\"/etc\"" -I.  -Wall -O2 -Wstrict-prototypes -Wshadow -Wpointer-arith -Wcast-qual -Wcast-align -Wwrite-strings -Wnested-externs -Winline -W -Wmissing-prototypes -Wundef -march=armv6zk -mtune=arm1176jzf-s -mabi=aapcs-linux -Wno-psabi -Wa,-mno-warn-deprecated -mfloat-abi=hard -mfpu=vfp -fomit-frame-pointer -Wall -pipe  -O2 lib/access.c | \
sed -e 's@^\(.*\)\.o:@lib/access.ad lib/access.ao: Makefile '`dirname lib/access.ad`/Module.mk' @' > lib/access.ad
/home/ubuntu/projects/LibreELEC.tv/build.LibreELEC-RPi.arm-9.80-devel/toolchain/bin/armv6zk-libreelec-linux-gnueabi-gcc -M -MG -DETCDIR="\"/etc\"" -I.  -Wall -O2 -Wstrict-prototypes -Wshadow -Wpointer-arith -Wcast-qual -Wcast-align -Wwrite-strings -Wnested-externs -Winline -W -Wmissing-prototypes -Wundef -march=armv6zk -mtune=arm1176jzf-s -mabi=aapcs-linux -Wno-psabi -Wa,-mno-warn-deprecated -mfloat-abi=hard -mfpu=vfp -fomit-frame-pointer -Wall -pipe  -O2 lib/error.c | \
sed -e 's@^\(.*\)\.o:@lib/error.ad lib/error.ao: Makefile '`dirname lib/error.ad`/Module.mk' @' > lib/error.ad
/home/ubuntu/projects/LibreELEC.tv/build.LibreELEC-RPi.arm-9.80-devel/toolchain/bin/armv6zk-libreelec-linux-gnueabi-gcc -M -MG -DETCDIR="\"/etc\"" -I.  -Wall -O2 -Wstrict-prototypes -Wshadow -Wpointer-arith -Wcast-qual -Wcast-align -Wwrite-strings -Wnested-externs -Winline -W -Wmissing-prototypes -Wundef -march=armv6zk -mtune=arm1176jzf-s -mabi=aapcs-linux -Wno-psabi -Wa,-mno-warn-deprecated -mfloat-abi=hard -mfpu=vfp -fomit-frame-pointer -Wall -pipe  -O2 lib/general.c | \
sed -e 's@^\(.*\)\.o:@lib/general.ad lib/general.ao: Makefile '`dirname lib/general.ad`/Module.mk' @' > lib/general.ad
/home/ubuntu/projects/LibreELEC.tv/build.LibreELEC-RPi.arm-9.80-devel/toolchain/bin/armv6zk-libreelec-linux-gnueabi-gcc -M -MG -DETCDIR="\"/etc\"" -I.  -Wall -O2 -Wstrict-prototypes -Wshadow -Wpointer-arith -Wcast-qual -Wcast-align -Wwrite-strings -Wnested-exter [... cut ...] xterns -Winline -W -Wmissing-prototypes -Wundef -march=armv6zk -mtune=arm1176jzf-s -mabi=aapcs-linux -Wno-psabi -Wa,-mno-warn-deprecated -mfloat-abi=hard -mfpu=vfp -fomit-frame-pointer -Wall -pipe  -O2 -c lib/test/test-scanner.c -o lib/test/test-scanner.ro
/home/ubuntu/projects/LibreELEC.tv/build.LibreELEC-RPi.arm-9.80-devel/toolchain/bin/armv6zk-libreelec-linux-gnueabi-gcc -Wl,-rpath,/usr/lib -o lib/test/test-scanner lib/test/test-scanner.ro lib/conf-lex.ao lib/error.ao lib/general.ao -Llib
```

PR #192 (2019-08-08): makefile: use compiler target arch when determining what programs to compile
`uname -m` lists the host machine architecture. To support cross-compilation
the Makefile should detect the compiler's target architecture. This approach
uses the `gcc`/`clang` compatible `-dumpmachine` option and parses the first
word in the tuple representing the target architecture.

Renaming `MACHINE` to `ARCH` to match cross-compiler conventions for
specifying the target architecture (e.g.
`make ARCH=arm CC=arm-linux-gnueabi-gcc`)

Just like for PPC the `isadump` and `isaset` tools should not be compiled
on ARM.  [... cut]
commit 0863eff8 (2019-08-08): makefile: use target arch when determining what programs to compile

To better support cross-compilation, the Makefile should detect the
compiler's target architecture instead of using `uname -m`, which
returns the host architecture. This approach uses the `gcc`/`clang`
compatible `-dumpmachine` option.

Cross-compilation worked but because of this bug, `isadump` and
`isaset` x86 ISA specific tools were built for PCC and ARM.

This run-time issue became a compile-time issue when `glibc` 2.30
removed the `sys/io.h` I/O port functions on ARM.

Also renamed `MACHINE` to `ARCH` to match cross-comp [... cut]
closed 2019-08-08 by olysonek