lm-sensors/lm-sensors #193: Ubuntu 18.04.3 LTS - New Updated kernel to 5.0.0.23-generic issue

mtytula, 2019-08-08:
Hello, after installed updates and rebooted my Precision 5530, most of sensors disappeared from the list. After running **sensors**, I get only these sensors.

`dell_smm-virtual-0
Adapter: Virtual device
fan1:           0 RPM
fan2:           0 RPM

coretemp-isa-0000
Adapter: ISA adapter
Package id 0:  +41.0°C  (high = +100.0°C, crit = +100.0°C)
Core 0:        +39.0°C  (high = +100.0°C, crit = +100.0°C)
Core 1:        +40.0°C  (high = +100.0°C, crit = +100.0°C)
Core 2:        +40.0°C  (high = +100.0°C, crit = +100.0°C)
Core 3:        +40.0°C  (high = +100.0°C, crit = +100.0°C)
Core 4:        +39.0°C  (high = +100.0°C, crit = +100.0°C)
Core 5:        +39.0°C  (high = +100.0°C, crit = +100.0°C)
`

Could you investigate?

olysonek, 2019-08-09:
What version of lm_sensors are you running? You'll probably need 3.5.0.

mtytula, 2019-08-10:
I have the package 1:3.4.0-4 (3.4.0). Running ```sudo apt-get install lm-sensors``` says that I have already the newest version.

olysonek, 2019-08-12:
Well, that may be the latest available version in Ubuntu, however I've already released 3.5.0 upstream. You'll need that version with the kernel you're running.

I'm closing this now. If upgrading to 3.5.0 doesn't fix the problem, please reopen.

closed 2019-08-12 by olysonek