lm-sensors/lm-sensors #195: ITE IT8665E support

imareo, 2019-08-17:
Hello! mb asus b450-i, ite it8665e 

> Probing for Super-I/O at 0x2e/0x2f
....
Trying family ITE...                                      Yes
Found unknown chip with ID 0x8665
    (logical device 4 has address 0x290, could be sensors)

RuedigerDieter, 2019-08-21:
Same Mobo, same error,
using force_id=0x8665 in the /etc/modprobe.d/it87.conf did not work

erikhakansson, 2019-10-19:
Same with X470F

guedressel, 2019-11-10:
Got an ASUS PRIME B450-PLUS here (running Debian buster/Proxmox PVE):

    [...]
    Trying family `ITE'...                                      Yes
    Found `ITE IT8665E Super IO Sensors'                        Success!
        (address 0x290, driver `to-be-written')
    [...]

Pq11911, 2020-02-10:
Same here ASUS PRIME X470-PRO .

Trying family `ITE'...                                      Yes
Found `ITE IT8665E Super IO Sensors'                        Success!
    (address 0x290, driver `to-be-written')

Driver `to-be-written':
  * ISA bus, address 0x290
    Chip `ITE IT8665E Super IO Sensors' (confidence: 9)

Driver `k10temp' (autoloaded):
  * Chip `AMD Family 17h thermal sensors' (confidence: 9)

Note: there is no driver for ITE IT8665E Super IO Sensors yet.
Check https://hwmon.wiki.kernel.org/device_support_status for updates.

gsedej, 2020-02-19:
It works for me using this github https://github.com/a1wong/it87 (can conform it works for IT8556E, IT8686E, and IT8628E)

UNIm95, 2020-04-25:
I can confirm message of @gsedej  
I have used same repository and now i got sensors reading on ASUS TUF B450M-Pro gaming.
They are have not proper low/high warnings.
```
asus-isa-0000
Adapter: ISA adapter
cpu_fan:        0 RPM

radeon-pci-0700
Adapter: PCI adapter
temp1:        +34.0°C  (crit = +120.0°C, hyst = +90.0°C)

it8655-isa-0290
Adapter: ISA adapter
in0:          +1.05 V  (min =  +2.76 V, max =  +2.14 V)  ALARM
in1:          +2.52 V  (min =  +2.00 V, max =  +1.74 V)  ALARM
in2:          +2.00 V  (min =  +2.04 V, max =  +2.66 V)  ALARM
in3:          +1.98 V  (min =  +1.47 V, max =  +2.73 V)
in4:          +2.00 V  (min =  +1.16 V, max =  +0.20 V)  ALARM
in5:          +1.92 V  (min =  +0.59 V, max =  +2.06 V)
in6:          +1.98 V  (min =  +2.76 V, max =  +2.62 V)  ALARM
3VSB:         +3.33 V  (min =  +5.32 V, max =  +4.12 V)  ALARM
Vbat:         +3.25 V  
+3.3V:        +3.38 V  
fan1:         654 RPM  (min =   10 RPM)
fan2:           0 RPM  (min =  124 RPM)  ALARM
fan3:           0 RPM  (min =   22 RPM)  ALARM
temp1:        +38.0°C  (low  = +59.0°C, high = -72.0°C)  ALARM
temp2:        +35.0°C  (low  = -105.0°C, high =  +3.0°C)  ALARM  sensor = disabled
temp3:        +39.0°C  (low  = +63.0°C, high = -124.0°C)  ALARM  sensor = disabled
temp4:        +39.0°C  (low  = +109.0°C, high = +115.0°C)  sensor = disabled
temp5:        +39.0°C  (low  = +95.0°C, high = -25.0°C)  ALARM  sensor = disabled
temp6:        +39.0°C  (low  = +17.0°C, high = -69.0°C)  ALARM  sensor = disabled
intrusion0:  ALARM

k10temp-pci-00c3
Adapter: PCI adapter
Tdie:         +38.8°C  (high = +70.0°C)
Tctl:         +38.8°C  
```

shag00, 2020-05-21:
OK, did:
* make clean
* make
* sudo make install
 on an ASUS Prime B450 and AMD 2600X and nothing happened in LM Sensors.
I thought something would be added in etc/modules but there was nothing. Is there another step after sudo>make>install?

UNIm95, 2020-05-21:
@shag00 
You forgot to load module.  
Run as root `sensors-detect`.  
At the end it will show you extra module `lm87`.   
You can load it with `modprobe lm87.`  
And for reboot survival you need to add it in `/etc/modules`

shag00, 2020-05-23:
Seems there must be another issue. After make install I sudo sensors detect and get the following both before and after running modprobe lm87:
`Now follows a summary of the probes I have just done.
Just press ENTER to continue: 

Driver `to-be-written':
  * ISA bus, address 0x290
    Chip `ITE IT8665E Super IO Sensors' (confidence: 9)

Driver `k10temp' (autoloaded):
  * Chip `AMD Family 17h thermal sensors' (confidence: 9)

Note: there is no driver for ITE IT8665E Super IO Sensors yet.
Check https://hwmon.wiki.kernel.org/device_support_status for updates.

No modules to load, skipping modules configuration.

Unloading cpuid... OK`

dgalli1, 2020-08-21:
Doesn't seem to work for me with Linux Kernel 5.5 and Asus PRIME B350M-K

caniko, 2021-02-04:
Found the AUR repo: https://aur.archlinux.org/packages/it87-dkms-git

SteavenGamerYT, 2024-07-01:
[it87-dkms-git](https://aur.archlinux.org/packages/it87-dkms-git)
Doesnt work with me
```sh
Driver `to-be-written':
  * ISA bus, address 0x290
    Chip `ITE IT8665E Super IO Sensors' (confidence: 9)

Note: there is no driver for ITE IT8665E Super IO Sensors yet.
Check https://hwmon.wiki.kernel.org/device_support_status for updates.
```

motherboard: ASUS PRIME B450-PLUS

amfern, 2025-02-02:
I have BIOSTAR B650EGTQ. Which also has the `ITE IT8665E`
Works for me, but you will need to also recompile the sensors-detect.
https://github.com/lm-sensors/lm-sensors/blob/42f240d2a457834bcbdf4dc8b57237f97b5f5854/prog/detect/sensors-detect#L2543
```
diff --git a/prog/detect/sensors-detect b/prog/detect/sensors-detect
index 4a73e1f9..8d2c8c09 100755
--- a/prog/detect/sensors-detect
+++ b/prog/detect/sensors-detect
@@ -2540,7 +2540,7 @@ use constant FEAT_SMBUS   => (1 << 7);
        features => FEAT_IN | FEAT_FAN | FEAT_TEMP,
    }, {
        name => "ITE IT8625E Super IO Sensors",
-       driver => "to-be-written",  # it87
+       driver => "it87",   # it87
        devid => 0x8625,
        logdev => 0x04,
        features => FEAT_IN | FEAT_FAN | FEAT_TEMP,
--
2.48.1

```

AstralPhnx, 2026-09-07:
> If you are on arch, you can install a patched version of lm_sensors [lm_sensors-it87-git](https://aur.archlinux.org/packages/lm_sensors-it87-git) that adds support for IT8665E and other new chips supported by https://github.com/frankcrawford/it87

Does this package exist in a way that is easily buildable for any other distro? Currently on Debian so a PKGBUILD is kinda useless

amfern, 2026-09-07:
@AstralPhnx Debian doesn't have an equivalent to Arch Linux's AUR, so you will have to manually apply the patches and build from source:

1. Apply the patches:
https://aur.archlinux.org/cgit/aur.git/tree/0001-Add-support-for-ITE-IT8625E-Super-IO-Sensors.patch?h=lm_sensors-it87-git 
https://aur.archlinux.org/cgit/aur.git/tree/0001-fix-incompatible-pointer-warnings.patch?h=lm_sensors-it87-git

2. Build from source.
https://github.com/lm-sensors/lm-sensors/blob/master/INSTALL