lm-sensors/lm-sensors #202: Support for Raspberry PI 4

anITiot, 2019-10-03:
Hi, I'm running Ubuntu 18.04 64bit on my RPI4 and the processor seems to be unsupported.

vielmetti, 2020-05-25:
Also the case under Ubuntu 20.04 64bit on the Pi 4.

sensors command returns this:

```
emv@pinnatus:~$ sensors
rpi_volt-isa-0000
Adapter: ISA adapter
in0:              N/A  

```

whereas Ubuntu 20.04 64-bit on the Pi 3 return this

```
ubuntu@ubuntu:/sys/class/thermal/thermal_zone0$ sensors
cpu_thermal-virtual-0
Adapter: Virtual device
temp1:        +52.1°C  

rpi_volt-isa-0000
Adapter: ISA adapter
in0:              N/A  
```

sg3524, 2021-04-06:
Verifying what @vielmetti has already reported. Under 20.04 I am seeing this

```
$ sensors
cpu_thermal-virtual-0
Adapter: Virtual device
temp1:        +43.3°C  

rpi_volt-isa-0000
Adapter: ISA adapter
in0:              N/A 
```

Lin-Admin, 2021-12-03:
It's now more than 2 years since that report has been created.
Is there any hope that the maintainers of lm-sensors will take action within a few weeks or should I look for other solutions?

aurel32, 2021-12-03:
Nothing will be done in lm-sensors, as this is a kernel issue. lm-sensors only displays what is exported by the kernel in /sys.

aurel32, 2021-12-03:
The problem has been solved with the following commit of kernel 5.6: https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/?id=59b781352dc4cb9ae27a8ddae0cda979d29d8af7

This bug should just be closed.

eugeneonline, 2022-10-24:
I still have same issue.

root@raspberrypi:/home/ubuntu# sensors
cpu_thermal-virtual-0
Adapter: Virtual device
temp1:        +38.5°C  

rpi_volt-isa-0000
Adapter: ISA adapter
in0:              N/A  

root@raspberrypi:/home/ubuntu# cat /etc/os-release 
PRETTY_NAME="Ubuntu 22.04.1 LTS"
NAME="Ubuntu"
VERSION_ID="22.04"
VERSION="22.04.1 LTS (Jammy Jellyfish)"
VERSION_CODENAME=jammy
ID=ubuntu
ID_LIKE=debian
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
UBUNTU_CODENAME=jammy

NUAAerJC, 2023-01-18:
I still have this issue.
```bash
 codery@raspberrypii:~$ cat /etc/os-release
PRETTY_NAME="Debian GNU/Linux 11 (bullseye)"
NAME="Debian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"
 codery@raspberrypii:~$ sensors
cpu_thermal-virtual-0
Adapter: Virtual device
temp1:        +27.3°C

rpi_volt-isa-0000
Adapter: ISA adapter
in0:              N/A

 codery@raspberrypi:~$ uname -r
5.15.76-v8+
 codery@raspberrypii:~$ uname -a
Linux raspberrypi 5.15.76-v8+ #1597 SMP PREEMPT Fri Nov 4 12:16:41 GMT 2022 aarch64 GNU/Linux
```
It seems that the thermal sensor driver for the  SoC on Raspberry Pi 4B doesn't work according to previous comments under this issue.

Lin-Admin, 2023-01-18:
> The problem has been solved with the following commit of kernel 5.6: https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/?id=59b781352dc4cb9ae27a8ddae0cda979d29d8af7
> 
> This bug should just be closed.

That link only handles the thermal problem, not the "rpi_volt-isa-0000"

Lin-Admin, 2023-01-18:
Linux R 5.15.0-1023-raspi #25-Ubuntu SMP still does not handle "rpi_volt-isa-0000".
Maybe this is relied to the dmesg warning _usb_phy_generic phy: supply vcc not found, using dummy regulator_?

HackerDaGreat57, 2023-02-21:
also not working under Ubuntu 22.04 Server LTS (aarch64 raspi 4 8gb ram)

lexfrei, 2023-03-17:
The same on the latest Rocky:

```
[root@node01 ~]# sensors
cpu_thermal-virtual-0
Adapter: Virtual device
temp1:        +53.6°C  (crit = +110.0°C)

rpi_volt-isa-0000
Adapter: ISA adapter
in0:              N/A

[root@node01 ~]# uname -a
Linux node01 6.1.8-v8.1.el9.altarch.1 #1 SMP PREEMPT Wed Feb  8 21:28:03 UTC 2023 aarch64 aarch64 aarch64 GNU/Linux
```

jwillikers, 2023-12-30:
Same issue on Fedora 39.

teslazap, 2024-01-03:
cpu_thermal-virtual-0 working
rpi_volt-isa-0000 not working

Ubuntu 23.10
Linux 6.5.0-1008-raspi #11-Ubuntu SMP PREEMPT_DYNAMIC Wed Nov 22 19:08:26 UTC 2023 aarch64