lm-sensors/lm-sensors #204: Multiple 1-wire sensors (e.g. DS18B20) indistinguishable

trs-mmuc, 2019-10-22:
I have 2 DS18B20 sensors attached to my Pi, but they both appear as w1_slave_temp-virtual-0 with no other properties to distinguish them from each other

**pi@relay-pi:~ $ ls /sys/bus/w1/devices/**
28-01144d29faaa  28-01144da0d8aa  w1_bus_master1
**pi@relay-pi:~ $ sensors**
rpi_volt-isa-0000
Adapter: ISA adapter
in0:              N/A

w1_slave_temp-virtual-0
Adapter: Virtual device
temp1:        +19.9°C

w1_slave_temp-virtual-0
Adapter: Virtual device
temp1:        +19.9°C

mishop, 2019-12-14:
I have same problem 

root@raspberrypi:/etc/telegraf# sensors
rpi_volt-isa-0000
Adapter: ISA adapter
in0:              N/A  

w1_slave_temp-virtual-0
Adapter: Virtual device
temp1:         +4.8°C  

w1_slave_temp-virtual-0
Adapter: Virtual device
temp1:        +14.7°C

olysonek, 2019-12-16:
Ack. We do need to do something about this. Patches welcome.

mishop, 2020-01-21:
I write own working solution:
https://github.com/mishop/sensors/

olysonek, 2020-01-21:
Thanks for the info.

tedb, 2020-08-25:
I am seeing the same issue with `sensors` on a Raspberry Pi with two 1-wire temperature probes (DS18B20).

It looks like the two sensors are not distinguished because they both have the same string in `/sys/class/hwmon/hwmon*/name`: `w1_slave_temp`.  Their 1-wire serial numbers show up in `/sys/class/hwmon/hwmon*/device/name`.

Would it make sense for `sensors` to read the device names from there?  Maybe read it conditionally if the name is `w1_slave_temp`?

Some data:

```
$ sensors -j
{
   "w1_slave_temp-virtual-0":{
      "Adapter": "Virtual device",
      "temp1":{
         "temp1_input": 23.937
      }
   },
   "cpu_thermal-virtual-0":{
      "Adapter": "Virtual device",
      "temp1":{
         "temp1_input": 46.540
      }
   },
   "rpi_volt-isa-0000":{
      "Adapter": "ISA adapter",
      "in0":{
         "in0_lcrit_alarm": 0.000
      }
   },
   "w1_slave_temp-virtual-0":{
      "Adapter": "Virtual device",
      "temp1":{
         "temp1_input": 24.187
      }
   }
}

cat /sys/class/hwmon/hwmon*/name
cpu_thermal
w1_slave_temp
w1_slave_temp

$ cat /sys/class/hwmon/hwmon*/device/name
28-0218508170ff
28-011850aecaff

$ cat /sys/bus/w1/devices/w1_bus_master1/w1_master_slaves
28-0218508170ff
28-011850aecaff

$ cat /sys/bus/w1/devices/28-*/name
28-011850aecaff
28-0218508170ff

```

dorinvlas, 2022-08-13:
> I am seeing the same issue with `sensors` on a Raspberry Pi with two 1-wire temperature probes (DS18B20).
> 
> It looks like the two sensors are not distinguished because they both have the same string in `/sys/class/hwmon/hwmon*/name`: `w1_slave_temp`. Their 1-wire serial numbers show up in `/sys/class/hwmon/hwmon*/device/name`.
> 
> Would it make sense for `sensors` to read the device names from there? Maybe read it conditionally if the name is `w1_slave_temp`?
> 
> An issue with Telegraf seems related: https://community.influxdata.com/t/implementation-of-multiple-ds18b20/8012
> 
> Some data:
> 
> ```
> $ sensors -j
> {
>    "w1_slave_temp-virtual-0":{
>       "Adapter": "Virtual device",
>       "temp1":{
>          "temp1_input": 23.937
>       }
>    },
>    "cpu_thermal-virtual-0":{
>       "Adapter": "Virtual device",
>       "temp1":{
>          "temp1_input": 46.540
>       }
>    },
>    "rpi_volt-isa-0000":{
>       "Adapter": "ISA adapter",
>       "in0":{
>          "in0_lcrit_alarm": 0.000
>       }
>    },
>    "w1_slave_temp-virtual-0":{
>       "Adapter": "Virtual device",
>       "temp1":{
>          "temp1_input": 24.187
>       }
>    }
> }
> 
> cat /sys/class/hwmon/hwmon*/name
> cpu_thermal
> w1_slave_temp
> w1_slave_temp
> 
> $ cat /sys/class/hwmon/hwmon*/device/name
> 28-0218508170ff
> 28-011850aecaff
> 
> $ cat /sys/bus/w1/devices/w1_bus_master1/w1_master_slaves
> 28-0218508170ff
> 28-011850aecaff
> 
> $ cat /sys/bus/w1/devices/28-*/name
> 28-011850aecaff
> 28-0218508170ff
> ```

did you found a solution to the problem you described? Seems same for my 4 temp probes... :(

Thanks

dorinvlas, 2022-08-13:
actually you have to compile it using `go build`, then you will have your sensors executable that you can put in /usr/bin and use it with inputs.sensors.