lm-sensors/lm-sensors #206: Support for ITE IT8987D

MaximeLT, 2019-10-25:
I'm getting "Found unknown chip with ID 0x8987".
And I can't get fan data, making Linux unable to handle fan speed and it drags some bug along.

If we could get a driver support for it, i'd be awesome.

ksofix, 2020-10-17:
I have the same problem on my Intel Nuc NUC7i5BNK.
It has a hardware monitoring subsystem, based on an ITE IT8987D embedded controller, including:
- Voltage sense to detect out of range power supply voltages
- Thermal sense to detect out of range thermal values
- One processor fan header
- Fan sense input used to monitor fan activity
- Simple fan speed control

It would be great to have support of this chip in lm-sensors.

RonnySvedman, 2020-11-14:
have you tried the latest betas out there of the it87 module? lm-sensors can only work with what hwmon reports.  ITE super IO chips has been a thorn in the butt of linux users since before 2010 and keeps on going. The maintainer quit in 2018 and now there are several unofficial forks of the code. [(https://github.com/a1wong/it87)] worked for me on my asus z97p, but you will have to edit sensors3.conf to fix scaling, labeling and limits by hand, once you get some values that actually change. (and you will get several other values that are bogus included in the deal)