lm-sensors/lm-sensors #209: Gigabyte Z390 UD

eyaleb, 2019-11-03:
Just started using this board and I do not see voltages or fan speeds.

I ran `sudo sensors-detect --auto` and it generated `/etc/sysconfig/lm_sensors` with one real line:
```
    HWMON_MODULES="coretemp"
```
How do I proceed from here? I can see that some adapters were not attempted:
```
    Next adapter: i915 gmbus dpb (i2c-0)
    Do you want to scan it? (yes/NO/selectively):
    Next adapter: i915 gmbus dpc (i2c-1)
    Do you want to scan it? (yes/NO/selectively):
    Next adapter: i915 gmbus misc (i2c-2)
    Do you want to scan it? (yes/NO/selectively):
    Next adapter: i915 gmbus dpd (i2c-3)
    Do you want to scan it? (yes/NO/selectively):
```
Should I try a manual run and allow these scans?

eyaleb, 2019-11-17:
I am on fedora 30, and installed is `lm_sensors.3.5.0-6.fc30.x86_64`.

Trying the skipped scans did not help. However, I do notice a line in the standard scan:
```
$ sensors -v
sensors version 3.5.0 with libsensors version 3.5.0

$ sudo sensors-detect
# sensors-detect revision $Revision$
# System: Gigabyte Technology Co., Ltd. Z390 UD [Default string]
# Kernel: 5.3.11-200.fc30.x86_64 x86_64
# Processor: Intel(R) Core(TM) i7-9700 CPU @ 3.00GHz (6/158/13)
...
Some Super I/O chips contain embedded sensors. We have to write to
standard I/O ports to probe them. This is usually safe.
Do you want to scan for Super I/O sensors? (YES/no):
Probing for Super-I/O at 0x2e/0x2f
...
Trying family `ITE'...                                      Yes
Found unknown chip with ID 0x8688
```

I find records of this chip discussed earlier this year [#154](https://github.com/lm-sensors/lm-sensors/issues/154)
The fix mentioned there works.

Is it supported in a newer sensors package now?

LXXero, 2020-01-06:
This might be related to a problem i'm having with the latest gigabyte bios, released the end of october 2019.

They recently released a new bios across all their various newer boards, as of this latest bios, I cannot get IT87 sensors to show up anymore. Normally, I'd just go back to the previous bios, but, this bios has fixed some other annoying graphics card related issues, so for now I've just been living with the broken sensors. Would love to dig some more (isadump, etc?)

Lamieur, 2020-02-01:
LXXero, my guess is that this is not a BIOS issue. This is a kernel issue.

I have an earlier BIOS (F6d, it's 1 year old) and everything works on kernel 5.3, but nothing works on kernel 5.4.

(By "everything works" I mean works after doing echo "options it87 force_id=0x8628" > /etc/modprobe.d/it87-gigabyte.conf + adding acpi_enforce_resources=lax to kernel command line)

My guess it that you've upgraded the BIOS at roughly the same time your distro moved to 5.4.

I would test newer BIOS with kernel 5.3 and 5.4, but Gigabyte is still to discover xml/json/whatever for profile export/import (so you could import a profile after BIOS update) and I really CBA redoing all the tuning I've done every time ;)

eyaleb, 2020-02-03:
I am now on fedora  `5.4.12-100.fc30.x86_64` and `BIOS F8`. Things still work with the above it87 cheat. I am holding off on upgrading to BIOS F9/F10 while watching #154.

Lamieur, 2020-02-28:
Update on the Linux 5.4 and 5.5 situation - I have bisected this problem to commit b84398d6d7f900805662b1619223fd644d862d7c - i2c: i801: Use iTCO version 6 in Cannon Lake PCH and beyond

By googling this commit id I found a thread on lkml where they're trying to fix it (apparently it broke more than just this board and it87), but it also contained a work-around - removing/blacklisting the module iTCO_wdt (you can test online: if you rmmod iTCO_wdt, then rmmod it87 and modprobe it87, it works). This disables the hardware watchdog, but I don't care.

LXXero, 2020-02-28:
interesting - that watchdog work-around seems to have fixed it here as well.