lm-sensors/lm-sensors #210: asus rog strix scar 3 g731gv support

azamet90, 2019-11-13:
this program i did install from kali repo but it is not detecting my fan!? can you please help me about that?

azamet90, 2019-11-16:
this is funny that someone is removing my postings

azamet90, 2019-12-07:
yeah after 25 days later if still no one saying anything means this tool already dead

olysonek, 2019-12-09:
The tool is absolutely *not* dead. There's just noone around who would provide *free* support for it.

What you're describing is most likely not a lm_sensors bug. It's either caused by a kernel bug, misconfiguration or an issue on the HW side. Therefore it's a support request and I feel no obligation (and have no time) to respond to those.

azamet90, 2019-12-20:
> The tool is absolutely _not_ dead. There's just noone around who would provide _free_ support for it.
>
> What you're describing is most likely not a lm_sensors bug. It's either caused by a kernel bug, [...]

the funny thing is that kali linux 2019.4 is giving the installation of distro with lm-sensor pre installed :) but i think this tool is working only on laptop models from 2010 maybe :D

aaronsb, 2020-06-30:
Try updating GRUB config with the line:
GRUB_CMDLINE_LINUX_DEFAULT="acpi_enforce_resources=lax"

It might allow modprobe to load additional drivers revealing more sensors.

RonnySvedman, 2020-11-14:
Asus products are very often using ITE superio chips, which have a long history of being tricky to support due to no specs available. look at the output of sensors-config. Try modprobe it87, and /or compiling it87 from unofficial beta sources [https://github.com/a1wong/it87] or if you find newer ones in some other fork.