lm-sensors/lm-sensors #217: lm-sensors.org is down

Talkless, 2020-01-14:
Hi,
man page hinted about `http://www.lm-sensors.org/wiki/Configurations` page, but it seems whole lm-sensors.org is down..? Browser does not get a response.

olysonek, 2020-01-14:
Hi. It's been down for a couple of years. The new location is at https://hwmon.wiki.kernel.org/lm_sensors.

Which man page are you talking about? Are you sure you're using a recent lm_sensors version?

Talkless, 2020-01-14:
Ouch.. Well, no, not the recent version. It's in Debian 9 (stretch). And actually it was not a man page, but `/etc/sensors3.conf`:
```
grep -B1 Configurations /etc/sensors3.conf
# Such custom configuration files for specific mainboards can be found at
# http://www.lm-sensors.org/wiki/Configurations
```

Sorry for the noise, and thanks for the link!

closed 2020-01-14 by Talkless

olysonek, 2020-01-14:
I see. BTW, currently the best place to find configuration files is the repository itself:
https://github.com/lm-sensors/lm-sensors/tree/master/configs