lm-sensors/lm-sensors #218: sensors showing Fan speed 0 RPM Gigabyte ga AB350 Gaming

wuast94, 2020-01-16:
i run sensors detect: 
`Now follows a summary of the probes I have just done.
Just press ENTER to continue: 

Driver `k10temp' (autoloaded):
  * Chip `AMD Family 17h thermal sensors' (confidence: 9)

Driver `it87':
  * ISA bus, address 0xa60
    Chip `ITE IT8792E Super IO Sensors' (confidence: 9)

Driver `to-be-written':
  * ISA bus, address 0xa40
    Chip `ITE IT8686E Super IO Sensors' (confidence: 9)`

senors -u:
`k10temp-pci-00c3
Adapter: PCI adapter
Tdie:
  temp1_input: 37.375
  temp1_max: 70.000
Tctl:
  temp2_input: 37.375

it8792-isa-0a60
Adapter: ISA adapter
in0:
  in0_input: 0.970
  in0_min: 0.000
  in0_max: 2.780
  in0_alarm: 0.000
in1:
  in1_input: 1.166
  in1_min: 0.000
  in1_max: 2.780
  in1_alarm: 0.000
in2:
  in2_input: 1.188
  in2_min: 0.000
  in2_max: 2.780
  in2_alarm: 0.000
+3.3V:
  in3_input: 1.679
  in3_min: 0.000
  in3_max: 2.780
  in3_alarm: 0.000
in4:
  in4_input: 1.352
  in4_min: 0.000
  in4_max: 2.780
  in4_alarm: 0.000
in5:
  in5_input: 1.275
  in5_min: 0.000
  in5_max: 2.780
  in5_alarm: 0.000
in6:
  in6_input: 2.780
  in6_min: 0.000
  in6_max: 2.780
  in6_alarm: 1.000
3VSB:
  in7_input: 1.668
  in7_min: 0.000
  in7_max: 2.780
  in7_alarm: 0.000
Vbat:
  in8_input: 1.548
fan1:
  fan1_input: 0.000
  fan1_min: 0.000
  fan1_alarm: 0.000
fan2:
  fan2_input: 0.000
  fan2_min: 0.000
  fan2_alarm: 0.000
fan3:
  fan3_input: 0.000
  fan3_min: 0.000
  fan3_alarm: 0.000
temp1:
  temp1_input: 26.000
  temp1_max: 127.000
  temp1_min: 127.000
  temp1_alarm: 0.000
  temp1_type: 4.000
  temp1_offset: 0.000
temp2:
  temp2_input: 12.000
  temp2_max: 127.000
  temp2_min: 127.000
  temp2_alarm: 0.000
  temp2_type: 4.000
  temp2_offset: 0.000
temp3:
  temp3_input: 27.000
  temp3_max: 127.000
  temp3_min: 127.000
  temp3_alarm: 0.000
  temp3_type: 4.000
  temp3_offset: 0.000
intrusion0:
  intrusion0_alarm: 1.000`

all fan speeds are at 0 but they are running.

jranke, 2020-07-23:
Running sensors-detect does not change your configuration. On my Debian system, I do not have an /etc/sensors directory, but an /etc/sensors.d directory. I have a link to conf/GA-AB350-GAMING.conf in /etc/sensors.d and sensors -u gives

```
it8792-isa-0a60
Adapter: ISA adapter
3VSB:
  in7_input: 3.335
  in7_min: 0.000
  in7_max: 5.559
  in7_alarm: 0.000
Vbat:
  in8_input: 3.357
temp1:
  temp1_input: 28.000
  temp1_max: 127.000
  temp1_min: 127.000
  temp1_alarm: 0.000
  temp1_type: 4.000
  temp1_offset: 0.000
temp2:
  temp2_input: 34.000
  temp2_max: 127.000
  temp2_min: 127.000
  temp2_alarm: 0.000
  temp2_type: 4.000
  temp2_offset: 0.000
temp3:
  temp3_input: 36.000
  temp3_max: 127.000
  temp3_min: 127.000
  temp3_alarm: 0.000
  temp3_type: 4.000
  temp3_offset: 0.000
intrusion0:
  intrusion0_alarm: 1.000

radeon-pci-0600
Adapter: PCI adapter
temp1:
  temp1_input: 62.500
  temp1_crit: 120.000
  temp1_crit_hyst: 90.000

k10temp-pci-00c3
Adapter: PCI adapter
Tdie:
  temp1_input: 41.000
  temp1_max: 70.000
Tctl:
  temp2_input: 41.000

it8686-isa-0a40
Adapter: ISA adapter
CPU Vcore:
  in0_input: 0.828
  in0_min: 0.000
  in0_max: 3.060
  in0_alarm: 0.000
  in0_beep: 0.000
DRAM Channel A/B:
  in6_input: 1.212
  in6_min: 0.000
  in6_max: 3.060
  in6_alarm: 0.000
  in6_beep: 0.000
3VSB:
  in7_input: 3.264
  in7_min: 0.000
  in7_max: 6.120
  in7_alarm: 0.000
  in7_beep: 0.000
Vbat:
  in8_input: 3.264
CPU_FAN:
  fan1_input: 901.000
  fan1_min: 0.000
  fan1_alarm: 0.000
  fan1_beep: 1.000
SYS_FAN1:
  fan2_input: 639.000
  fan2_min: 10.000
  fan2_alarm: 0.000
  fan2_beep: 1.000
SYS_FAN2:
  fan3_input: 523.000
  fan3_min: 10.000
  fan3_alarm: 0.000
  fan3_beep: 1.000
SYS_FAN3_PUMP:
  fan4_input: 0.000
  fan4_min: 0.000
  fan4_alarm: 0.000
  fan4_beep: 1.000
System 1:
  temp1_input: 38.000
  temp1_max: 127.000
  temp1_min: 127.000
  temp1_alarm: 0.000
  temp1_type: 4.000
  temp1_offset: 0.000
  temp1_beep: 1.000
Chipset Temp:
  temp2_input: 51.000
  temp2_max: 127.000
  temp2_min: 127.000
  temp2_alarm: 0.000
  temp2_type: 4.000
  temp2_offset: 0.000
  temp2_beep: 1.000
CPU Temp:
  temp3_input: 46.000
  temp3_max: 85.000
  temp3_min: 15.000
  temp3_alarm: 0.000
  temp3_type: 5.000
  temp3_offset: -10.000
  temp3_beep: 1.000
PCI-EX16 Temp:
  temp4_input: 40.000
  temp4_max: 127.000
  temp4_min: 127.000
  temp4_alarm: 0.000
  temp4_type: 4.000
  temp4_offset: 0.000
  temp4_beep: 1.000
VRM MOS Temp:
  temp5_input: 44.000
  temp5_max: -122.000
  temp5_min: 0.000
  temp5_alarm: 0.000
  temp5_type: 4.000
  temp5_offset: 0.000
  temp5_beep: 1.000
vSOC MOS Temp:
  temp6_input: 52.000
  temp6_max: -122.000
  temp6_min: 0.000
  temp6_alarm: 0.000
  temp6_type: 4.000
  temp6_offset: 0.000
  temp6_beep: 1.000
intrusion0:
  intrusion0_alarm: 1.000
```

Of course you will probably not have the `radeon-pci-0600` component, as this is provided by my graphics card.

duckty, 2022-11-16:
@wuast94 : I got the same issue like you. Have you check the lines after this ?
```Driver to-be-written:
  * ISA bus, address 0xa40
    Chip `ITE IT8686E Super IO Sensors' (confidence: 9)
Note: there is no driver for ITE IT8686E Super IO Sensors yet.
Check https://hwmon.wiki.kernel.org/device_support_status for updates.

No modules to load, skipping modules configuration.

Unloading cpuid... OK
```
You need to load ITE IT8686E. Follow the instruction [here](https://wiki.tnonline.net/w/Blog/IT87_driver_for_the_it8686_sensor_chip) to compile, install and load the module. You will be able to see the CPU fan speed

This is the sensors' output from my computer
```
# sensors
it8686-isa-0a40
Adapter: ISA adapter
Vcore:       900.00 mV (min =  +0.00 V, max =  +3.06 V)
+3.3V:         3.35 V  (min =  +0.00 V, max =  +5.05 V)
+12.0V:       12.02 V  (min =  +0.00 V, max = +18.36 V)
+5.0V:         5.04 V  (min =  +0.00 V, max =  +7.65 V)
VSOC:          1.01 V  (min =  +0.00 V, max =  +3.06 V)
VDDP:        900.00 mV (min =  +0.00 V, max =  +3.06 V)
DRAM:          1.22 V  (min =  +0.00 V, max =  +3.06 V)
3VSB:          3.31 V  (min =  +0.00 V, max =  +6.12 V)
VBAT:          3.22 V  
CPU Fan:     1956 RPM  (min =   10 RPM)
SYS Fan 1:      0 RPM  (min =    0 RPM)
System:       +41.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
Chipset:      +52.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
CPU:          +61.0°C  (low  =  +0.0°C, high = +90.0°C)  sensor = AMD AMDSI
PCIe x16:     +17.0°C  (low  =  +0.0°C, high = +127.0°C)  sensor = thermistor
VRM:          +45.0°C  (low  =  +0.0°C, high = -113.0°C)  sensor = thermistor
VSoC:         +46.0°C  (low  =  +0.0°C, high = -113.0°C)  sensor = thermistor
Intrusion:   ALARM

```

tomchiverton, 2026-02-16:
Confirm same issue and fix. Could we ship it ?