lm-sensors/lm-sensors #219: Unsupported SMSC SCH5317 (id=0x85, rev=0x87)

typoworx-de, 2020-01-17:
I found an unsupported sensor-chip:

> Trying family `National Semiconductor/ITE'...               Yes
> Found unknown chip with ID 0x8587
> Probing for Super-I/O at 0x4e/0x4f

$ sudo superiotool
> superiotool r6637
> Found SMSC SCH5317 (id=0x85, rev=0x87) at 0x2e

(https://www.mouser.de/ProductDetail/Microchip-Technology/SCH5317-NS/?qs=iGJE2gFRJMOFcJOmxm1xjw==)

May there be any existing chip which may cover this one? I already found these hints searching for the chip-name, that it may be covered by existing sensor-chip implementations:
https://narkive.com/14dqdnxi.4

typoworx-de, 2020-01-17:
Is it somehow possible to map the given hardware-id to the module 'smsc47b397' (mentioned in the link given above as compatible device)?

LicoMonch, 2020-10-08:
> Is it somehow possible to map the given hardware-id to the module 'smsc47b397' (mentioned in the link given above as compatible device)?

I'm not sure this would work. For me the driver smsc47b397 fails to init the device .. I'm no expert for ISA/LPC devices and a bit lost here:

```
~$ dmesg | grep -i smsc
~$ lsmod | grep smsc
~$ modprobe smsc47b397
~$ lsmod | grep smsc
smsc47b397             20480  0
~$ dmesg | grep -i smsc
[  258.265981] smsc47b397: found SMSC SCH5317 (base address 0x0000, revision 135)
[  258.266048] smsc47b397 smsc47b397.0: Region 0x0-0x1 already in use!
[  258.266050] smsc47b397: probe of smsc47b397.0 failed with error -16
```
In the latest sensors-detect there is already a part about vendorID 85 (see [here](https://github.com/lm-sensors/lm-sensors/blob/master/prog/detect/sensors-detect#L2080-L2094)). My hope was, that with a successful loading of smsc47b397 all would work out of the box .. but . meh -.-'

LicoMonch, 2020-10-08:
I"ve found out, why it wasn't detected:
```
Some Super I/O chips contain embedded sensors. We have to write to
standard I/O ports to probe them. This is usually safe.
Do you want to scan for Super I/O sensors? (YES/no): 
Probing for Super-I/O at 0x2e/0x2f
Trying family `National Semiconductor/ITE'...               Yes
Found unknown chip with ID 0x8587
Probing for Super-I/O at 0x4e/0x4f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               No
```
Here it simply skips the SMSC detection, because it found an 'National Semiconductor/ITE' chip.

I don't know why the probe for 'National Semiconductor/ITE' got the SMSC-Chip, but if you remove the whole 'National Semiconductor/ITE' group and skip it that way, you'll get the propper detection:
```
Some Super I/O chips contain embedded sensors. We have to write to
standard I/O ports to probe them. This is usually safe.
Do you want to scan for Super I/O sensors? (YES/no): 
Probing for Super-I/O at 0x2e/0x2f
Trying family `SMSC'...                                     Yes
Found `SMSC SCH5317 Super IO'                               
    (but not activated)
Probing for Super-I/O at 0x4e/0x4f
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               No

```

That should be the main problem of this issue, at least it seems so to me ;)

And now I need to figure out, how to load that damn device. I bet thats some kind of IRQ sharing problem .. and that was already a pain 20 years ago :(

memfissoftware, 2024-10-19:
@LicoMonch 
then were you able to activate SCH5317?