lm-sensors/lm-sensors #220: No CPU FAN sensor monitoring for ASUS ROG STRIX X570-I MB

NTMan, 2020-01-19:
```
# sensors -u
iwlwifi-virtual-0
Adapter: Virtual device
temp1:
ERROR: Can't get value of subfeature temp1_input: Can't read

k10temp-pci-00c3
Adapter: PCI adapter
Tdie:
  temp1_input: 72.625
  temp1_max: 70.000
Tctl:
  temp2_input: 72.625

nvme-pci-0100
Adapter: PCI adapter
Composite:
  temp1_input: 40.850
  temp1_alarm: 0.000

amdgpu-pci-0b00
Adapter: PCI adapter
vddgfx:
  in0_input: 0.731
fan1:
  fan1_input: 908.000
  fan1_min: 0.000
  fan1_max: 3850.000
edge:
  temp1_input: 48.000
  temp1_crit: 100.000
  temp1_crit_hyst: -273.150
  temp1_emergency: 105.000
junction:
  temp2_input: 50.000
  temp2_crit: 110.000
  temp2_crit_hyst: -273.150
  temp2_emergency: 115.000
mem:
  temp3_input: 50.000
  temp3_crit: 94.000
  temp3_crit_hyst: -273.150
  temp3_emergency: 99.000
power1:
  power1_average: 42.000
  power1_cap: 250.000
```

<details><summary>sensors-detect says that this MB uses Nuvoton NCT6798D-R super I/O chip</summary>

```
# sensors-detect
# sensors-detect version 3.6.0
# Board: ASUSTeK COMPUTER INC. ROG STRIX X570-I GAMING
# Kernel: 5.5.0-0.rc6.git3.2.fc32.x86_64 x86_64
# Processor: AMD Ryzen 9 3950X 16-Core Processor (23/113/0)

This program will help you determine which kernel modules you need
to load to use lm_sensors most effectively. It is generally safe
and recommended to accept the default answers to all questions,
unless you know what you're doing.

Some south bridges, CPUs or memory controllers contain embedded sensors.
Do you want to scan for them? This is totally safe. (YES/no): YES
Silicon Integrated Systems SIS5595...                       No
VIA VT82C686 Integrated Sensors...                          No
VIA VT8231 Integrated Sensors...                            No
AMD K8 thermal sensors...                                   No
AMD Family 10h thermal sensors...                           No
AMD Family 11h thermal sensors...                           No
AMD Family 12h and 14h thermal sensors...                   No
AMD Family 15h thermal sensors...                           No
AMD Family 16h thermal sensors...                           No
AMD Family 17h thermal sensors...                           Success!
    (driver `k10temp')
AMD Family 15h power sensors...                             No
AMD Family 16h power sensors...                             No
Hygon Family 18h thermal sensors...                         No
Intel digital thermal sensor...                             No
Intel AMB FB-DIMM thermal sensor...                         No
Intel 5500/5520/X58 thermal sensor...                       No
VIA C7 thermal sensor...                                    No
VIA Nano thermal sensor...                                  No

Some Super I/O chips contain embedded sensors. We have to write to
standard I/O ports to probe them. This is usually safe.
Do you want to scan for Super I/O sensors? (YES/no): YES
Probing for Super-I/O at 0x2e/0x2f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               Yes
Found `Nuvoton NCT6798D Super IO Sensors'                   Success!
    (address 0x290, driver `nct6775')
Probing for Super-I/O at 0x4e/0x4f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               No
Trying family `ITE'...                                      No

Some systems (mainly servers) implement IPMI, a set of common interfaces
through which system health data may be retrieved, amongst other things.
We first try to get the information from SMBIOS. If we don't find it
there, we have to read from arbitrary I/O ports to probe for such
interfaces. This is normally safe. Do you want to scan for IPMI
interfaces? (YES/no): YES
Probing for `IPMI BMC KCS' at 0xca0...                      No
Probing for `IPMI BMC SMIC' at 0xca8...                     No

Some hardware monitoring chips are accessible through the ISA I/O ports.
We have to write to arbitrary I/O ports to probe them. This is usually
safe though. Yes, you do have ISA I/O ports even if you do not have any
ISA slots! Do you want to scan the ISA I/O ports? (yes/NO): NO

Lastly, we can probe the I2C/SMBus adapters for connected hardware
monitoring devices. This is the most risky part, and while it works
reasonably well on most systems, it has been reported to cause trouble
on some systems.
Do you want to probe the I2C/SMBus adapters now? (YES/no): YES
Using driver `i2c-piix4' for device 0000:00:14.0: AMD KERNCZ SMBus
Module i2c-dev loaded successfully.

Next adapter: AMDGPU DM i2c hw bus 0 (i2c-0)
Do you want to scan it? (yes/NO/selectively): NO

Next adapter: dmdc (i2c-1)
Do you want to scan it? (yes/NO/selectively): NO

Next adapter: AMDGPU DM i2c hw bus 1 (i2c-2)
Do you want to scan it? (yes/NO/selectively): NO

Next adapter: dmdc (i2c-3)
Do you want to scan it? (yes/NO/selectively): NO

Next adapter: AMDGPU DM i2c hw bus 2 (i2c-4)
Do you want to scan it? (yes/NO/selectively): NO

Next adapter: dmdc (i2c-5)
Do you want to scan it? (yes/NO/selectively): NO

Next adapter: AMDGPU DM i2c hw bus 3 (i2c-6)
Do you want to scan it? (yes/NO/selectively): NO

Next adapter: SMBus PIIX4 adapter port 0 at 0b00 (i2c-7)
Do you want to scan it? (YES/no/selectively): YES
Client found at address 0x50
Probing for `Analog Devices ADM1033'...                     No
Probing for `Analog Devices ADM1034'...                     No
Probing for `SPD EEPROM'...                                 Yes
    (confidence 8, not a hardware monitoring chip)
Probing for `EDID EEPROM'...                                No
Client found at address 0x52
Probing for `Analog Devices ADM1033'...                     No
Probing for `Analog Devices ADM1034'...                     No
Probing for `SPD EEPROM'...                                 Yes
    (confidence 8, not a hardware monitoring chip)
Client found at address 0x77
Probing for `Asus Mozart-2'...                              No

Next adapter: SMBus PIIX4 adapter port 2 at 0b00 (i2c-8)
Do you want to scan it? (YES/no/selectively): YES

Now follows a summary of the probes I have just done.
Just press ENTER to continue: 

Driver `k10temp' (autoloaded):
  * Chip `AMD Family 17h thermal sensors' (confidence: 9)

Driver `nct6775':
  * ISA bus, address 0x290
    Chip `Nuvoton NCT6798D Super IO Sensors' (confidence: 9)

Do you want to overwrite /etc/sysconfig/lm_sensors? (YES/no): YES
Unloading i2c-dev... OK
```
</details>

Motherboard photo confirms this:
[image]

risyasin, 2020-01-22:
I have the same problem as well 
```
[yas@dxarch ~]$ sudo dmidecode -t 2
# dmidecode 3.2
Getting SMBIOS data from sysfs.
SMBIOS 3.2.0 present.

Handle 0x0002, DMI type 2, 15 bytes
Base Board Information
	Manufacturer: ASUSTeK COMPUTER INC.
	Product Name: ROG STRIX X570-I GAMING
	Version: Rev X.0x
	Serial Number: 190957676100539
	Asset Tag: Default string
	Features:
		Board is a hosting board
		Board is replaceable
	Location In Chassis: Default string
	Chassis Handle: 0x0003
	Type: Motherboard
	Contained Object Handles: 0

[yas@dxarch ~]$ sudo dmidecode -t 34
# dmidecode 3.2
Getting SMBIOS data from sysfs.
SMBIOS 3.2.0 present.

Handle 0x0023, DMI type 34, 11 bytes
Management Device
	Description: NCT6798D-R
	Type: Other
	Address: 0x00000295
	Address Type: I/O Port

[yas@dxarch ~]$ sensors
sensors               sensors-conf-convert  sensors-detect        
[yas@dxarch ~]$ sensors
iwlwifi-virtual-0
Adapter: Virtual device
temp1:        +41.0°C  

amdgpu-pci-0b00
Adapter: PCI adapter
vddgfx:      775.00 mV 
fan1:           0 RPM  (min =    0 RPM, max = 3200 RPM)
edge:         +47.0°C  (crit = +118.0°C, hyst = -273.1°C)
                       (emerg = +99.0°C)
junction:     +47.0°C  (crit = +99.0°C, hyst = -273.1°C)
                       (emerg = +99.0°C)
mem:          +48.0°C  (crit = +99.0°C, hyst = -273.1°C)
                       (emerg = +99.0°C)
power1:        8.00 W  (cap = 165.00 W)

```
Sensor3.conf
```
# libsensors configuration file
# -----------------------------
#
# This default configuration file only includes statements which do not
# differ from one mainboard to the next. Only label, compute and set
# statements for internal voltage and temperature sensors are included.
#
# In general, local changes should not be added to this file, but rather
# placed in custom configuration files located in /etc/sensors.d. This
# approach makes further updates much easier.
#
# Such custom configuration files for specific mainboards can be found in
# "configs" directory of lm-sensors package.
#
[... 467 lines cut ...]

    set in2_min  3.3 * 0.90
    set in2_max  3.3 * 1.10

    label temp2 "Board Temp"

chip "adt7476-*" "adt7490-*"

    label in1 "Vcore"
    label in2 "+3.3V"
    label in3 "+5V"
    label in4 "+12V"

    set in2_min  3.3 * 0.90
    set in2_max  3.3 * 1.10
    set in3_min  5.0 * 0.90
    set in3_max  5.0 * 1.10
# Depending on how your ADT7476 is hardwired, you may or may not have
# +12V readings.
#    set in4_min 12.0 * 0.90
#    set in4_max 12.0 * 1.10

    label temp2 "M/B Temp"
```
I've tried both k10temp and zenpower mods. No luck so far.

risyasin, 2020-01-22:
Update: 
I've installed new driver for Nuvoton NCT6798D-R [https://github.com/tonylambiris/nct6775](https://github.com/tonylambiris/nct6775)

Now I can see more data provided to lm_sensors via new `nct6775` but no fan speeds :(  
```
[yas@dxarch ~]$ sensors 
iwlwifi-virtual-0
Adapter: Virtual device
temp1:        +41.0°C  

zenpower-pci-00c3
Adapter: PCI adapter
SVI2_Core:     1.46 V  
SVI2_SoC:      1.09 V  
Tdie:         +46.4°C  (high = +70.0°C)
Tctl:         +46.4°C  
Tccd1:        +41.0°C  
SVI2_P_Core:   7.71 W  
SVI2_P_SoC:    9.93 W  
SVI2_C_Core:   8.56 A  
SVI2_C_SoC:    9.12 A  

amdgpu-pci-0b00
Adapter: PCI adapter
vddgfx:      775.00 mV 
fan1:           0 RPM  (min =    0 RPM, max = 3200 RPM)
edge:         +46.0°C  (crit = +118.0°C, hyst = -273.1°C)
                       (emerg = +99.0°C)
junction:     +46.0°C  (crit = +99.0°C, hyst = -273.1°C)
                       (emerg = +99.0°C)
mem:          +48.0°C  (crit = +99.0°C, hyst = -273.1°C)
                       (emerg = +99.0°C)
power1:        8.00 W  (cap = 165.00 W)
```

berniyh, 2020-04-01:
Same thing here for an ASUS ROG STRIX X570-E.
Has the same nuvoton NCT6798D-R chip on it.

berniyh, 2020-04-01:
Just noticed that there is already an issue about that, see:
https://github.com/lm-sensors/lm-sensors/issues/197

With said boot parameter acpi_enforce_resources=lax the sensor monitoring and also fan monitoring does work for me now.

lbibass, 2020-07-27:
Hey, I still can't get this to work. I'm new to linux, but I've put "acpi_enforce_resources=lax" into the kernel parameters after "quiet spash" (no quotes). It can see the sensor, but it never shows me anything from it. Here's the output from sensors. 

```evan@SM560:~$ sensors
iwlwifi_1-virtual-0
Adapter: Virtual device
temp1:        +56.0°C  

k10temp-pci-00c3
Adapter: PCI adapter
Tdie:         +78.0°C  (high = +70.0°C)
Tctl:         +78.0°C  

amdgpu-pci-0b00
Adapter: PCI adapter
vddgfx:      725.00 mV 
fan1:         787 RPM  (min =    0 RPM, max = 3200 RPM)
edge:         +46.0°C  (crit = +118.0°C, hyst = -273.1°C)
                       (emerg = +99.0°C)
junction:     +51.0°C  (crit = +99.0°C, hyst = -273.1°C)
                       (emerg = +99.0°C)
mem:          +62.0°C  (crit = +99.0°C, hyst = -273.1°C)
                       (emerg = +99.0°C)
power1:       30.00 W  (cap = 180.00 W)
```

lbibass, 2020-07-27:
Sorry: I ended up getting it to work. I just needed to type in 

```sudo modprobe nct6775``` 

into the terminal. Now it's reading from the sensor. Hope this helps anyone else who had the same problem.

Darxus, 2020-10-13:
Just wanted to confirm I had a similar experience.  What I needed to do was run sensors-detect, and allow it to add nct6775 to /etc/modules, then change /etc/default/grub to include:
GRUB_CMDLINE_LINUX_DEFAULT="quiet splash acpi_enforce_resources=lax"
Then I ran update-grub.  

Before, I had video and CPU temps, but no fans.  Now I also have 7 fans.  On my Asus Tuf Gaming X570-Plus motherboard.  Ubuntu 20.04, kernel 5.4.0-48-generic.

regulator-g, 2021-05-15:
Hey guys

I have the same issue on my b550 motherboard

On the newer kernel(s) I was able to get sensors using the steps suggested. No luck on 5.4 LTS kernel though.

will these sensors eventually become native in the kernel and not require extra steps to function?

I had a lot of issues with the built in Ethernet and WiFi too but again these are mostly resolved in newer kernels

sudo dmidecode -t 2
# dmidecode 3.3
Getting SMBIOS data from sysfs.
SMBIOS 3.3.0 present.

Handle 0x0002, DMI type 2, 15 bytes
Base Board Information
	Manufacturer: ASUSTeK COMPUTER INC.
	Product Name: ROG STRIX B550-F GAMING (WI-FI)
	Version: Rev X.0x
	Serial Number: 
	Asset Tag: Default string
	Features:
		Board is a hosting board
		Board is replaceable
	Location In Chassis: Default string
	Chassis Handle: 0x0003
	Type: Motherboard
	Contained Object Handles: 0

peacepenguin, 2021-09-08:
This issue is related to an upstream kernel driver bug in the nct6775 driver. The nct6775 driver/kernel module is used for the chip many asus boards have: nct6798. Sensors (lm_sensors) uses the nct6775 driver to query the fan, temp, and voltage info on these asus motherboards.

https://bugzilla.kernel.org/show_bug.cgi?id=204807

There's a patch being actively worked on, and works great if you get it compiled. So far these boards have been reported as affected and fixed by the patch:

ROG CROSSHAIR VIII IMPACT
ROG STRIX B550-E GAMING
ROG STRIX B550-F GAMING (WI-FI)
ROG STRIX Z490-I GAMING
TUF GAMING B550M-PLUS
TUF GAMING B550M-PLUS (WI-FI)
TUF GAMING B550-PLUS
TUF GAMING X570-PLUS
TUF GAMING X570-PRO (WI-FI)

The patch is available for testing at the bugzilla link above. So far the patch works great on my Fedora 34 system, with a TUF GAMING X570-PLUS motherboard. Some more clean-up work is needed to get the patch committed to the kernel tree, so anyone with Kernel driver skills please check out the bugzilla link in this post to see if you can help out in that effort.

For now though, the workaround mentioned by @Darxus works fine, but it is a "risk" to system stability. The new patch removes the need for "lax", and instead uses WMI to get the sensor data instead of direct ACPI calls.

> run sensors-detect, and allow it to add nct6775 to /etc/modules, then change /etc/default/grub to include:
GRUB_CMDLINE_LINUX_DEFAULT="quiet splash acpi_enforce_resources=lax"
Then I ran update-grub.

NoahDar, 2021-09-27:
> This issue is related to an upstream kernel driver bug in the nct6775 driver. The nct6775 driver/kernel module is used for the chip many asus boards have: nct6798. Sensors (lm_sensors) uses the [...]

I can tell you this works perfectly fine with my ASUS ROG STRIX X570-E GAMING WIFI running on Debian 11 KDE with Debian 5.10.46-5 (2021-09-23) x86_64 GNU/Linux kernel.  AMD Ryzen 9 5950X

What is the system stability you're referring to?  I've read somewhere using WMI on this motherboard may cause the fans to stop working altogether.  Also been suggested not to leave the computer alone?    Can WMI just set as read-only since that's all we care about?

laurensbl, 2021-10-04:
Getting no data from sensors (doesn't show in output) on:  TUF H370-PRO GAMING, so TUF H370-PRO GAMING seems also affected.

It does detect it though:
```
sudo sensors-detect
...
Some Super I/O chips contain embedded sensors. We have to write to
standard I/O ports to probe them. This is usually safe.
Do you want to scan for Super I/O sensors? (YES/no):
Probing for Super-I/O at 0x2e/0x2f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               Yes
Found `Nuvoton NCT6796D Super IO Sensors'                   Success!
    (address 0x290, driver `nct6775')
Probing for Super-I/O at 0x4e/0x4f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               No
Trying family `ITE'...                                      No
...
```

```
sudo dmidecode -t 2
# dmidecode 3.3
Getting SMBIOS data from sysfs.
SMBIOS 3.2.1 present.

Handle 0x0002, DMI type 2, 15 bytes
Base Board Information
        Manufacturer: ASUSTeK COMPUTER INC.
        Product Name: TUF H370-PRO GAMING
        Version: Rev 1.xx
        Serial Number: 180221078400849
        Asset Tag: Default string
        Features:
                Board is a hosting board
                Board is replaceable
        Location In Chassis: Default string
        Chassis Handle: 0x0003
        Type: Motherboard
        Contained Object Handles: 0
```

```
sudo dmidecode -t 34
# dmidecode 3.3
Getting SMBIOS data from sysfs.
SMBIOS 3.2.1 present.

Handle 0x002B, DMI type 34, 11 bytes
Management Device
        Description: Nuvoton NCT5582D
        Type: Other
        Address: 0x00000295
        Address Type: I/O Port
```

```
lsmod | grep nct6775
nct6775                69632  0
hwmon_vid              16384  1 nct6775
```

winklevos, 2021-10-25:
> Just a reminder, this issue is caused by an upstream kernel hwmon driver problem with the nct6775 driver that the nct6798 devices use. Nothing to fix on the lm_sensors side.
> 
> Patch for a variety of asus mobos is almost ready to be merged into the linux kernel, status and the patches being merged can be found here:
> 
> https://bugzilla.kernel.org/show_bug.cgi?id=204807
> 
> If you're up for it, you can implement the fixes now by pulling in the patches and compiling the nct6775 driver yourself, usually using dkms. Working well for me on a tuf gaming x570 plus by manually compiling the patches.
> 
> Good luck!

Yeah I found the upstream bug shortly after, though it looks like the solution is a specific work around for a list of motherboards. And to have a motherboard added to the list requires someone to compile and test on that motherboard. 

I'm not across linux compiling so they won't add my MB

GregDuhamel, 2021-12-05:
@NTMan Should work with Paul's patch, can you confirm ? You need to use Kernel 5.16. It's working on mine (Asus ROG Crosshair VIII Impact).

NTMan, 2021-12-05:
> @NTMan Should work with Denis Pauk's [patch](https://bugzilla.kernel.org/show_bug.cgi?id=204807), can you confirm ? You need to use Kernel 5.16. It's working on mine (Asus ROG Crosshair VIII Impact), if your card is not working / listed it may be useful to just request it on bugzilla.

Unfortunately I didn't find any changes in output of sensors app after adding Denis Pauk's patch.
[image]

Maybe because my motherboard "ROG STRIX X570-I GAMING" not listed here.
[image]

VeniceNerd, 2022-07-26:
Is there any update to this? I own the ROG STRIX Z490-I GAMING (which I see is one of the affected boards) and while I do see my CPU temperature in Unraid, I do not see my motherboard temperature:
[image]

Any help would be greatly appreciated!

shunlir, 2022-08-05:
Same problem.

As a workaround, the issue is resolved using https://github.com/tonylambiris/nct6775 , and `acpi_enforce_resources=lax` is not needed.

```
# sensors
k10temp-pci-00c3
Adapter: PCI adapter
Tctl:         +36.6°C  
Tccd1:        +26.5°C  
Tccd2:        +27.5°C  

nct6798-isa-0290
Adapter: ISA adapter
in0:                      160.00 mV (min =  +0.00 V, max =  +1.74 V)
in1:                        1.01 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in2:                        3.39 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in3:                        3.34 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in4:                        1.01 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in5:                      840.00 mV (min =  +0.00 V, max =  +0.00 V)  ALARM
in6:                      240.00 mV (min =  +0.00 V, max =  +0.00 V)  ALARM
in7:                        3.39 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in8:                        3.34 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in9:                      904.00 mV (min =  +0.00 V, max =  +0.00 V)  ALARM
in10:                     416.00 mV (min =  +0.00 V, max =  +0.00 V)  ALARM
in11:                     416.00 mV (min =  +0.00 V, max =  +0.00 V)  ALARM
in12:                       1.04 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in13:                     1000.00 mV (min =  +0.00 V, max =  +0.00 V)  ALARM
in14:                       1.02 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
fan1:                        0 RPM  (min =    0 RPM)
fan2:                      774 RPM  (min =    0 RPM)
fan3:                        0 RPM  (min =    0 RPM)
fan7:                      804 RPM  (min =    0 RPM)
SYSTIN:                    +35.0°C  (high = +80.0°C, hyst = +75.0°C)  sensor = thermistor
CPUTIN:                    +29.5°C  (high = +80.0°C, hyst = +75.0°C)  sensor = thermistor
AUXTIN0:                   +87.5°C    sensor = thermistor
AUXTIN1:                   +35.0°C    sensor = thermistor
AUXTIN2:                   +35.0°C    sensor = thermistor
AUXTIN3:                   +25.0°C    sensor = thermistor
PECI Agent 0 Calibration:  +30.0°C  
PCH_CHIP_CPU_MAX_TEMP:      +0.0°C  
PCH_CHIP_TEMP:              +0.0°C  
PCH_CPU_TEMP:               +0.0°C  
TSI0_TEMP:                 +37.0°C  
intrusion0:               ALARM
intrusion1:               ALARM
beep_enable:              disabled
```

I'm using:
- `TUF GAMING B550M-PLUS (WI-FI)` which has `NCT6798D-R`
- kernel version `5.15.52-gentoo` where the `nct6775` driver has `nct6798d` support (but not work)

tudo75, 2022-08-11:
Hi, 
I have an `ASUS PRIME Z690-P WIFI` with kernel `5.15.0-46-generic` and the default `nct6775` driver and adding the `acpi_enforce_resources=lax` to the boot parameters solved the problem

```
nick   SIGMUNDX  ~  sensors
iwlwifi_2-virtual-0
Adapter: Virtual device
temp1:        +30.0°C  

coretemp-isa-0000
Adapter: ISA adapter
Package id 0:  +30.0°C  (high = +80.0°C, crit = +100.0°C)
Core 0:        +24.0°C  (high = +80.0°C, crit = +100.0°C)
Core 4:        +28.0°C  (high = +80.0°C, crit = +100.0°C)
Core 8:        +29.0°C  (high = +80.0°C, crit = +100.0°C)
Core 12:       +28.0°C  (high = +80.0°C, crit = +100.0°C)
Core 16:       +27.0°C  (high = +80.0°C, crit = +100.0°C)
Core 20:       +27.0°C  (high = +80.0°C, crit = +100.0°C)
Core 28:       +30.0°C  (high = +80.0°C, crit = +100.0°C)
Core 29:       +30.0°C  (high = +80.0°C, crit = +100.0°C)
Core 30:       +30.0°C  (high = +80.0°C, crit = +100.0°C)
Core 31:       +30.0°C  (high = +80.0°C, crit = +100.0°C)

acpitz-acpi-0
Adapter: ACPI interface
temp1:        +27.8°C  (crit = +105.0°C)

nct6798-isa-0290
Adapter: ISA adapter
in0:                      496.00 mV (min =  +0.00 V, max =  +1.74 V)
in1:                      1000.00 mV (min =  +0.00 V, max =  +0.00 V)  ALARM
in2:                        3.39 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in3:                        3.34 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in4:                        1.02 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in5:                        8.00 mV (min =  +0.00 V, max =  +0.00 V)  ALARM
in6:                      880.00 mV (min =  +0.00 V, max =  +0.00 V)  ALARM
in7:                        3.39 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in8:                        3.22 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in9:                      536.00 mV (min =  +0.00 V, max =  +0.00 V)  ALARM
in10:                     560.00 mV (min =  +0.00 V, max =  +0.00 V)  ALARM
in11:                       0.00 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in12:                       1.06 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in13:                     520.00 mV (min =  +0.00 V, max =  +0.00 V)  ALARM
in14:                     904.00 mV (min =  +0.00 V, max =  +0.00 V)  ALARM
fan1:                      817 RPM  (min =    0 RPM)
fan2:                      691 RPM  (min =    0 RPM)
fan3:                        0 RPM  (min =    0 RPM)
fan4:                      720 RPM  (min =    0 RPM)
fan5:                      574 RPM  (min =    0 RPM)
fan6:                        0 RPM  (min =    0 RPM)
SYSTIN:                    +31.0°C  (high = +80.0°C, hyst = +75.0°C)  sensor = thermistor
CPUTIN:                    +33.0°C  (high = +80.0°C, hyst = +75.0°C)  sensor = thermistor
AUXTIN0:                   +32.5°C    sensor = thermistor
AUXTIN1:                   +19.0°C    sensor = thermistor
AUXTIN2:                  +127.0°C    sensor = thermistor
AUXTIN3:                   +31.0°C    sensor = thermistor
PECI Agent 0 Calibration:  +33.5°C  
PCH_CHIP_CPU_MAX_TEMP:      +0.0°C  
PCH_CHIP_TEMP:             +59.0°C  
PCH_CPU_TEMP:               +0.0°C  
intrusion0:               ALARM
intrusion1:               ALARM
beep_enable:              disabled

nvme-pci-0100
Adapter: PCI adapter
Composite:    +34.9°C  (low  = -273.1°C, high = +81.8°C)
                       (crit = +84.8°C)
Sensor 1:     +34.9°C  (low  = -273.1°C, high = +65261.8°C)
Sensor 2:     +36.9°C  (low  = -273.1°C, high = +65261.8°C)
```

isFakeAccount, 2023-02-07:
I have the same problem ob MSI Pro Z690-A. What it would take to add the support?

isFakeAccount, 2023-02-10:
This is linux kernel issue rather than lm-sensor issue. You can add support by following instructions here

ricciolino, 2023-03-08:
I also can confirm that [this](https://github.com/Fred78290/nct6687d) kernel module (linked by @isFakeAccount) supports the **nuvoTon 3961S** chipsets installed on the MAG Z690 TOMAHAWK WIFI DDR4 (MS-7D32) motherboard.
After the installation (make sure to load the kernel module at boot time, adding it to `/etc/modules`) and rebooting the PC, the system is able to correctly read system fan sensors.

berniyh, 2023-03-12:
> I also can confirm that [this](https://github.com/Fred78290/nct6687d) kernel module (linked by @isFakeAccount) supports the **nuvoTon 3961S** chipsets installed on the MAG Z690 TOMAHAWK WIFI DDR4 [...]

Just a minor thing: better add a .conf file containing the module name in /etc/modules-load.d instead of using /etc/modules.
The latter is deprecated and does not exist on all distributions anymore. 
iirc it's mainly Debian (and possibly derivatives like Ubuntu, Mint etc.) that still use this.

GuannanHu, 2023-07-25:
> 

As pointed out in the website mentioned in the quoted relay, Linux kernel 6.4 has patched the driver for nct6798d. Now the fan info is detected.

GuannanHu, 2023-07-25:
As pointed out in the website mentioned in the quoted reply, Linux kernel 6.4 has patched the driver for nct6798d. Now the fan info is detected.

> Just a reminder, this issue is caused by an upstream kernel hwmon driver problem with the nct6775 driver that the nct6798 devices use. Nothing to fix on the lm_sensors side.
>
> Patch for a variety of [...]

bilogic, 2023-12-02:
I'm using the `TUF Gaming B550M-Plus (Wi-Fi)` and managed to compile https://github.com/Fred78290/nct6687d, but running `modprobe nct6678` gave me `modprobe: ERROR: could not insert 'nct6687': No such device`

Any idea what I'm missing?

tkuebler, 2024-10-30:
the grub fix worked for me too - Ubuntu 22.x LTS,  X99-E WS Asus board