lm-sensors/lm-sensors #227: fancontrol: use sensors' stable names

zdzichu, 2020-03-12:
Currently, `fancontrol` operates directly on sysfs files. The number of hwmon drivers increases. For example, my computer currently provides hwmon readings from acpitz, it87, coretemp, nvme_hwmon and drivetemp. Mapping between sensors and hwmonX names often changes during boot.

`sensors` command helps by providing stable names for specific sensors. It would be nice to be able to use those stable names in `fancontrol`

Darxus, 2020-11-15:
I agree this would be nice.  Those names are contained in /sys/class/hwmon/hwmon*/name

Would a patch be accepted?  Any thoughts on how this should be accomplished?  Reading the existing /etc/fancontrol format would need to continue to be supported.  A less tidy way of dealing with this would be to leave that file format as is, and use the DEVNAME values to seek out the correct hwmon device numbers, ignoring the configured device numbers.  But it would probably be better to just replace the "hwmon1" type values with the names?

My existing /etc/fancontrol looks like this:

# Configuration file generated by pwmconfig, changes will be lost
INTERVAL=10
DEVPATH=hwmon1=devices/platform/nct6775.656 hwmon2=devices/pci0000:00/0000:00:18.3
DEVNAME=hwmon1=nct6798 hwmon2=k10temp
FCTEMPS=hwmon1/pwm1=hwmon1/temp1_input hwmon1/pwm3=hwmon1/temp1_input hwmon1/pwm4=hwmon1/temp1_input hwmon1/pwm6=hwmon1/temp1_input hwmon1/pwm2=hwmon2/temp1_input
FCFANS=hwmon1/pwm1=hwmon1/fan1_input hwmon1/pwm3=hwmon1/fan3_input hwmon1/pwm4=hwmon1/fan4_input hwmon1/pwm6=hwmon1/fan6_input hwmon1/pwm2=hwmon1/fan2_input
MINTEMP=hwmon1/pwm1=20 hwmon1/pwm3=20 hwmon1/pwm4=20 hwmon1/pwm6=20 hwmon1/pwm2=20
MAXTEMP=hwmon1/pwm1=60 hwmon1/pwm3=60 hwmon1/pwm4=60 hwmon1/pwm6=60 hwmon1/pwm2=60
MINSTART=hwmon1/pwm1=33 hwmon1/pwm3=33 hwmon1/pwm4=33 hwmon1/pwm6=33 hwmon1/pwm2=33
MINSTOP=hwmon1/pwm1=33 hwmon1/pwm3=33 hwmon1/pwm4=33 hwmon1/pwm6=33 hwmon1/pwm2=33
MINPWM=hwmon1/pwm1=33 hwmon1/pwm3=33 hwmon1/pwm4=33 hwmon1/pwm6=33 hwmon1/pwm2=33

What do you think of changing it to this?  What would be better?

# Configuration file generated by pwmconfig, changes will be lost
INTERVAL=10
DEVPATH=nct6798=devices/platform/nct6775.656 k10temp=devices/pci0000:00/0000:00:18.3
FCTEMPS=nct6798/pwm1=nct6798/temp1_input nct6798/pwm3=nct6798/temp1_input nct6798/pwm4=nct6798/temp1_input nct6798/pwm6=nct6798/temp1_input nct6798/pwm2=k10temp/temp1_input
FCFANS=nct6798/pwm1=nct6798/fan1_input nct6798/pwm3=nct6798/fan3_input nct6798/pwm4=nct6798/fan4_input nct6798/pwm6=nct6798/fan6_input nct6798/pwm2=nct6798/fan2_input
MINTEMP=nct6798/pwm1=20 nct6798/pwm3=20 nct6798/pwm4=20 nct6798/pwm6=20 nct6798/pwm2=20
MAXTEMP=nct6798/pwm1=60 nct6798/pwm3=60 nct6798/pwm4=60 nct6798/pwm6=60 nct6798/pwm2=60
MINSTART=nct6798/pwm1=33 nct6798/pwm3=33 nct6798/pwm4=33 nct6798/pwm6=33 nct6798/pwm2=33
MINSTOP=nct6798/pwm1=33 nct6798/pwm3=33 nct6798/pwm4=33 nct6798/pwm6=33 nct6798/pwm2=33
MINPWM=nct6798/pwm1=33 nct6798/pwm3=33 nct6798/pwm4=33 nct6798/pwm6=33 nct6798/pwm2=33

There is a ten year old ubuntu bug which this would fix: https://bugs.launchpad.net/ubuntu/+source/lm-sensors-3/+bug/576602
It contains a module dependency workaround in comment #13, which I have successfully used.

The fancontrol error this problem causes is: "Configuration appears to be outdated, please run pwmconfig again".

Darxus, 2020-11-15:
fancontrol has a section marked "# Detect path to sensors" which effectively describes these possible values.  It looks like it would be useful to use names that can be easily differentiated.  So maybe replacing "hwmon1" with "hwmon_nct6798", for example?

Darxus, 2020-11-15:
Arch advice for this problem: https://wiki.archlinux.org/index.php/Fan_speed_control#Device_Paths_have_Changed_in_%2Fetc%2Ffancontrol

Darxus, 2020-11-15:
A script that controls fans which handles changing hwmon numbers (not mine):

https://gist.github.com/Ropid/077816cec9e5a826ad417fc6ce5ac41a
https://www.reddit.com/r/archlinux/comments/fhcxs5/is_it_possible_to_use_fancontrol_on_3rd_gen_ryzen/fkaevnq/?context=3

markusressel, 2021-01-09:
I have had multiple system freezes over the past months because of fancontrol failing to ramp up fans because of this. I tried the advice in the Arch wiki but it does not work. I did not try other scripts yet like mentioned by @Darxus (thx for that!) because I never found one. I am almost at a point where I write my own tool because of this issue. Is this such a complicated thing to resolve that it takes this long to find and implement a solution? I don't want to be disrespectful here, but this has a severe impact on the stability of a system if you trust a tool like fancontrol to keep your CPU temperatures in check.

markusressel, 2021-01-18:
I have written [my own tool](https://github.com/markusressel/fan2go) now, based on the files in `/sys/class/hwmon` now. It should work no matter the path and even after suspend/resume (without the need for a restart). Feel free to use, create issues, contribute etc.

7eggert, 2021-02-09:
My solution:

**/usr/lib/systemd/system/fancontrol.service**
`ExecStartPre=/root/.lib/bin/mklinks-hwmon`

**cat /root/.lib/bin/mklinks-hwmon**
```
#!/bin/sh
set -e
mount hwmon-tmpfs /sys/class/hwmon/ -t tmpfs -osize=64k
cd /sys/class/hwmon/
ln -s ../../devices/platform/coretemp.0/hwmon/hwmon? hwmon0
ln -s ../../devices/pci0000:00/0000:00:03.0/0000:01:00.0/hwmon/hwmon? hwmon2
ln -s ../../devices/virtual/hwmon/hwmon? hwmon1
ln -s ../../devices/platform/f71882fg.2560/hwmon/hwmon? hwmon3
```

These entries are the symlinks when the configurations were created, with the hwmon number being replaced with a wildcard (maybe you'll need to use "*" instead of "?"). I imagine that a more general solution would create a set of symlinks in /var/run/fancontrol/devices/ on boot, but I'll leave that as an exercise to the reader :-)

Cryotize, 2021-11-17:
Any update on this issue? It's been quite a long time and i run into the same problem.

felipealmeida, 2023-07-20:
Hello,

I've had the same problem here. I changed the logic a bit, I don't think this works for i2c sensors. However, it works for my use-case. So, if somebody feels comfortable to generalize it, here is a patch.

https://gist.github.com/felipealmeida/bf25420c9b750bcfb3483cf576346a85

Rudd-O, 2023-12-04:
This is a big issue with many computers.  I'm surprised it still hasn't been fixed.

lukolszewski, 2024-09-07:
For anyone that came here 4 years later like me hoping it would be resolved it isn't (and I have no time to write it). But I created this workaround script.

Essentially if like me you have all your devices under hwmonX and that X is changing on every start. You can use this. One of my devices is called pwm2 - make sure to adjust if your's isn't.
(you need python3)
Place this in a file called /usr/bin/update_fancontrol.py
```
import os
import re

# define the device we look for
target_file = "pwm2"

hwmon_path = "/sys/class/hwmon/"
found_folder = None

for folder in os.listdir(hwmon_path):
    if re.match(r'hwmon\d+', folder):
        folder_path = os.path.join(hwmon_path, folder)
        if target_file in os.listdir(folder_path):
            found_folder = folder
            break

if found_folder:
    with open("/etc/fancontrol", "r") as file:
        fancontrol_content = file.read()

    updated_content = re.sub(r'hwmon\d+', found_folder, fancontrol_content)

    # Print the updated content instead of saving for testing
#    print(updated_content) # uncomment for testing

    # Comment out for testing
    with open("/etc/fancontrol", "w") as file:
        file.write(updated_content)
else:
    print("Target file not found in any hwmon folder.")
```
and place this in a file /etc/systemd/system/update_fancontrol.service:
```
[Unit]
Before=fancontrol.service
Description=config updater for fancontrol

[Service]
ExecStart=/usr/bin/python3 /usr/bin/update_fancontrol.py
WorkingDirectory=/tmp
Type=oneshot
RemainAfterExit=true

[Install]
WantedBy=fancontrol.service
```
Then run sudo systemctl enable update_fancontrol
and sudo systemctl start update_fancontrol

And on every startup this script will run before fancontrol, it will find where this pwm2 device is and it will update all hwmonX entries in your /etc/fancontrol to that folder.

**Important:**
Don't use this if you have more than one hwmonX in your /etc/fancontrol. So for eaxmple if your normal /etc/fancontrol contains hwmon4 and hwmon5 (essentially more than one number). This will break it.

webdev23, 2025-01-30:
I like to tinker with scripts and all, but here it's too much, i am at 2 finger to make a separate PWM circuit with a temperature sensor, using whatever 50 ct eeprom and live in peace.

markusressel, 2025-05-08:
There is no package (yet), but building youself is very simple. You could also create a package and I can add it to the list. I will not create packages for systems that I do not use myself. Take a look at the [README.md](https://github.com/markusressel/fan2go#manual).