lm-sensors/lm-sensors #230: cannot label 'drivetemp' sensors - bus handling for scsi busted?

zdzichu, 2020-03-25:
I have number of `drivetemp` sensors (`drivetemp-scsi-0-0, drivetemp-scsi-1-0, drivetemp-scsi-2-0`) and I wanted to have labels to tell them apart. I've tried with simple entry in config:

```
chip "drivetemp-scsi-1-0"
        label temp1     "SSD_850_PRO"
```

but it fails with:

```
# sensors drivetemp-\*
Error: File /etc/sensors.d/mother.conf, line 16: Undeclared bus id referenced
sensors_init: Can't parse bus name
```

I've tried few more combination, all failing:
* chip "drivetemp-*-1-0" -> Parse error in chip name
* chip "*-scsi-1-0" -> Undeclared bus id referenced; sensors_init: Can't parse bus name
* chip "\*-\*-0-0"->  Parse error in chip name

Then I've tried to use `bus` directive (`bus "scsi-1" "SATA drive 1"`) but it fails, because `sensors_parse_bus_id()` function is limited to *i2c* buses only (https://github.com/lm-sensors/lm-sensors/blob/master/lib/data.c#L216). 
`sensors_substitute_busses()` in `parse_config()` (https://github.com/lm-sensors/lm-sensors/blob/master/lib/init.c#L92) calls parse_bus_id() and fails.

The function `sensors_parse_chip_name()` seem to handle names like `drivetemp-scsi-1-0` properly: bus type 8 (SCSI), bus number 1, addr 0. Only that "bus number 1" is not defined, so substitution fails.

I think buses should be handled differently for SCSI devices, or maybe substitution function should be extended.

@groeck , what do you think?

JiiPee74, 2020-11-06:
```
sensors drivetemp-scsi-*-*
sensors drivetemp-scsi-*-0
sensors drivetemp-scsi-0-*
```

seems to work at least for me

However right now I'm not sure if it's good idea to even use drivetemp module.

```
sensors
drivetemp-scsi-39-0
Adapter: SCSI adapter
Storage temp:  +37.0°C  (low  =  +0.0°C, high = +60.0°C)
                        (crit low = -41.0°C, crit = +85.0°C)
                        (lowest = +36.0°C, highest = +39.0°C)

drivetemp-scsi-11-0
Adapter: SCSI adapter
Storage temp:  +32.0°C  (low  =  +0.0°C, high = +60.0°C)
                        (crit low = -41.0°C, crit = +85.0°C)
                        (lowest = +30.0°C, highest = +35.0°C)

drivetemp-scsi-4-0
Adapter: SCSI adapter
Storage temp:  +35.0°C

drivetemp-scsi-9-0
Adapter: SCSI adapter
Storage temp:  +31.0°C  (low  =  +0.0°C, high = +60.0°C)
                        (crit low = -41.0°C, crit = +85.0°C)
                        (lowest = +29.0°C, highest = +34.0°C)

drivetemp-scsi-2-0
Adapter: SCSI adapter
Storage temp:  +33.0°C  (low  =  +0.0°C, high = +60.0°C)
                        (crit low = -41.0°C, crit = +85.0°C)
                        (lowest = +31.0°C, highest = +36.0°C)

drivetemp-scsi-7-0
Adapter: SCSI adapter
Storage temp:  +30.0°C  (low  =  +0.0°C, high = +60.0°C)
                        (crit low = -41.0°C, crit = +85.0°C)
                        (lowest = +28.0°C, highest = +33.0°C)

drivetemp-scsi-0-0
Adapter: SCSI adapter
Storage temp:  +32.0°C  (low  =  +0.0°C, high = +60.0°C)
                        (crit low = -41.0°C, crit = +85.0°C)
                        (lowest = +29.0°C, highest = +35.0°C)

it8728-isa-0a30
Adapter: ISA adapter
Vcore:         1.24 V  (min =  +0.50 V, max =  +1.40 V)
+3.3V:         3.30 V  (min =  +3.14 V, max =  +3.47 V)
+12V:         12.10 V  (min = +11.38 V, max = +12.60 V)
+5V:           5.13 V  (min =  +4.74 V, max =  +5.25 V)
Vaxg:         12.00 mV (min =  +0.80 V, max =  +1.00 V)
Vrin:          1.78 V  (min =  +1.66 V, max =  +1.85 V)
Vdram:         1.56 V  (min =  +1.45 V, max =  +1.55 V)
3VSB:          3.38 V  (min =  +3.14 V, max =  +3.48 V)
Vbat:          3.12 V
CPU fan:     1013 RPM  (min =  700 RPM)
Top fan:      644 RPM  (min =  300 RPM)
Back fan:    1108 RPM  (min =  300 RPM)
Front fan:    910 RPM  (min =  300 RPM)
PCH temp:     +34.0°C  (low  = +10.0°C, high = +85.0°C)  sensor = thermistor
Socket temp:  +59.0°C  (low  = +10.0°C, high = +85.0°C)  sensor = Intel PECI

[... 139 lines cut ...]
Front fan:    908 RPM  (min =  300 RPM)
PCH temp:     +34.0°C  (low  = +10.0°C, high = +85.0°C)  sensor = thermistor
Socket temp:  +64.0°C  (low  = +10.0°C, high = +85.0°C)  sensor = Intel PECI
CPU Package:  +71.0°C  (high = +80.0°C, crit = +100.0°C)
Core 1:       +61.0°C  (high = +80.0°C, crit = +100.0°C)
Core 2:       +68.0°C  (high = +80.0°C, crit = +100.0°C)
Core 3:       +71.0°C  (high = +80.0°C, crit = +100.0°C)
Core 4:       +66.0°C  (high = +80.0°C, crit = +100.0°C)

sda WDC WD30EFRX-68E Storage temp:  +32.0°C  (low  =  +0.0°C, high = +60.0°C)
sdb WDC WD30EFRX-68E Storage temp:  +32.0°C  (low  =  +0.0°C, high = +60.0°C)
sdc WDC WD30EFRX-68A Storage temp:  +33.0°C  (low  =  +0.0°C, high = +60.0°C)
sdd OCZ-TRION100     Storage temp:  +36.0°C
sde OCZ-TRION100     Storage temp:  +35.0°C
sdf WDC WD20EARX-00P Storage temp:  +29.0°C  (low  =  +0.0°C, high = +60.0°C)
sdg WDC WD20EARX-00P Storage temp:  +30.0°C  (low  =  +0.0°C, high = +60.0°C)
sdh WDC WD30EFRX-68A Storage temp:  +31.0°C  (low  =  +0.0°C, high = +60.0°C)
sdi WDC WD30EFRX-68A Storage temp:  +31.0°C  (low  =  +0.0°C, high = +60.0°C)
sdj WDC WD30EFRX-68A Storage temp:  +31.0°C  (low  =  +0.0°C, high = +60.0°C)
sdk WDC WD30EFRX-68A Storage temp:  +32.0°C  (low  =  +0.0°C, high = +60.0°C)
sdl WDC WD1002FBYS-0 Storage temp:  +38.0°C  (low  =  +0.0°C, high = +60.0°C)
sdm WDC WD1002FBYS-0 Storage temp:  +37.0°C  (low  =  +0.0°C, high = +60.0°C)
```

I would like to see output something like that directly from sensors. It would require some sort of grouping of all the drivetemp-scsi-*-* "adapters" together I think.

groeck, 2020-11-07:
On 11/6/20 3:09 PM, JiiPee wrote:
> |sensors drivetemp-scsi-*-* sensors drivetemp-scsi-*-0 sensors drivetemp-scsi-0-* |
> 
> seems to work at least for me
> 
> However right now I'm not sure if it's good idea to even use drivetemp module.
> 

You don't have to use / load it. On Ubuntu, for example, all
you need to do is to add it to /etc/modprobe.d/blacklist.conf.

Guenter

JiiPee74, 2020-11-07:
My point isn't how to disable it, I know how to blacklist stuff and in fact I think most distros do not even load it. I had to do it manually also.

My point is how to make lmsensors better so you can actually use this module so you don't need to use extra service like hddtemp to archive about same result.

```
 hddtemp
/dev/sda: WDC WD30EFRX-68EUZN0: 31°C
/dev/sdb: WDC WD30EFRX-68EUZN0: 32°C
/dev/sdc: WDC WD30EFRX-68AX9N0: 32°C
/dev/sdd: OCZ-TRION100: 35°C
/dev/sde: OCZ-TRION100: 33°C
/dev/sdf: WDC WD20EARX-00PASB0: 29°C
/dev/sdg: WDC WD20EARX-00PASB0: 30°C
/dev/sdh: WDC WD30EFRX-68AX9N0: 31°C
/dev/sdi: WDC WD30EFRX-68AX9N0: 30°C
/dev/sdj: WDC WD30EFRX-68AX9N0: 31°C
/dev/sdk: WDC WD30EFRX-68AX9N0: 31°C
/dev/sdl: WDC WD1002FBYS-02A6B0: 38°C
/dev/sdm: WDC WD1002FBYS-01A6B0: 37°C
```

zdzichu, 2020-11-08:
@JiiPee74 how this works for you? Can you show your config where you change label for `temp1` on specific disk?

jonasmalacofilho, 2020-11-08:
@zdzichu I think he meant that it works on `sensors <chip>`, not on the config files.

JiiPee74, 2020-11-08:
> @JiiPee74 how this works for you? Can you show your config where you change label for `temp1` on specific disk?

As I said, I made dirty script to parse sensors output and do some filtering and finding block device name and model for drives.

```
#!/bin/bash

cat /proc/cpuinfo | grep "model name" | head -n 1 | awk -F':' '{print $2}'
cat /proc/cpuinfo | grep "cpu MHz"
echo ""

sens=$(sensors | grep -vE "(Adapter|coretemp|acpitz|it8728|crit low|lowest)")

sensors=$(grep -vE "(Storage temp|drivetemp)" <<< $sens)
sensors=$(sed -r '/^\s*$/d' <<< $sensors)

storage=$(grep -E "(Storage temp|drivetemp)" <<< $sens)
storage=$(tr '\n' ' ' <<< $storage)
storage=$(sed 's/\sdrivetemp-/\|drivetemp-/g' <<< $storage)

echo "$sensors"

IFS="\|"
stout=

for i in $storage
do
        # drivetemp-scsi-39-0
        driveid="$(awk -F'-' '{print $3}' <<< ${i})"
        blockid="$(ls /sys/bus/scsi/drivers/sd/${driveid}:0:0:0/block/)"
        model="$(cat /sys/bus/scsi/drivers/sd/${driveid}\:0\:0\:0/model)"
        stout="$stout$(echo "$(sed -e "s/drivetemp-scsi-$driveid-0/$blockid $model/" <<< ${i})\n")"
done

echo -e "$stout" | sort -h
```

This ofcourse requires that you have something like this also
```
cat /etc/sensors.d/drivetemp.conf
###
chip "drivetemp-*"

label temp1 "Storage temp"
```

zdzichu, 2020-11-09:
This bug is about labeling using config file. Please do not use it for other issues.

pallaswept, 2024-01-31:
> ```
> sensors drivetemp-scsi-*-*
> sensors drivetemp-scsi-*-0
> sensors drivetemp-scsi-0-*
> ```
> seems to work at least for me

But it does not allow you to uniquely identify devices, because you do not specify the bus and address - respectively, you:

> sensors drivetemp-scsi-*-*
Don't specify *either* bus *or* address, so you cannot uniquely identify the drive, in order to uniquely label it
> sensors drivetemp-scsi-*-0
Don't specify the bus, but do specify the address, but the address is ignored if you do not specify the bus, so this is effectively just the same lack of a solution for unique identification as the above
> sensors drivetemp-scsi-0-*
Specify the bus but not the address, again not uniquely specifying the drive.

All of your examples will work in that they will allow the drive to be seen by `sensors`, but none of them help OP with his request to be able to uniquely identify drives for labelling. Your shell script is a nice workaround if you only want to see the correct label in shell output, but it doesn't help us with software-based solutions, and that's where I'm finding this problem.

It's 4 years later and I have the same problem, so I found this thread, and #371 potentially contains a resolution but it seems to have been lost :( I don't mean to 'necro', but these things are still alive, just.... not doing much with their lives :D @groeck have you seen that patch? Would you please give it some consideration? Or perhaps some other means to enhance the `sensors_parse_bus_id` function to be aware of other bus types beyond i2c? Thanks in advance! :)