lm-sensors/lm-sensors #234: nct6775: Invalid temperature source

birdie-github, 2020-04-12:
This looks like it should probably be taken care of:

```
nct6775: Found NCT6798D or compatible chip at 0x2e:0x290
nct6775 nct6775.656: Invalid temperature source 28 at index 0, source register 0x100, temp register 0x73
nct6775 nct6775.656: Invalid temperature source 28 at index 1, source register 0x200, temp register 0x75
nct6775 nct6775.656: Invalid temperature source 28 at index 2, source register 0x300, temp register 0x77
nct6775 nct6775.656: Invalid temperature source 28 at index 3, source register 0x800, temp register 0x79
nct6775 nct6775.656: Invalid temperature source 28 at index 5, source register 0xa00, temp register 0x7d
```

I'm unable to find any additional information about this chip.

olysonek, 2020-04-14:
Hi. Why do you think this is an lm_sensors issue? Does sensors-detect suggest loading the nct6775 module?

birdie-github, 2020-04-16:
@olysonek 

I don't know what to say. These are kernel messages on boot. The bug is in the driver. If it's the wrong place to file this bug report - I apologize. In this case I'd like to know where I should file it. Kernel bugzilla? I thought if you had a project bug tracker here, it would be a proper place.

olysonek, 2020-04-16:
I was honestly just asking for context... I meant no harm :). No need to apologize. Sorry if it came out wrong.

lm_sensors is a userspace tool. This is not the right place to report kernel bugs. You should report this to the hwmon kernel subsystem maintainers. I'm not aware of any bugzilla instance that the subsystem has, so I think you should report this to the mailing list linux-hwmon at vger.kernel.org. Here's some basic info about the list:
http://vger.kernel.org/vger-lists.html#linux-hwmon

closed 2020-04-16 by birdie-github