lm-sensors/lm-sensors #236: Asus ZenBook UX431DA_UM431DA / asus-isa-0000 cpu_fan 0 RPM

axet, 2020-04-25:
Hello! CPU fan shows 0 RPM (I hear it spinning - wrong detection). Adding 'acpi_enforce_resources=lax' to kernel args has no effect.

```
axet@axet-laptop:~$ sensors
iwlwifi_1-virtual-0
Adapter: Virtual device
temp1:        +42.0°C  

k10temp-pci-00c3
Adapter: PCI adapter
Tdie:         +56.1°C  (high = +70.0°C)
Tctl:         +56.1°C  

BATT-acpi-0
Adapter: ACPI interface
in0:           7.85 V  
curr1:         0.00 A  

asus-isa-0000
Adapter: ISA adapter
cpu_fan:        0 RPM

amdgpu-pci-0300
Adapter: PCI adapter
vddgfx:           N/A  
vddnb:            N/A  
edge:         +56.0°C  (crit = +80.0°C, hyst =  +0.0°C)

acpitz-acpi-0
Adapter: ACPI interface
temp1:        +48.0°C  

axet@axet-laptop:~$ 
```

```
sudo sensors-detect
# sensors-detect version 3.6.0
# System: ASUSTeK COMPUTER INC. ZenBook UX431DA_UM431DA [1.0] (laptop)
# Board: ASUSTeK COMPUTER INC. UX431DA
# Kernel: 5.4.0-26-generic x86_64
# Processor: AMD Ryzen 5 3500U with Radeon Vega Mobile Gfx (23/24/1)

This program will help you determine which kernel modules you need
to load to use lm_sensors most effectively. It is generally safe
and recommended to accept the default answers to all questions,
unless you know what you're doing.

Some south bridges, CPUs or memory controllers contain embedded sensors.
Do you want to scan for them? This is totally safe. (YES/no): 
Module cpuid loaded successfully.
Silicon Integrated Systems SIS5595...                       No
VIA VT82C686 Integrated Sensors...                          No
VIA VT8231 Integrated Sensors...                            No
AMD K8 thermal sensors...                                   No
AMD Family 10h thermal sensors...                           No
AMD Family 11h thermal sensors...                           No
AMD Family 12h and 14h thermal sensors...                   No
AMD Family 15h thermal sensors...                           No
AMD Family 16h thermal sensors...                           No
AMD Family 17h thermal sensors...                           Success!
    (driver `k10temp')
AMD Family 15h power sensors...                             No
AMD Family 16h power sensors...                             No
Hygon Family 18h thermal sensors...                         No
Intel digital thermal sensor...                             No
Intel AMB FB-DIMM thermal sensor...                         No
Intel 5500/5520/X58 thermal sensor...                       No
VIA C7 thermal sensor...                                    No
VIA Nano thermal sensor...                                  No
[... 920 lines cut ...]
Probing for `Texas Instruments TMP441'...                   No
Probing for `Texas Instruments AMC6821'...                  No
Probing for `National Semiconductor LM95234'...             No
Probing for `National Semiconductor LM64'...                No
Probing for `National Semiconductor LM73'...                No
Probing for `Maxim MAX6633/MAX6634/MAX6635'...              No
Probing for `NXP/Philips SA56004'...                        No
Probing for `Fintek F75121R/F75122R/RG (VID+GPIO)'...       No
Probing for `Fintek F75111R/RG/N (GPIO)'...                 No
Probing for `ITE IT8201R/IT8203R/IT8206R/IT8266R'...        No
Client found at address 0x77
Probing for `Asus Mozart-2'...                              No

Now follows a summary of the probes I have just done.
Just press ENTER to continue: 

Driver `k10temp' (autoloaded):
  * Chip `AMD Family 17h thermal sensors' (confidence: 9)

No modules to load, skipping modules configuration.

Unloading cpuid... OK

axet@axet-laptop:~$ 
```

RonnySvedman, 2020-11-14:
THe fan speed on many Asus products are read from the ITE chip instead of the asus-isa-0000. try modprobe it87, or compile the unofficial driver sources from [https://github.com/a1wong/it87](url) or some fork of it whichever is newest.  I did this for my z97p yesterday and it worked. you will have to edit scaling and limits in sensors3.conf after you get working changing values from the it87 driver.

axet, 2020-11-14:
Thanks for suggestion.

```
# sudo modprobe it87
modprobe: ERROR: could not insert 'it87': No such device
```