lm-sensors/lm-sensors #238: sensors 3.6.0 the change of sensors3.conf has no effect

edefranco, 2020-04-30:
Hi,
I compiled sensors 3.6.0 (before I had installed distro package version 3.4.0 and the result was the same) on ubuntu server 18.04 but for my chip w83627hf any label modification on the sensors3.conf has no effect (yes, I applied sensors -s after that).
Not only, but if I remove sensors3.conf I see the same resul as when sensors3.conf was in /etc.
Any suggested is very apreciated,
thank in advance,

Emilio

olysonek, 2020-05-04:
Hi, can you share your config?

edefranco, 2020-05-04:
In attachment my cfg file: org.sensors3.conf is the file original and the sensors3.conf is the file in use with the only cfg for my chip.

[org.sensors3.conf.txt](https://github.com/lm-sensors/lm-sensors/files/4573934/org.sensors3.conf.txt)
[sensors3.conf.txt](https://github.com/lm-sensors/lm-sensors/files/4573935/sensors3.conf.txt)

olysonek, 2020-05-05:
OK, and what is the output of `sensors`?

edefranco, 2020-05-05:
This is the output and is thealways the same any change I make in the .conf:

k8temp-pci-00c3
Adapter: PCI adapter
temp1:        +47.0°C  
temp3:        +45.0°C  

nouveau-pci-0600
Adapter: PCI adapter
temp1:        +40.0°C  (high = +95.0°C, hyst =  +3.0°C)
                       (crit = +130.0°C, hyst = +10.0°C)
                       (emerg = +135.0°C, hyst =  +5.0°C)

w83697hf-isa-0290
Adapter: ISA adapter
in0:           1.39 V  (min =  +0.51 V, max =  +0.02 V)  ALARM
in2:           3.28 V  (min =  +0.54 V, max =  +0.00 V)  ALARM
in3:           2.98 V  (min =  +0.27 V, max =  +2.05 V)  ALARM
in4:           2.93 V  (min =  +2.13 V, max =  +0.00 V)  ALARM
in5:           3.12 V  (min =  +0.00 V, max =  +3.86 V)
in6:           3.31 V  (min =  +3.14 V, max =  +1.02 V)  ALARM
in7:           3.26 V  (min =  +0.18 V, max =  +1.60 V)  ALARM
in8:           3.36 V  (min =  +0.13 V, max =  +0.00 V)  ALARM
fan1:           0 RPM  (min = 28125 RPM, div = 8)  ALARM
fan2:        3013 RPM  (min = 84375 RPM, div = 8)  ALARM
temp1:        +45.0°C  (high =  +9.0°C, hyst = -44.0°C)  ALARM  sensor = thermistor
temp2:        +46.5°C  (high = +80.0°C, hyst = +75.0°C)  sensor = thermistor
beep_enable: enabled

olysonek, 2020-05-06:
Well, none of the expressions in the `chip` statement match any of the chips you have in the `sensors` output.

edefranco, 2020-05-07:
Well, you are right !
configuring the chip w83697hf-* all working fine, but why the sensors don't notify that are not none config for the chip in use ? And, why, sensors also without cfg file print out random values ?
Emilio

edefranco, 2020-05-07:
I am not be able to setup hyst form temp: the temp1_hyst is not accepted. Have you suggestion ?

Thank in advance 
emilio

olysonek, 2020-05-11:
> but why the sensors don't notify that are not none config for the chip in use

You mean, why doesn't it warn that there is no config for the chip you have on the board? The answer to that would be that no config is strictly necessary.

> why, sensors also without cfg file print out random values

What do you mean by "random values"?

> I am not be able to setup hyst form temp: the temp1_hyst is not accepted. Have you suggestion ?

I don't know. What does it say when you try to change it? It's possible that the driver and/or the hardware doesn't support changing it.

subterfugium, 2020-07-23:
I might have similar issue

sensors output:
```
amdgpu-pci-0900
Adapter: PCI adapter
vddgfx:      725.00 mV 
fan1:         740 RPM  (min =    0 RPM, max = 3200 RPM)
edge:         +36.0°C  (crit = +94.0°C, hyst = -273.1°C)
power1:       33.13 W  (cap = 175.00 W)
```
Then I added file
`/etc/sensors.d/renames`
with content:
```
chip "amdgpu-pci-*"
  # Radeon labels
  label edge "GPU Temp"
  label vddgfx "GPU Voltage"
  label power1 "GPU Power"
  label fan1 "GPU Fan"
```
And sensors output:
```
amdgpu-pci-0900
Adapter: PCI adapter
vddgfx:      725.00 mV 
GPU Fan:      741 RPM  (min =    0 RPM, max = 3200 RPM)
edge:         +37.0°C  (crit = +94.0°C, hyst = -273.1°C)
GPU Power:    33.14 W  (cap = 175.00 W)
```
edge seems to be GPU temperature which I was trying to change but it does not seem to work.

olysonek, 2020-07-23:
> edge seems to be GPU temperature which I was trying to change but it does not seem to work.

That's because `edge` is not the name of the input. Rather it's the kernel-provided label for the input. Please post the output of `sensors -u`

subterfugium, 2020-07-23:
```
sensors -u

amdgpu-pci-0900
Adapter: PCI adapter
vddgfx:
  in0_input: 0.725
GPU Fan:
  fan1_input: 860.000
  fan1_min: 0.000
  fan1_max: 3200.000
edge:
  temp1_input: 26.000
  temp1_crit: 94.000
  temp1_crit_hyst: -273.150
GPU Power:
  power1_average: 33.231
  power1_cap: 175.000
```
It's temp1 right?

Changed
`label edge "GPU Temp"` to `label temp1 "GPU Temp"`
```
amdgpu-pci-0900
Adapter: PCI adapter
vddgfx:      725.00 mV 
GPU Fan:      744 RPM  (min =    0 RPM, max = 3200 RPM)
GPU Temp:     +29.0°C  (crit = +94.0°C, hyst = -273.1°C)
GPU Power:    33.16 W  (cap = 175.00 W)
```
Thanks!

olysonek, 2020-07-23:
Right. You may want to replace `vddgfx` wilth `in0` too in your config.