lm-sensors/lm-sensors #239: no support for fan on Intel NUC8i5BEH

fpoto, 2020-05-05:
Fan speed is read by BIOS, but no fan sensor is detected by sensors-detect.
By reading around, this seems to be the case for all Intel NUCs.

If I can help, try to ask: this the system where I work, but I can do some experiments if told how,

olysonek, 2020-05-06:
What lm_sensors version are you using? Can you post the output of `sensors-detect`?

fpoto, 2020-05-06:
`
# sensors-detect version 3.6.0
# System: Intel(R) Client Systems NUC8i5BEH [J72747-306]
# Board: Intel Corporation NUC8BEB
# Kernel: 5.6.0-1-amd64 x86_64
# Processor: Intel(R) Core(TM) i5-8259U CPU @ 2.30GHz (6/142/10)

This program will help you determine which kernel modules you need
to load to use lm_sensors most effectively. It is generally safe
and recommended to accept the default answers to all questions,
unless you know what you're doing.

Some south bridges, CPUs or memory controllers contain embedded sensors.
Do you want to scan for them? This is totally safe. (YES/no): 
Module cpuid loaded successfully.
Silicon Integrated Systems SIS5595...                       No
VIA VT82C686 Integrated Sensors...                          No
VIA VT8231 Integrated Sensors...                            No
AMD K8 thermal sensors...                                   No
AMD Family 10h thermal sensors...                           No
AMD Family 11h thermal sensors...                           No
AMD Family 12h and 14h thermal sensors...                   No
AMD Family 15h thermal sensors...                           No
AMD Family 16h thermal sensors...                           No
AMD Family 17h thermal sensors...                           No
AMD Family 15h power sensors...                             No
AMD Family 16h power sensors...                             No
Hygon Family 18h thermal sensors...                         No
Intel digital thermal sensor...                             Success!
    (driver `coretemp')
Intel AMB FB-DIMM thermal sensor...                         No
Intel 5500/5520/X58 thermal sensor...                       No
VIA C7 thermal sensor...                                    No
VIA Nano thermal sensor...                                  No

Some Super I/O chips contain embedded sensors. We have to write to
standard I/O ports to probe them. This is usually safe.
Do you want to scan for Super I/O sensors? (YES/no): 
Probing for Super-I/O at 0x2e/0x2f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               No
Trying family `ITE'...                                      No
Probing for Super-I/O at 0x4e/0x4f
Trying family `National Semiconductor/ITE'...               Yes
Found unknown chip with ID 0x8987

Some systems (mainly servers) implement IPMI, a set of common interfaces
through which system health data may be retrieved, amongst other things.
We first try to get the information from SMBIOS. If we don't find it
there, we have to read from arbitrary I/O ports to probe for such
interfaces. This is normally safe. Do you want to scan for IPMI
interfaces? (YES/no): 
Probing for `IPMI BMC KCS' at 0xca0...                      No
Probing for `IPMI BMC SMIC' at 0xca8...                     No

Some hardware monitoring chips are accessible through the ISA I/O ports.
We have to write to arbitrary I/O ports to probe them. This is usually
safe though. Yes, you do have ISA I/O ports even if you do not have any
ISA slots! Do you want to scan the ISA I/O ports? (YES/no): 
Probing for `National Semiconductor LM78' at 0x290...       No
Probing for `National Semiconductor LM79' at 0x290...       No
Probing for `Winbond W83781D' at 0x290...                   No
Probing for `Winbond W83782D' at 0x290...                   No

Lastly, we can probe the I2C/SMBus adapters for connected hardware
monitoring devices. This is the most risky part, and while it works
reasonably well on most systems, it has been reported to cause trouble
on some systems.
Do you want to probe the I2C/SMBus adapters now? (YES/no): 
Using driver `i2c-i801' for device 0000:00:1f.4: Cannon Lake-LP (PCH)
Module i2c-dev loaded successfully.

Next adapter: SMBus I801 adapter at efa0 (i2c-0)
Do you want to scan it? (YES/no/selectively): 
Client found at address 0x50
Probing for `Analog Devices ADM1033'...                     No
Probing for `Analog Devices ADM1034'...                     No
Probing for `SPD EEPROM'...                                 Yes
    (confidence 8, not a hardware monitoring chip)
Probing for `EDID EEPROM'...                                No
Client found at address 0x52
Probing for `Analog Devices ADM1033'...                     No
Probing for `Analog Devices ADM1034'...                     No
Probing for `SPD EEPROM'...                                 Yes
    (confidence 8, not a hardware monitoring chip)

Next adapter: i915 gmbus dpb (i2c-1)
Do you want to scan it? (yes/NO/selectively): yes

Next adapter: i915 gmbus dpc (i2c-2)
Do you want to scan it? (yes/NO/selectively): yes

Next adapter: i915 gmbus misc (i2c-3)
Do you want to scan it? (yes/NO/selectively): yes

Next adapter: i915 gmbus dpd (i2c-4)
Do you want to scan it? (yes/NO/selectively): yes

Next adapter: DPDDC-B (i2c-5)
Do you want to scan it? (yes/NO/selectively): yes

Next adapter: DPDDC-C (i2c-6)
Do you want to scan it? (yes/NO/selectively): yes

Now follows a summary of the probes I have just done.
Just press ENTER to continue: 

Driver `coretemp':
  * Chip `Intel digital thermal sensor' (confidence: 9)

To load everything that is needed, add this to /etc/modules:
#----cut here----
# Chip drivers
coretemp
#----cut here----
If you have some drivers built into your kernel, the list above will
contain too many modules. Skip the appropriate ones!

Do you want to add these lines automatically to /etc/modules? (yes/NO)

Unloading i2c-dev... OK
Unloading cpuid... OK
`

olysonek, 2020-05-06:
And what is the output of `sensors`?

> Trying family `National Semiconductor/ITE'...               Yes
> Found unknown chip with ID 0x8987

If the fan speed is read from an ITE Super IO chip, chances are that there's no easy solution for your problem. As far as I understand it, support for ITE chips is generally not good due to ITE not providing documentation for the chips. Searching for ITE-related issues in this repo could give you an idea.

fpoto, 2020-05-06:
Ok, thanks, I'll look for ITE. Here is the output from sensors:

```
pch_cannonlake-virtual-0
Adapter: Virtual device
temp1:        +39.0°C  

acpitz-acpi-0
Adapter: ACPI interface
temp1:       -263.2°C  
temp2:        +27.8°C  (crit = +119.0°C)

iwlwifi_1-virtual-0
Adapter: Virtual device
temp1:            N/A  

coretemp-isa-0000
Adapter: ISA adapter
Package id 0:  +39.0°C  (high = +100.0°C, crit = +100.0°C)
Core 0:        +37.0°C  (high = +100.0°C, crit = +100.0°C)
Core 1:        +37.0°C  (high = +100.0°C, crit = +100.0°C)
Core 2:        +37.0°C  (high = +100.0°C, crit = +100.0°C)
Core 3:        +39.0°C  (high = +100.0°C, crit = +100.0°C)
```

zdzichu, 2020-05-12:
You may try this unofficial driver:  https://github.com/a1wong/it87

fpoto, 2020-05-12:
Thanks for the hint.
However, trying `modprobe it87` (and its variations it87_wdt and it8712f_wdt) with the stock kernel 5.6.0 drivers results in "no such device", so I suspect the chip is not the right one, am I wrong?

zdzichu, 2020-05-13:
In-kernel driver supports fewer chips than the unofficial one.

fpoto, 2020-05-13:
Nope :(
I just tried both with the source you suggested and with https://github.com/hannesha/it87/tree/rebase
In both cases I get `modprobe: ERROR: could not insert 'it87': No such device`

paulo-loureiro, 2020-10-13:
Do I understand correctly that a solution is not available or even possible? I just got an Intel NUC8i5BEH and installed debian on it. Fan runs at full speed all the time, independently of the BIOS settings. I get the same error output from sensors-detect:

Trying family `National Semiconductor/ITE'...               Yes
Found unknown chip with ID 0x8987

fpoto, 2020-11-11:
As far as I know, it is currently not possible. Do you still have the fan running fast? My fan is almost inaudible, even in a silent room.

paulo-loureiro, 2020-11-13:
> As far as I know, it is currently not possible. Do you still have the fan running fast? My fan is almost inaudible, even in a silent room.

I got an bios/uefi update from Intel (for the NUC8) which solves the issue and now I can control the fan speed.

fpoto, 2020-11-13:
I have a NUC8i5BEH2. How did you get the update? Is it publicly available?

paulo-loureiro, 2020-11-16:
Yes, it is publicly available(v85). Mine is a NUC8i5BEH. Check it here:

https://downloadcenter.intel.com/download/29959/BIOS-Update-BECFL357-86A-?wapkw=nuc8i5beh

fpoto, 2020-11-16:
Thank you.  I had no problems with the fan, apart from the fact that it
cannot be monitored on Linux as the it87 driver does not work on this
board.  So I installed the new BIOS, but nothing changed: it87 still
does not work :(