lm-sensors/lm-sensors #240: No CPU FAN sensor monitoring on Acer Nitro AN517-51

maqsudjon, 2020-05-07:
# sensors-detect version 3.6.0
# System: Acer Nitro AN517-51 [V1.28] (laptop)
# Board: CFL Superb_CFS
# Kernel: 5.4.0-29-generic x86_64
# Processor: Intel(R) Core(TM) i7-9750H CPU @ 2.60GHz (6/158/10)

**$ sudo sensors**
iwlwifi_1-virtual-0
Adapter: Virtual device
temp1:        +49.0°C  

coretemp-isa-0000
Adapter: ISA adapter
Package id 0:  +45.0°C  (high = +100.0°C, crit = +100.0°C)
Core 0:        +45.0°C  (high = +100.0°C, crit = +100.0°C)
Core 1:        +36.0°C  (high = +100.0°C, crit = +100.0°C)
Core 2:        +36.0°C  (high = +100.0°C, crit = +100.0°C)
Core 3:        +36.0°C  (high = +100.0°C, crit = +100.0°C)
Core 4:        +38.0°C  (high = +100.0°C, crit = +100.0°C)
Core 5:        +36.0°C  (high = +100.0°C, crit = +100.0°C)

pch_cannonlake-virtual-0
Adapter: Virtual device
temp1:        +61.0°C  

BAT1-acpi-0
Adapter: ACPI interface
in0:          17.21 V  
curr1:         0.00 A

RonnySvedman, 2020-11-14:
run sensors-config and see if it finds some more sensors, add the output here if not or it finds an unsupported one.

maqsudjon, 2020-11-14:
**$ sudo sensors-detect** 
#-- sensors-detect version 3.6.0
#-- System: Acer Nitro AN517-51 [V1.31] (laptop)
#-- Board: CFL Superb_CFS
#-- Kernel: 5.4.0-52-generic x86_64
#-- Processor: Intel(R) Core(TM) i7-9750H CPU @ 2.60GHz (6/158/10)

This program will help you determine which kernel modules you need
to load to use lm_sensors most effectively. It is generally safe
and recommended to accept the default answers to all questions,
unless you know what you're doing.

Some south bridges, CPUs or memory controllers contain embedded sensors.
Do you want to scan for them? This is totally safe. (YES/no): YES
Silicon Integrated Systems SIS5595...                       No
VIA VT82C686 Integrated Sensors...                          No
VIA VT8231 Integrated Sensors...                            No
AMD K8 thermal sensors...                                   No
AMD Family 10h thermal sensors...                           No
AMD Family 11h thermal sensors...                           No
AMD Family 12h and 14h thermal sensors...                   No
AMD Family 15h thermal sensors...                           No
AMD Family 16h thermal sensors...                           No
AMD Family 17h thermal sensors...                           No
AMD Family 15h power sensors...                             No
AMD Family 16h power sensors...                             No
Hygon Family 18h thermal sensors...                         No
Intel digital thermal sensor...                             Success!
    (driver `coretemp')
Intel AMB FB-DIMM thermal sensor...                         No
Intel 5500/5520/X58 thermal sensor...                       No
VIA C7 thermal sensor...                                    No
VIA Nano thermal sensor...                                  No

Some Super I/O chips contain embedded sensors. We have to write to
standard I/O ports to probe them. This is usually safe.
Do you want to scan for Super I/O sensors? (YES/no): YES
/dev/port: Operation not permitted

Some hardware monitoring chips are accessible through the ISA I/O ports.
We have to write to arbitrary I/O ports to probe them. This is usually
safe though. Yes, you do have ISA I/O ports even if you do not have any
ISA slots! Do you want to scan the ISA I/O ports? (YES/no): YES
/dev/port: Operation not permitted

Lastly, we can probe the I2C/SMBus adapters for connected hardware
monitoring devices. This is the most risky part, and while it works
reasonably well on most systems, it has been reported to cause trouble
on some systems.
Do you want to probe the I2C/SMBus adapters now? (YES/no): YES
Using driver `i2c-i801' for device 0000:00:1f.4: Cannon Lake-H (PCH)

Next adapter: SMBus I801 adapter at 5040 (i2c-0)
Do you want to scan it? (YES/no/selectively): YES
Client found at address 0x50
Probing for `Analog Devices ADM1033'...                     No
Probing for `Analog Devices ADM1034'...                     No
Probing for `SPD EEPROM'...                                 Yes
    (confidence 8, not a hardware monitoring chip)
Probing for `EDID EEPROM'...                                No
Client found at address 0x52
Probing for `Analog Devices ADM1033'...                     No
Probing for `Analog Devices ADM1034'...                     No
Probing for `SPD EEPROM'...                                 Yes
    (confidence 8, not a hardware monitoring chip)

Next adapter: Synopsys DesignWare I2C adapter (i2c-1)
Do you want to scan it? (YES/no/selectively): YES
Adapter doesn't support all probing functions.
Some addresses won't be probed.
[... 875 lines cut ...]
to load them.

**$ sensors**
iwlwifi_1-virtual-0
Adapter: Virtual device
temp1:        +57.0°C  

coretemp-isa-0000
Adapter: ISA adapter
Package id 0:  +41.0°C  (high = +100.0°C, crit = +100.0°C)
Core 0:        +40.0°C  (high = +100.0°C, crit = +100.0°C)
Core 1:        +40.0°C  (high = +100.0°C, crit = +100.0°C)
Core 2:        +39.0°C  (high = +100.0°C, crit = +100.0°C)
Core 3:        +39.0°C  (high = +100.0°C, crit = +100.0°C)
Core 4:        +41.0°C  (high = +100.0°C, crit = +100.0°C)
Core 5:        +40.0°C  (high = +100.0°C, crit = +100.0°C)

pch_cannonlake-virtual-0
Adapter: Virtual device
temp1:        +62.0°C  

BAT1-acpi-0
Adapter: ACPI interface
in0:          16.86 V  
curr1:         0.00 A

RuShiiiiii, 2022-03-31:
Everyone'd have guessed it, but same with Acer Nitro 5 ANS515-54

LanShang1206, 2022-11-21:
the same problem with Acer Nitro AN515-54, SOS pls......

macmoi, 2022-11-27:
Same problem on Acer Nitro AN517-54.