lm-sensors/lm-sensors #243: No sensors detected for ASUS TUF Gaming FA506IU (AMD Ryzen 7 4800H)

julenpardo, 2020-06-02:
Hello,

I'm using Manjaro 20.0.2 with 5.6.15 kernel.

```
❯❯❯ sudo sensors-detect
# sensors-detect version 3.6.0
# System: ASUSTeK COMPUTER INC. TUF Gaming FA506IU_FA506IU [1.0] (laptop)
# Board: ASUSTeK COMPUTER INC. FA506IU
# Kernel: 5.6.15-1-MANJARO x86_64
# Processor: AMD Ryzen 7 4800H with Radeon Graphics (23/96/1)

This program will help you determine which kernel modules you need
to load to use lm_sensors most effectively. It is generally safe
and recommended to accept the default answers to all questions,
unless you know what you're doing.

Some south bridges, CPUs or memory controllers contain embedded sensors.
Do you want to scan for them? This is totally safe. (YES/no):
Module cpuid loaded successfully.
Silicon Integrated Systems SIS5595...                       No
VIA VT82C686 Integrated Sensors...                          No
VIA VT8231 Integrated Sensors...                            No
AMD K8 thermal sensors...                                   No
AMD Family 10h thermal sensors...                           No
AMD Family 11h thermal sensors...                           No
AMD Family 12h and 14h thermal sensors...                   No
AMD Family 15h thermal sensors...                           No
AMD Family 16h thermal sensors...                           No
AMD Family 17h thermal sensors...                           No
AMD Family 15h power sensors...                             No
AMD Family 16h power sensors...                             No
Hygon Family 18h thermal sensors...                         No
Intel digital thermal sensor...                             No
Intel AMB FB-DIMM thermal sensor...                         No
Intel 5500/5520/X58 thermal sensor...                       No
VIA C7 thermal sensor...                                    No
VIA Nano thermal sensor...                                  No

Some Super I/O chips contain embedded sensors. We have to write to
standard I/O ports to probe them. This is usually safe.
Do you want to scan for Super I/O sensors? (YES/no):
Probing for Super-I/O at 0x2e/0x2f
Trying family `National Semiconductor/ITE'...               Yes
Found unknown chip with ID 0x8987
Probing for Super-I/O at 0x4e/0x4f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               No
Trying family `ITE'...                                      No

Some hardware monitoring chips are accessible through the ISA I/O ports.
We have to write to arbitrary I/O ports to probe them. This is usually
safe though. Yes, you do have ISA I/O ports even if you do not have any
ISA slots! Do you want to scan the ISA I/O ports? (YES/no):

Lastly, we can probe the I2C/SMBus adapters for connected hardware
monitoring devices. This is the most risky part, and while it works
reasonably well on most systems, it has been reported to cause trouble
on some systems.
Do you want to probe the I2C/SMBus adapters now? (YES/no):
Sorry, no sensors were detected.
This is relatively common on laptops, where thermal management is
handled by ACPI rather than the OS.
```
```
❯❯❯ sudo sensors
amdgpu-pci-0500
Adapter: PCI adapter
vddgfx:           N/A
vddnb:            N/A
edge:         +47.0°C  (crit =  +0.0°C, hyst =  +0.0°C)

acpitz-acpi-0
Adapter: ACPI interface
temp1:        +69.0°C  (crit = +115.0°C)

BAT1-acpi-0
Adapter: ACPI interface
in0:           4.00 V
curr1:         0.00 A

nvme-pci-0400
Adapter: PCI adapter
Composite:    +34.9°C  (low  = -20.1°C, high = +84.8°C)
                       (crit = +89.8°C)

```

rileyweber13, 2020-06-05:
I had the same problem on a machine with a Ryzen 5 3600. Looks like zen/zen+/zen2 architectures are not (yet?) supported. See https://hwmon.wiki.kernel.org/device_support_status_a_c

RonnySvedman, 2020-11-14:
Asus keeps mounting IT87 series chips on their motherboards since way back (at least before 2010). ITE refuses to release specs for their chips. This led to the maintainer of the it87 module giving up in 2018, and it87 is real old in the kernel. There are newer beta sources floating around, I compiled https://github.com/a1wong/it87[](https://github.com/a1wong/it87) and got fan speeds (and fancontrol)  working on my z97p in ubuntu 20.04 as recently as yesterday. (it broke when upgrading ubuntu 18.04LTS to 20.04LTS, but I think I've had it running since 13.10 at least). If your chip is not supported by newer sources you should ask whoever is runnnig the it87 module show to research your chip,  and/or ask the hwmon developers to include/update the it87  in the kernel.  

You will have to edit sensors3.conf for labels, scaling and limits by hand once you get "changing values" working, unless you want most values doubled or halved - this is not the fault of lm-sensors maintainers, they support chips, not motherboards.

miurahr, 2021-03-08:
k10temp module detects CPU Tctl and Tdie temperature with Linux kernel 5.10 on AMD Ryzen 7 PRO 4750U 
Ubuntu 20.04 now ships 5.10.0-oem kernel and it works well on ThinkPad T14s-gen1-amd notePC.