lm-sensors/lm-sensors #244: Gigabyte B450 AORUS ELITE: Fan speed 0 rpm

MichaelTrip, 2020-06-08:
Hi,

I just bought a new PC, but the fan speed is reporting 0 rpm. I am running Manjaro. 

Version of sensors:

```
sensors version 3.6.0 with libsensors version 3.6.0
` ``
output of sensors command

```

sensors
it8792-isa-0a60
Adapter: ISA adapter
in0:           1.80 V  (min =  +0.00 V, max =  +2.78 V)
in1:           1.34 V  (min =  +0.00 V, max =  +2.78 V)
in2:           1.22 V  (min =  +0.00 V, max =  +2.78 V)
+3.3V:         1.69 V  (min =  +0.00 V, max =  +2.78 V)
in4:           1.33 V  (min =  +0.00 V, max =  +2.78 V)
in5:           1.22 V  (min =  +0.00 V, max =  +2.78 V)
in6:           2.78 V  (min =  +0.00 V, max =  +2.78 V)  ALARM
3VSB:          1.68 V  (min =  +0.00 V, max =  +2.78 V)
Vbat:          1.64 V  
fan1:           0 RPM  (min =    0 RPM)
fan2:           0 RPM  (min =    0 RPM)
fan3:           0 RPM  (min =    0 RPM)
temp1:        +28.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp2:        +31.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
temp3:        +33.0°C  (low  = +127.0°C, high = +127.0°C)  sensor = thermistor
intrusion0:  ALARM

nvme-pci-0100
Adapter: PCI adapter
Composite:    +34.9°C  (low  =  -0.1°C, high = +74.8°C)
                       (crit = +79.8°C)

amdgpu-pci-0900
Adapter: PCI adapter
vddgfx:      775.00 mV 
fan1:           0 RPM  (min =    0 RPM, max = 3800 RPM)
edge:         +51.0°C  (crit = +118.0°C, hyst = -273.1°C)
                       (emerg = +99.0°C)
junction:     +51.0°C  (crit = +99.0°C, hyst = -273.1°C)
                       (emerg = +99.0°C)
mem:          +50.0°C  (crit = +99.0°C, hyst = -273.1°C)
                       (emerg = +99.0°C)
power1:       16.00 W  (cap = 160.00 W)

k10temp-pci-00c3
Adapter: PCI adapter
Vcore:       1000.00 mV 
Vsoc:          1.02 V  
Tdie:         +39.2°C  
Tctl:         +39.2°C  
Tccd1:        +40.8°C  
Icore:         7.00 A  
Isoc:          7.75 A  

```

ga72Aa2mBQ2SnmivoyymQqH4, 2020-10-13:
The technical base of Manjaro is: https://wiki.archlinux.org/index.php/Lm_sensors#Gigabyte_B250/Z370/B450M_motherboards
Maybe you are using an super IO chip current unknown to the OS.
Update driver for chip IT87: https://github.com/bbqlinux/it87

available configurations for Aorus / Gigabyte motherboards:
https://github.com/lm-sensors/lm-sensors/tree/master/configs/Gigabyte

francoism90, 2023-04-23:
I'm having the same issue on a Gigabyte B450 I Aorus Pro WiFi.

How can I help to support this MB? It seems more sensors have been added recently.

closed 2024-10-26 by MichaelTrip