lm-sensors/lm-sensors #255: Conversion error

bmn-m, 2020-07-15:
Hello, as i run sensors i get the following results: 

`k10temp-pci-00c3
Adapter: PCI adapter
Tdie:         +45.8°C  (high = +70.0°C)
Tctl:         +45.8°C  

amdgpu-pci-0400
Adapter: PCI adapter
vddgfx:           N/A  
vddnb:            N/A  
edge:         +45.0°C  (crit = +80.0°C, hyst =  +0.0°C)

nct6793-isa-0290
Adapter: ISA adapter
in0:                   592.00 mV (min =  +0.00 V, max =  +1.74 V)
in1:                     1.86 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in2:                     3.41 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in3:                     3.39 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in4:                   256.00 mV (min =  +0.00 V, max =  +0.00 V)  ALARM
in5:                   136.00 mV (min =  +0.00 V, max =  +0.00 V)  ALARM
in6:                   720.00 mV (min =  +0.00 V, max =  +0.00 V)  ALARM
in7:                     3.39 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in8:                     3.26 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in9:                     1.85 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in10:                  184.00 mV (min =  +0.00 V, max =  +0.00 V)  ALARM
in11:                  136.00 mV (min =  +0.00 V, max =  +0.00 V)  ALARM
in12:                    1.86 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in13:                    1.73 V  (min =  +0.00 V, max =  +0.00 V)  ALARM
in14:                  192.00 mV (min =  +0.00 V, max =  +0.00 V)  ALARM
fan1:                     0 RPM  (min =    0 RPM)
fan2:                  1365 RPM  (min =    0 RPM)
fan3:                     0 RPM  (min =    0 RPM)
fan4:                     0 RPM  (min =    0 RPM)
fan5:                     0 RPM  (min =    0 RPM)
SYSTIN:                +115.0°C  (high =  +0.0°C, hyst =  +0.0°C)  sensor = thermistor
CPUTIN:                 +50.5°C  (high = +80.0°C, hyst = +75.0°C)  sensor = thermistor
AUXTIN0:                +42.0°C  (high =  +0.0°C, hyst =  +0.0°C)  ALARM  sensor = thermistor
AUXTIN1:               +109.0°C    sensor = thermistor
AUXTIN2:               +107.0°C    sensor = thermistor
AUXTIN3:               +106.0°C    sensor = thermistor
SMBUSMASTER 0:          +45.5°C  
PCH_CHIP_CPU_MAX_TEMP:   +0.0°C  
PCH_CHIP_TEMP:           +0.0°C  
PCH_CPU_TEMP:            +0.0°C  
intrusion0:            OK
intrusion1:            ALARM
beep_enable:           disabled`

I think the results from SYSINT and AUXTIN 1-3 are possible conversion errors, bc converted to °C they nearly match the other results.

olysonek, 2020-07-20:
Hi.

lm_sensors doesn't do any conversion on its own. I suggest to check lm_sensors configuration files to see if there are any `compute` statements that could cause this.

If there aren't any and you believe the conversion error should get corrected globally in the hwmon driver, feel free to report it to the hwmon subsystem.

Otherwhise, if it's a board specific issue (this is most likely the case), I suggest to add your own `compute` statements a config file to fix it for your case.

bmn-m, 2022-01-27:
thanks. ill try.

closed 2022-01-27 by bmn-m