lm-sensors/lm-sensors #261: Support for additional (AMD)GPU metrics like frequency and power?

simonvanderveldt, 2020-08-02:
Hi! I was wondering if it would be possible to add GPU frequency support to lm-sensors?
AFAIK lm-sensors only uses the hwmon subsystem and the AMDGPU module nowadays exposes several additional metrics via hwmon, they don't show up in sensors-detect though.

These are the sysfs hwmon files
```
# GPU clock
$ cat /sys/class/drm/card0/device/hwmon/hwmon4/freq1_input
351590000
# Memory clock
$ cat /sys/class/drm/card0/device/hwmon/hwmon4/freq2_input 
300000000
# Power
$ cat /sys/class/drm/card0/device/hwmon/hwmon4/power1_average 
9103000
# Voltage
$ cat /sys/class/drm/card0/device/hwmon/hwmon4/in0_input 
750
```

olysonek, 2020-08-03:
Hi, is this just about `sensors-detect` or also `sensors`? Can you post the output of `sensors-detect` and `sensors -u`? What kind of GPU is it?

simonvanderveldt, 2020-08-03:
Hi @olysonek Thanks for the reply.
This if for an RX 480, although I expect the same applies to all GPUs that use the amdgpu kernel driver.
It's mainly about being able to monitor the frequencies, so command-line that would be `sensors` (although I'm using collectd's lm-sensors input, which I assume uses a libsensors).

I just noticed `power1_average` and `in0_input` are already shown, sorry about that. I've removed them from the initial description.
The GPU and memory clock aren't shown though.

Here are the outputs you asked for. Note that `sensors-detect` doesn't really show much about the amdgpu kernel module, it's already loaded though, so not sure if that's intentional or not. `sensors` show it just fine though.
```
$ sudo sensors-detect 
# sensors-detect version 3.6.0
# Board: ASUSTeK COMPUTER INC. P8Z77-I DELUXE
# Kernel: 5.4.52 x86_64
# Processor: Intel(R) Core(TM) i5-3570K CPU @ 3.40GHz (6/58/9)

This program will help you determine which kernel modules you need
to load to use lm_sensors most effectively. It is generally safe
and recommended to accept the default answers to all questions,
unless you know what you're doing.

Gentoo config file protection is active. Every file this program will
modify must be merged before the change will become active using
default Gentoo tools such as dispatch-conf, cfg-update, and etc-update.

Some south bridges, CPUs or memory controllers contain embedded sensors.
Do you want to scan for them? This is totally safe. (YES/no): 
Silicon Integrated Systems SIS5595...                       No
VIA VT82C686 Integrated Sensors...                          No
VIA VT8231 Integrated Sensors...                            No
AMD K8 thermal sensors...                                   No
AMD Family 10h thermal sensors...                           No
AMD Family 11h thermal sensors...                           No
AMD Family 12h and 14h thermal sensors...                   No
AMD Family 15h thermal sensors...                           No
AMD Family 16h thermal sensors...                           No
AMD Family 17h thermal sensors...                           No
AMD Family 15h power sensors...                             No
AMD Family 16h power sensors...                             No
Hygon Family 18h thermal sensors...                         No
Intel digital thermal sensor...                             Success!
    (driver `coretemp')
Intel AMB FB-DIMM thermal sensor...                         No
Intel 5500/5520/X58 thermal sensor...                       No
VIA C7 thermal sensor...                                    No
VIA Nano thermal sensor...                                  No

Some Super I/O chips contain embedded sensors. We have to write to
standard I/O ports to probe them. This is usually safe.
Do you want to scan for Super I/O sensors? (YES/no): 
Probing for Super-I/O at 0x2e/0x2f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               Yes
Found `Nuvoton NCT5532D/NCT6779D Super IO Sensors'          Success!
    (address 0x290, driver `nct6775')
Probing for Super-I/O at 0x4e/0x4f
Trying family `National Semiconductor/ITE'...               No
Trying family `SMSC'...                                     No
Trying family `VIA/Winbond/Nuvoton/Fintek'...               No
Trying family `ITE'...                                      No

Some systems (mainly servers) implement IPMI, a set of common interfaces
through which system health data may be retrieved, amongst other things.
We first try to get the information from SMBIOS. If we don't find it
there, we have to read from arbitrary I/O ports to probe for such
interfaces. This is normally safe. Do you want to scan for IPMI
interfaces? (YES/no): 
Probing for `IPMI BMC KCS' at 0xca0...                      No
Probing for `IPMI BMC SMIC' at 0xca8...                     No

[... 121 lines cut ...]
  temp1_input: 46.000
  temp1_max: 85.000
  temp1_crit: 105.000
  temp1_crit_alarm: 0.000
core 0:
  temp2_input: 46.000
  temp2_max: 85.000
  temp2_crit: 105.000
  temp2_crit_alarm: 0.000
core 1:
  temp3_input: 33.000
  temp3_max: 85.000
  temp3_crit: 105.000
  temp3_crit_alarm: 0.000
core 2:
  temp4_input: 35.000
  temp4_max: 85.000
  temp4_crit: 105.000
  temp4_crit_alarm: 0.000
core 3:
  temp5_input: 37.000
  temp5_max: 85.000
  temp5_crit: 105.000
  temp5_crit_alarm: 0.000
```

olysonek, 2020-08-03:
> Note that `sensors-detect` doesn't really show much about the amdgpu kernel module, it's already loaded though, so not sure if that's intentional or not.

That's expected. The module probably autoloads. Regarding sensors-detect, I think there was an issue filed already about the amdgpu detection that I haven't yet got around to.

Thanks for the report. I'll do my best. Patches are welcome.

reinerh, 2023-01-02:
A patch for this is available in #347.