lm-sensors/lm-sensors #264: values are way too low to be real

Moth-Tolias, 2020-08-12:
hi! i seem to be having some trouble with sensors, because it seems to be saying my computer is only 11 degrees? that would be really impressive if it was true, because it's a fan-cooled machine and room temperature is as hot as hell right now here in the uk.

here's the output i get:

> fam15h_power-pci-00c4
> Adapter: PCI adapter
> power1:           N/A  (crit =  65.19 W)
> 
> radeon-pci-0008
> Adapter: PCI adapter
> temp1:        +11.0°C  (crit = +120.0°C, hyst = +90.0°C)
> 
> k10temp-pci-00c3
> Adapter: PCI adapter
> temp1:        +11.1°C  (high = +70.0°C)
>                        (crit = +70.0°C, hyst = +69.0°C)

what should i do?

also, what do the values in parentheses mean? i'm guessing it's like, maximum safe values, but why are there two or three of them?

leighbb, 2020-08-12:
Hello,

According to [this article](https://www.overclock.net/forum/10-amd-cpus/1128821-amd-temp-information-guide.html), the k10temp readings are not "real" temperatures, but "a non-physical temperature on an arbitrary scale measured in degrees", which is not terribly helpful.  I don't know about the graphics card, but since it is an AMD card, it could be the same.

Regarding the value in parentheses, they indicate values above which the component is too hot.  The "hyst" value is the hysteresis value. The reading must fall below that value before the temperature of the component can be considered to be okay again.

I'm not sure why k10temp has both a "high" and a "crit" value.

Sometimes it is informative to put some load on the system and see what happens to the temperatures.  If they go up when the system is under load and come down again when idle, it's showing you something at least!

Regards,

Leigh.

RonnySvedman, 2020-11-14:
perhaps it needs to be scaled in /sensors3.conf ? 11 is half of 22 and a third of 33 or a 4th of 44.  33 is a believable value for an idle graphics card? Play a game and see if the reading changes, and you might find out the right scaling to map the "unitless value" to approximate degrees.