lm-sensors/lm-sensors #268: MSI MPG B550 Gaming Carbon

mdx0111, 2020-08-20:
Hi,

I was wondering if you have any plan on providing configs for reading temperature sensors of MSI MPG B550 Gaming Carbon motherboard please.

Many thanks.

Regards.

olysonek, 2020-08-20:
Nope. We rely on users like you to provide configs ;).

closed 2020-08-20 by olysonek

mdx0111, 2020-08-20:
Thanks for your quick response. I think part of the issue is that the chip nct6687d isn't supported by lm sensors yet. I wonder if you could point me in the write direction to be able to contribute to get this chip added and supported please.

Many thanks.

mdx0111, 2020-08-20:
I'm happy to run any script to collect the required data to make this possible. It'd be great if you could help to add support for this chip too.

Many thanks.

olysonek, 2020-08-20:
> I think part of the issue is that the chip nct6687d isn't supported by lm sensors yet. I wonder if you could point me in the write direction to be able to contribute to get this chip added and supported please.

The chip is not supported by the kernel. There's nothing I  can do about that in lm_sensors.

Try searching the linux-hwmon mailing list archives or writing to that list.
http://vger.kernel.org/vger-lists.html#linux-hwmon

mdx0111, 2020-08-20:
Brilliant.

Thanks a lot. :)

tuxick, 2021-08-14:
Good news: turns out it's about chip Nuvoton NCT6687-R, this module works: https://github.com/Fred78290/nct6687d

mdx0111, 2021-08-14:
> Good news: turns out it's about chip Nuvoton NCT6687-R, this module works: https://github.com/Fred78290/nct6687d

It's not neede anymore. Forced the driver provided by the kernel enabled and everything works like a treat.