lm-sensors/lm-sensors #274: sensord dies on mk aarch64

necrose99, 2020-08-30:
[ebuild   R    ] sys-apps/lm-sensors-3.6.0:0/5.0.0::gentoo  USE="contrib* static-libs* -sensord" 0 KiB
 (however will build just not demon for now ok...   so for now i have to shift to no use of sensord. 
[build-logs.zip](https://github.com/lm-sensors/lm-sensors/files/5147456/build-logs.zip)
)

```
/usr/lib/gcc/aarch64-unknown-linux-gnu/10.1.0/../../../../aarch64-unknown-linux-gnu/bin/ld: /usr/lib/gcc/aarch64-unknown-linux-gnu/10.1.0/../../../../lib64/librrd.so: undefined reference to `dbi_shutdown'
collect2: error: ld returned 1 exit status
make: *** [prog/sensord/Module.mk:45: prog/sensord/sensord] Error 1
 * ERROR: sys-apps/lm-sensors-3.6.0::gentoo failed (compile phase):
 *   emake failed
```

olysonek, 2020-08-31:
Have you reported it to the Gentoo maintainer(s)? It doesn't look like an upstream issue to me.

necrose99, 2020-08-31:
that i can do , if it hasn't already.. just checking if sensord /arm64 was reported prior on any platform/distro .

olysonek, 2020-09-01:
I don't recall this having been reported.