lm-sensors/lm-sensors #279: Ryzen 9 3900x & MSI MPG-x570 Gaming MB (Dead Because of THIS SOFTWARE)

incryptx, 2020-09-24:
After installing this software, I now have to RMA my motherboard. I installed this and Psensor and rebooted, only to find my bios (nvram) was toast, No ability to see the bios screen loading devices which was bypassed and after one minute, I was able to boot back to Ubuntu 20.04.

After going into Grub and forcing grub to boot to firmware (UEFI Bios) the system will not boot at all. No ability to get into bios, no OS will load now. It's toast.

I went to my USB thumbdrive, downloaded the current bios, but it on the USB thumbdrive and using the Flash button on the motherboard, reloaded the bios four times to no avail. I did answer "Yes" to all of the questions and it looks like SMBus toasted it.

My motherboard is toast. I now have to RMA it back to MSI,

Before I do, you have any ideas, because I don't know what the problem is, but reloading the bios does nothing?

Frankly, I am livid.

BEWARE of this software.

NostalgiaRunner, 2020-09-24:
Sorry to hear that. That is shocking behavior, as I believe most motherboards these days come with a dual-bios system. Did you install lm-sensors from the (I'm assuming) Ubuntu 20 apt? Or did you build and compile the latest Github release? You are running Ubuntu 20, correct? Or did you just boot into it for testing.

incryptx, 2020-09-30:
Hello again.

I RMA-ed the Motherboard and still the system will not go into the bios at all. It will boot to a Ubuntu DVD install disk, but no matter what I do, it will not go into bios. I don't even get the visual boot up sequence showing the version and what is connected, just a black screen until Ubuntu loads. I think your "lm-sensors" application has done something to the Ryzen 9 3900x processor, as I have tried a different graphics card and memory. I also took out the NvMe module and nothing will allow the system to go into bios.

You have any ideas?

My next step was to boot into a Windows install DVD and install it somewhere where I could install the Ryzen Master software, but the system will not even boot on a Windows install DVD. Why it will boot on a Ubuntu boot DVD and not a Windows install DVD has me proplexed. Is there any way to reverse what LM-sensors did?

incryptx, 2020-09-30:
I am contacting AMD Technical Support and letting them know this software may have toasted my processor and see if they have any ideas before I RMA the processor. I can't figure out why I can boot on Ubuntu and use it if the processor were bad, as I am now typing this on Ubuntu with the affected processor and a new Motherboard.

incryptx, 2020-10-05:
After receiving an RMA replacement for my motherboard, the system still would not post. I upgraded the bios to the current version using the flashback process for this motherboard and the system would still not post. Fortunately the default for the bios is to return to CMS as opposed to UEFI so I was able to connect a SSD and reinstall Ubuntu 20.04. After doing this, I continually installed and uninstalled lm-sensors and after a week of doing this, the system finally posted and I was able to get back into the bios. 

I have no clue or idea why lm-sensors caused the problem or what lm-sensors did to the system to fix the problem, but be warned that installing lm-sensors on a x570 chipset with a Ryzen processor will cause problems. I suspect the problem is a low level hardware issue within the SMBus and one indication I got was AMD KERNCZ and the IC2-Pii4 and the SPD EEPROM. These indications came up in running "hardinfo" and several CPU stress tests applications I ran to see what the CPU was doing.

The system has returned to normal operation but I am unable to determine what caused the failure or what lead to the solution except running several CPU type stress applications.

MartinJM, 2020-12-08:
> but be warned that installing lm-sensors on a x570 chipset with a Ryzen processor will cause problems.

I haven't had any problems with a 3900X on a Gigabyte X570 Aorus Master. So I don't think the problem is only due to the CPU/chipset combination with this software.

Some system/version info:
|      | Version | Comment |
| ---- | ------- | ------- |
| CPU  | Ryzen 9 3900X | |
| MB   | Gigabyte X570 Aorus Master rev 1.0 | |
| BIOS | F22     | Although I haven't had any problems on older versions either. |
| OS   | Fedora Workstation 33 | It worked fine on 32 as well. |
| Sensors | 3.6.0 | Installed from repo's using `dnf`. |

raynalm, 2021-01-21:
>  but be warned that installing lm-sensors on a x570 chipset with a Ryzen processor will cause problems.

No issues to report (beyond not seeing all temps because I'm on 5.8 kernel) on this config:

  | Version | Comment
-- | -- | --
CPU | Ryzen 9 5900X |  
GPU | RTX 3070 FE |
MB | Asus ROG Strix X570 E |  
BIOS | 3001 | last stable version; PBO & DOCP activated |
OS | Ubuntu 20.04 LTS |  kernel 5.8.0-38-generic |
Sensors | 3.6.0 | Installed from repo |

If your report remains isolated, I'd start looking into other possible causes, especially if, as I think I understood, you toasted already two MBs. Did you try swapping hardware devices too (DDRAM, GPU, PSU, drives ...) ?