lm-sensors/lm-sensors #282: Unexpected Output

nurbles, 2020-10-05:
One of the webmin developers directed me here because (according to him) the output from **sensors** is "odd."  He is trying to have something automatically pull the CPU/core temperatures (and ranges, I think) from the sensors output.  The output from my computer (an old Dell Precision 390) is confusing him:

    > sensors
    adt7473-i2c-2-2e
    Adapter: nvkm-0000:01:00.0-bus-0002
    in1:          +3.00 V  (min =  +0.00 V, max =  +2.99 V)
    +3.3V:        +3.29 V  (min =  +2.96 V, max =  +3.63 V)
    fan1:        1129 RPM  (min =    0 RPM)
    fan2:           0 RPM  (min =    0 RPM)
    fan3:           0 RPM  (min =  164 RPM)  ALARM
    fan4:           0 RPM  (min =    0 RPM)
    temp1:        +47.2 C  (low  = +72.0 C, high = +92.0 C)  ALARM
                           (crit = +191.0 C, hyst = +189.0 C)
    Board Temp:   +45.5 C  (low  = +20.0 C, high = +60.0 C)
                           (crit = +100.0 C, hyst = +96.0 C)
    temp3:        +47.2 C  (low  = +80.0 C, high = +105.0 C)  ALARM
                           (crit = +100.0 C, hyst = +96.0 C)

    coretemp-isa-0000
    Adapter: ISA adapter
    Core 0:       +58.0 C  (high = +84.0 C, crit = +100.0 C)
    Core 1:       +61.0 C  (high = +84.0 C, crit = +100.0 C)

    nouveau-pci-0100
    Adapter: PCI adapter
    GPU core:     +1.30 V  (min =  +1.30 V, max =  +1.40 V)
    temp1:        +48.0 C  (high = +95.0 C, hyst =  +3.0 C)
                           (crit = +110.0 C, hyst =  +2.0 C)
                           (emerg = +135.0 C, hyst =  +5.0 C)

    dell_smm-virtual-0
    Adapter: Virtual device
    Processor Fan:   1025 RPM
    Motherboard Fan: 1243 RPM

Apparently, it is difficult (for the code) to determine which temperatures are the CPU temps.  It keeps listing _temp1_, _temp3_, _Core 0_ and _Core 1_ as the **four** CPU temps for my **TWO** core system.  If there is any info I can collect from the system for you, please let me know.

olysonek, 2020-10-07:
A quick search suggests that `adt7473` (with the temp1 and temp3) might be a chip on the GPU. Is that a possibility?

Anyway, coretemp gives CPU temperatures, so I'd rely on that.

Also note that `sensors -u` or `sensors -j` will give you a more easily parseable output.

nurbles, 2020-10-07:
The machine does have an nVidia GeForce 7900 GTX video adapter, so you are likely correct about the `adt7473` chip.

`sensors -u` may be easier to parse (and it may be what he uses) but the actual information it provides is (obviously) identical to the formatted output.  I think he's trying to find a general way to successfully extract the CPU/core temp info from the output for all systems that report them.  Apparently there's some inconsistency there that requires him to consider other sections that include "temp" or "core".

`sensors -j` is a syntax error in my version:

    root@mybox:~# sensors -v
    sensors version 3.4.0 with libsensors version 3.4.0
    root@mybox:~# sensors -j
    sensors: invalid option -- 'j'
    Try `sensors -h' for more information