lm-sensors/lm-sensors #283: nct6775 PECI vs coretemp

yutayu, 2020-10-09:
Hi.
My uses drivers of nct6775 and coretemp.
I want to know which cpu temperature reports is correct .

nct6775 uses PECI.
intel drier is coretemp.

nct6776-isa-0a30
Adapter: ISA adapter
Vcore:
  in0_input: 0.704
  in0_min: 0.248
  in0_max: 1.528
  in0_alarm: 0.000
  in0_beep: 0.000
in1:
  in1_input: 1.224
  in1_min: 1.080
  in1_max: 1.320
  in1_alarm: 0.000
  in1_beep: 0.000
AVCC:
  in2_input: 3.312
  in2_min: 2.976
  in2_max: 3.632
  in2_alarm: 0.000
  in2_beep: 0.000
+3.3V:
  in3_input: 3.312
  in3_min: 2.976
  in3_max: 3.632
  in3_alarm: 0.000
  in3_beep: 0.000
in4:
  in4_input: 1.024
  in4_min: 0.896
  in4_max: 1.104
  in4_alarm: 0.000
  in4_beep: 0.000
in5:
  in5_input: 1.008
  in5_min: 0.896
  in5_max: 1.104
  in5_alarm: 0.000
  in5_beep: 0.000
in6:
  in6_input: 0.704
  in6_min: 0.000
  in6_max: 0.000
  in6_alarm: 1.000
  in6_beep: 0.000
3VSB:
  in7_input: 3.328
  in7_min: 2.976
  in7_max: 3.632
  in7_alarm: 0.000
  in7_beep: 0.000
Vbat:
  in8_input: 3.232
  in8_min: 2.704
  in8_max: 3.632
  in8_alarm: 0.000
  in8_beep: 0.000
fan1:
  fan1_input: 982.000
  fan1_min: 0.000
  fan1_alarm: 0.000
  fan1_beep: 0.000
  fan1_pulses: 2.000
fan2:
  fan2_input: 792.000
  fan2_min: 0.000
  fan2_alarm: 0.000
  fan2_beep: 0.000
  fan2_pulses: 2.000
fan3:
  fan3_input: 5744.000
  fan3_min: 0.000
  fan3_alarm: 0.000
  fan3_beep: 0.000
  fan3_pulses: 2.000
fan4:
  fan4_input: 0.000
  fan4_min: 0.000
  fan4_alarm: 0.000
  fan4_beep: 0.000
  fan4_pulses: 3.000
SYSTIN:
  temp1_input: 33.000
  temp1_max: 65.000
  temp1_max_hyst: 0.000
  temp1_alarm: 0.000
  temp1_type: 4.000
  temp1_offset: 0.000
  temp1_beep: 0.000
CPUTIN:
  temp2_input: 123.500
  temp2_max: 80.000
  temp2_max_hyst: 75.000
  temp2_alarm: 1.000
  temp2_type: 1.000
  temp2_offset: 0.000
  temp2_beep: 0.000
AUXTIN:
  temp3_input: 49.000
  temp3_max: 80.000
  temp3_max_hyst: 75.000
  temp3_alarm: 0.000
  temp3_type: 4.000
  temp3_offset: 0.000
  temp3_beep: 0.000
PECI Agent 0:
  temp7_input: 39.000
  temp7_max: 80.000
  temp7_max_hyst: 75.000
  temp7_crit: 100.000
  temp7_beep: 0.000
PCH_CHIP_TEMP:
  temp8_input: 0.000
PCH_CPU_TEMP:
  temp9_input: 0.000
PCH_MCH_TEMP:
  temp10_input: 0.000
intrusion0:
  intrusion0_alarm: 1.000
  intrusion0_beep: 0.000
intrusion1:
  intrusion1_alarm: 1.000
  intrusion1_beep: 0.000
beep_enable:
  beep_enable: 0.000

acpitz-acpi-0
Adapter: ACPI interface
temp1:
  temp1_input: 27.800
  temp1_crit: 103.000
temp2:
  temp2_input: 29.800
  temp2_crit: 103.000

coretemp-isa-0000
Adapter: ISA adapter
Package id 0:
  temp1_input: 42.000
  temp1_max: 82.000
  temp1_crit: 102.000
  temp1_crit_alarm: 0.000
Core 0:
  temp2_input: 43.000
  temp2_max: 82.000
  temp2_crit: 102.000
  temp2_crit_alarm: 0.000
Core 1:
  temp3_input: 42.000
  temp3_max: 82.000
  temp3_crit: 102.000
  temp3_crit_alarm: 0.000

yutayu, 2020-10-09:
https://www.kernel.org/doc/Documentation/hwmon/nct6775

Usage Notes
-----------

On various ASUS boards with NCT6776F, it appears that CPUTIN is not really
connected to anything and floats, or that it is connected to some non-standard
temperature measurement device. As a result, the temperature reported on CPUTIN
will not reflect a usable value. It often reports unreasonably high
temperatures, and in some cases the reported temperature declines if the actual
temperature increases (similar to the raw PECI temperature value - see PECI
specification for details). CPUTIN should therefore be be ignored on ASUS
boards. The CPU temperature on ASUS boards is reported from PECI 0.

My mother board is not ASUS.

JiiPee74, 2020-11-05:
PECI Agent 0:
most likely CPU socket temp

Package id 0:
CPU packege temp

and then 2 core temps.

They are all correct, socket temp in lower than package/core temps because temp sensor is actually located on motherboard, so it won't heat up as much as sensors on CPU package or inside on cores.

yutayu, 2020-11-05:
@JiiPee74 
Thanks for your reply , Okay , I  will pick package temp.

closed 2020-11-05 by yutayu