lm-sensors/lm-sensors #284: asus-isa-0000 shows Fan Speed but pwmconfig doesn't set the speed

FranceBB, 2020-10-11:
Hi there,
I've recently upgraded from kernel 5.4.70 to kernel 5.8.14, but there's a problem: I can no longer control the fan speed on my Asus VivoBook Pro N552VX.
Unfortunately, I can't roll back to the old kernel as I upgraded from Fedora 32 to Fedora 33, so I'm stuck with kernel 5.8.14.
This is the output of sensors-detect: https://pastebin.com/HvKi18Pi
and this is the list of sensors:
`
[FranceBB@router-localhost ~]$ sensors
iwlwifi_1-virtual-0
Adapter: Virtual device
temp1:            N/A  

asus-isa-0000
Adapter: ISA adapter
**CPU Fan:        2400 RPM**
NVIDIA GTX950M:  +58.0°C  

nouveau-pci-0100
Adapter: PCI adapter
NVIDIA GTX950M: 900.00 mV (min =  +0.60 V, max =  +1.20 V)

BAT0-acpi-0
Adapter: ACPI interface
Battery Voltage:  14.40 V  

pch_skylake-virtual-0
Adapter: Virtual device
Platform Controller Hub:  +54.0°C  

nvme-pci-0500
Adapter: PCI adapter
Intel SSD NVME:  +34.9°C  (low  =  -0.1°C, high = +76.8°C)
                          (crit = +79.8°C)

acpitz-acpi-0
Adapter: ACPI interface
Motherboard:  +58.0°C  (crit = +103.0°C)
`

And we can clearly see the speed of the CPU fan (which is correct and updates itself automatically), however, when I try to run pwmconfig I get:

`
FranceBB@router-localhost ~]$ sudo pwmconfig
[sudo] password for FranceBB: 
# pwmconfig version 3.6.0
This program will search your sensors for pulse width modulation (pwm)
controls, and test each one to see if it controls a fan on
your motherboard. Note that many motherboards do not have pwm
circuitry installed, even if your sensor chip supports pwm.

We will attempt to briefly stop each fan using the pwm controls.
The program will attempt to restore each fan to full speed
after testing. However, it is ** very important ** that you
physically verify that the fans have been to full speed
after the program has completed.

**/usr/sbin/pwmconfig: There are no pwm-capable sensor modules installed**
`
Why is that?

FranceBB, 2020-10-15:
By opening Nautilus as root (sudo nautilus) and navigating to /sys/devices/platform/asus-nb-wmi/hwmon/hwmon6 I was able to modify the file called "pwm1_enable" and set it from 2 to 0 and it worked with the Fan now spinning at maximum speed (25500RPM). So... it's weird 'cause everything works but pwmconfig doesn't seem to detect it...

RonnySvedman, 2020-11-14:
if you set the pwm to 0 does the reading freeze where it was or go to 0 ? (dont forget to start it up again quickly so you dont melt your cpu)
on my radeon rx580 the reading freezes when pwmconfig sets zero rpm to check which controls which, and thus the pwmconfig gets confused. I had to manually craft a /etc/fancontrol for my machine, with a >0 MINPWM and >0 MINSTOP for this reason after upgrading to 20.04. (copied my old one from a backup)

sirfoga, 2021-03-04:
> if you set the pwm to 0 does the reading freeze where it was or go to 0 ? (dont forget to start it up again quickly so you dont melt your cpu)
> on my radeon rx580 the reading freezes when pwmconfig [...]

@RonnySvedman  can you share your `/etc/fancontrol` ?

RonnySvedman, 2021-03-04:
[fancontrol_z97p.txt](https://github.com/lm-sensors/lm-sensors/files/6085751/fancontrol_z97p.txt)
This is what worked on my asus z97p with a 8GB ddr5 Sapphire Pulse RX580 OC lite.

The gfx card was the cause iof pwmconfig script failing, since it stops reporting RPM immediately and freezes the reported number when the PWM is set below 21 ( Where the fan physically stops), sp pwmconfig never understands that the fan has stopped, an then it will mix up what controls what, reaulting in no output file at all. My solution was to unplug the graphics card, re-run pwmconfig without it using the on-cpu graphics. this resulted in a pwmconfig tha worked for the front, top and cpu fans (motherboard connected fans). with the unofficial it87 kernel module compiled on my machine. At this point the graphics card was following the factory curve withour control from lmsensors. Then I manually found out which pwm device file controlled the gfx card fan and edited the /etc/fancontrol file accordingly, and the attached file is the result. I no longer have this motherboard since I destroyed the cpu socket by accident when remounting the cpu, so I can't really help with research about the it8606, but I do have the RX580 installed in another (older) motherboard and it has the same behavior here - I have to manually edit the /etc/fancontrol to gain control over the graphics card fans, but this time i do't need any unofficial module -this motherboard chipset is supported in the generic kernel. The RX580 was bought new in store in october 2020, although it had been on shelf a while and was on sale.

RonnySvedman, 2021-03-04:
Also, if you set theh pwm_enable file to 1, the corresponding pwm should become writeable for manual control, write a number betwwen 0-255 there. thsi is what fancontrol does once its configured right. 2 in ghe pwmX_enable means "automatic" or bios control, 1 means teh value in pwmX decides, and 0 means full power, no control

FranceBB, 2021-03-06:
Thanks Ronny for sharing your experience.
For what is worth, I've been able to set 
sudo bash -c 'echo 0 > /sys/devices/platform/asus-nb-wmi/hwmon/hwmon6/pwm1_enable'
and apply it every time I boot to make the fan spin at maximum and that's enough for me, actually. ;)

closed 2021-03-06 by FranceBB