lm-sensors/lm-sensors #285: Z490 Phantom Gaming-ITX/TB3 (NCT6683D) - only CPU/nvme temperatures are shown

mikeyo, 2020-10-20:
Nuvoton chip and only CPU and ISA/nvme temps shown, no motherboard temp or voltages.

olysonek, 2020-10-21:
Have you run `sensors-detect`? What's the output? And what's the output of `sensors`? Is the `nct6683` driver loaded?

mikeyo, 2020-10-21:
> 
> 
> Have you run `sensors-detect`? What's the output? And what's the output of `sensors`? Is the `nct6683` driver loaded?

Here is my sensors output. Driver loaded OK.

Found `Nuvoton NCT6683D eSIO' Success!

(address 0xa20, driver `nct6683')

-----------------------------------------------------------

# sensors-detect version 3.6.0

# Board: ASRock Z490 Phantom Gaming-ITX/TB3

# Processor: Intel(R) Core(TM) i9-10850K CPU @ 3.60GHz (6/165/5)

Silicon Integrated Systems SIS5595... No

VIA VT82C686 Integrated Sensors... No

VIA VT8231 Integrated Sensors... No

AMD K8 thermal sensors... No

AMD Family 10h thermal sensors... No

AMD Family 11h thermal sensors... No

AMD Family 12h and 14h thermal sensors... No

AMD Family 15h thermal sensors... No

AMD Family 16h thermal sensors... No

AMD Family 17h thermal sensors... No

AMD Family 15h power sensors... No

AMD Family 16h power sensors... No

Hygon Family 18h thermal sensors... No

Intel digital thermal sensor... Success!

(driver `coretemp')

Intel AMB FB-DIMM thermal sensor... No

Intel 5500/5520/X58 thermal sensor... No

VIA C7 thermal sensor... No

VIA Nano thermal sensor... No

Some Super I/O chips contain embedded sensors. We have to write to

standard I/O ports to probe them. This is usually safe.

Do you want to scan for Super I/O sensors? (YES/no):

Probing for Super-I/O at 0x2e/0x2f

Trying family `National Semiconductor/ITE'... No

Trying family `SMSC'... No

Trying family `VIA/Winbond/Nuvoton/Fintek'... Yes

Found `Nuvoton NCT6683D eSIO' Success!

(address 0xa20, driver `nct6683')

Probing for Super-I/O at 0x4e/0x4f

Trying family `National Semiconductor/ITE'... No

Trying family `SMSC'... No

Trying family `VIA/Winbond/Nuvoton/Fintek'... No

Trying family `ITE'... No

olysonek, 2020-10-21:
Hm, that's odd. How many chips are there exported in `/sys/class/hwmon/`? What are their names (`cat /sys/class/hwmon/*/name`)? If there's no nct chip, maybe share your dmesg so that we can see if the driver has logged anything.

mikeyo, 2020-10-21:
> 
> 
> Hm, that's odd. How many chips are there exported in `/sys/class/hwmon/`? What are their names (`cat /sys/class/hwmon/*/name`)? If there's no nct chip, maybe share your dmesg so that we can see if the driver has logged anything.

Sure, here you go....
---------------------------------
rwxrwxrwx  1 root root 0 Oct 21 10:29 hwmon0 -> ../../devices/pci0000:00/0000:00:1b.0/0000:01:00.0/hwmon/hwmon0/
lrwxrwxrwx  1 root root 0 Oct 21 10:29 hwmon1 -> ../../devices/pci0000:00/0000:00:1d.0/0000:3d:00.0/hwmon/hwmon1/
lrwxrwxrwx  1 root root 0 Oct 21 10:29 hwmon2 -> ../../devices/platform/coretemp.0/hwmon/hwmon2/

at /sys/class/hwmon/*/name
nvme
nvme
coretemp

------------
dmesg

[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000005cfff] usable
[    0.000000] BIOS-e820: [mem 0x000000000005d000-0x000000000005dfff] reserved
[    0.000000] BIOS-e820: [mem 0x000000000005e000-0x000000000009ffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000000a0000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x000000003fffffff] usable
[    0.000000] BIOS-e820: [mem 0x0000000040000000-0x00000000403fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000040400000-0x000000007561efff] usable
[    0.000000] BIOS-e820: [mem 0x000000007561f000-0x000000007561ffff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x0000000075620000-0x0000000075642fff] usable
[    0.000000] BIOS-e820: [mem 0x0000000075643000-0x0000000075643fff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000075644000-0x00000000757ebfff] usable
[    0.000000] BIOS-e820: [mem 0x00000000757ec000-0x0000000076dc6fff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000076dc7000-0x0000000076ef6fff] ACPI data
[    0.000000] BIOS-e820: [mem 0x0000000076ef7000-0x00000000773a6fff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x00000000773a7000-0x0000000077ffefff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000077fff000-0x0000000077ffffff] usable
[    0.000000] BIOS-e820: [mem 0x0000000078000000-0x00000000897fffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000e0000000-0x00000000efffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fe000000-0x00000000fe010fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fec00000-0x00000000fec00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fee00000-0x00000000fee00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ff000000-0x00000000ffffffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x00000010747fffff] usable
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] e820: update [mem 0x65fcf018-0x65fdf057] usable ==> usable
[    0.000000] e820: update [mem 0x65fcf018-0x65fdf057] usable ==> usable
[    0.000000] e820: update [mem 0x65fc1018-0x65fce857] usable ==> usable
[    0.000000] e820: update [mem 0x65fc1018-0x65fce857] usable ==> usable
[    0.000000] extended physical RAM map:
[    0.000000] reserve setup_data: [mem 0x0000000000000000-0x000000000005cfff] usable
[    0.000000] reserve setup_data: [mem 0x000000000005d000-0x000000000005dfff] reserved
[    0.000000] reserve setup_data: [mem 0x000000000005e000-0x000000000009ffff] usable
[    0.000000] reserve setup_data: [mem 0x00000000000a0000-0x00000000000fffff] reserved
[    0.000000] reserve setup_data: [mem 0x0000000000100000-0x000000003fffffff] usable
[    0.000000] reserve setup_data: [mem 0x0000000040000000-0x00000000403fffff] reserved
[    0.000000] reserve setup_data: [mem 0x0000000040400000-0x0000000065fc1017] usable
[    0.000000] reserve setup_data: [mem 0x0000000065fc1018-0x0000000065fce857] usable
[    0.000000] reserve setup_data: [mem 0x0000000065fce858-0x0000000065fcf017] usable
[    0.000000] reserve setup_data: [mem 0x0000000065fcf018-0x0000000065fdf057] usable
[    0.000000] reserve setup_data: [mem 0x0000000065fdf058-0x000000007561efff] usable
[    0.000000] reserve setup_data: [mem 0x000000007561f000-0x000000007561ffff] ACPI NVS
[    0.000000] reserve setup_data: [mem 0x0000000075620000-0x0000000075642fff] usable
[    0.000000] reserve setup_data: [mem 0x0000000075643000-0x0000000075643fff] reserved
[    0.000000] reserve setup_data: [mem 0x0000000075644000-0x00000000757ebfff] usable
[    0.000000] reserve setup_data: [mem 0x00000000757ec000-0x0000000076dc6fff] reserved
[    0.000000] reserve setup_data: [mem 0x0000000076dc7000-0x0000000076ef6fff] ACPI data
[    0.000000] reserve setup_data: [mem 0x0000000076ef7000-0x00000000773a6fff] ACPI NVS
[    0.000000] reserve setup_data: [mem 0x00000000773a7000-0x0000000077ffefff] reserved
[    0.000000] reserve setup_data: [mem 0x0000000077fff000-0x0000000077ffffff] usable
[    0.000000] reserve setup_data: [mem 0x0000000078000000-0x00000000897fffff] reserved
[    0.000000] reserve setup_data: [mem 0x00000000e0000000-0x00000000efffffff] reserved
[... 921 lines cut ...]
[   42.109838] virbr0: port 1(virbr0-nic) entered blocking state
[   42.109839] virbr0: port 1(virbr0-nic) entered disabled state
[   42.109862] device virbr0-nic entered promiscuous mode
[   42.121107] cgroup: cgroup: disabling cgroup2 socket matching due to net_prio or net_cls activation
[   42.182826] virbr0: port 1(virbr0-nic) entered blocking state
[   42.182827] virbr0: port 1(virbr0-nic) entered listening state
[   42.191253] virbr0: port 1(virbr0-nic) entered disabled state
[   42.209235] docker0: port 1(veth7d2d15a) entered blocking state
[   42.209236] docker0: port 1(veth7d2d15a) entered disabled state
[... cut ...]
[   42.349551] IPv6: ADDRCONF(NETDEV_CHANGE): docker0: link becomes ready
[   42.456742] docker0: port 2(veth42fb46f) entered blocking state
[   42.456742] docker0: port 2(veth42fb46f) entered disabled state
[   42.456766] device veth42fb46f entered promiscuous mode
[   42.456793] docker0: port 2(veth42fb46f) entered blocking state
[   42.456793] docker0: port 2(veth42fb46f) entered forwarding state
[   42.597390] eth0: renamed from veth9a6f534
[   42.604462] IPv6: ADDRCONF(NETDEV_CHANGE): veth42fb46f: link becomes ready

olysonek, 2020-10-21:
So apparently the driver has found the chip:
```
 [ 29.353934] nct6683: Found NCT6683D or compatible chip at 0x2e:0xa20
```

But no data is exported from it. No idea why that is. Maybe try reaching out to the kernel-side maintainers.

mikeyo, 2020-10-21:
> 
> 
> So apparently the driver has found the chip:
> 
> ```
>  [ 29.353934] nct6683: Found NCT6683D or compatible chip at 0x2e:0xa20
> ```
> 
> But no data is exported from it. No idea why that is. Maybe try reaching out to the kernel-side maintainers.

OK will do. Which maintainer is best to report this to? Please provide a link.

cootem, 2021-02-23:
I added the Asus Z490-I to the Kernal bug comments. [https://bugzilla.kernel.org/show_bug.cgi?id=204807](https://bugzilla.kernel.org/show_bug.cgi?id=204807). It was opened a while ago, but it is now assigned and a P1.

alexanderek, 2021-03-01:
Same issue with ASRock x570 Phantom Gaming-ITX :(

michal-gh, 2021-03-13:
Same issue with Asrock TRX40 Creator.

oscarkey, 2021-06-15:
Hi @mikeyo, did you ever reach out to the kernel-side maintainer? I am also hitting what I think is this issue on an asrock TRX40 creator (which is a real pain because one of the fans on the motherboard is really loud!)

qRoC, 2021-08-02:
> Same issue with ASRock x570 Phantom Gaming-ITX :(

> Same issue with Asrock TRX40 Creator.

This motherboards for Ryzen.
Please read https://www.kernel.org/doc/Documentation/hwmon/nct6683.rst

> The driver has only been tested with the Intel firmware, and by default
only instantiates on Intel boards. To enable it on non-Intel boards,
set the 'force' module parameter to 1.

So: 

```sh
echo "options nct6683 force=1" >> /etc/modprobe.d/sensors.conf
reboot
sensors
```

oscarkey, 2021-08-02:
After adding the option above, unloading and reloading nct6683 (or rebooting), I can now see various additional temperatures, voltages, and fan speeds by running `sensors`!

michal-gh, 2021-08-02:
Confirmed, I also see this data on my ASRock TRX40 Creator

vincele, 2023-04-10:
Asrock boards may not need the `force=1` parameter since:
https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/drivers/hwmon/nct6683.c?id=bd433537fef88d76e7f427bafda18791ae60e721

There is another Asrock ID added a bit later:
https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/drivers/hwmon/nct6683.c?id=d55532f771372c60e01a67d1fa9789b71869eb27

qRoC, 2023-04-10:
> Asrock boards may not need the `force=1` parameter since: [...]

Yep, info for users:

NCT6683_CUSTOMER_ID_ASROCK - kernel v5.12+
NCT6683_CUSTOMER_ID_ASROCK2 - kernel v5.16+

1368129224, 2024-03-04:
https://github.com/branchmispredictor/asrock-nct6683

It works for me (ASRock Z370M Pro4)