lm-sensors/lm-sensors #286: Lot of disk access with linux kernel 5.8.0-25

Gilbert-de-Gaspe, 2020-10-24:
I am using Ubuntu Mate, after upgrade to 20.10, I had a constant blinking led for disk access.
When booting with the previous kernel 5.4.0-52, I was not having this behavior.
I use conky, so after removing the line ${exec sensors|grep 'Core 0'|cut -c17-23} that give me the CPU temperature, the behavior stopped.
Something changed in the kernel that affect sensors behavior.
Could be a bug in the kernel or a change that will need sensors to be modified.
Passing the information along.
Gilbert

V10lator, 2020-12-26:
This is caused by the kernel setting CONFIG_SENSORS_DRIVETEMP:
```
If you say yes you get support for the temperature sensor on
hard disk drives.

This driver can also be built as a module. If so, the module
will be called drivetemp.
```
So if your distibutor ships this as a module your fist try should be to blacklist it: Try adding the line
```
blacklist drivetemp
```
to `/etc/modprobe.d/blacklist`, then reboot.

Gilbert-de-Gaspe, 2020-12-26:
Thanks for your answer **V10lator**

I did try-it and it made no change.

In the mean time my 'work around' is:
In a script that I have to monitor my VPN (run every second) I added this

**CpuTemp=$(cat** **/sys/class/thermal/thermal_zone0/temp)**
**CpuTemp="${CpuTemp:0:2}"**
**echo "$CpuTemp C" > ~/.config/session//CPUTEMP**

Get the cpu temp ex: "38000"
cut to "38"
save in a file in a tmpfs folder (no disk access) as "38 C"

in Conky I run this line to display the cpu temp
**${exec cat ~/.config/session/CPUTEMP}**

I also look which modules are loaded for temperature:

**x86_pkg_temp_thermal 20480 0 - Live 0x0000000000000000**
**coretemp 20480 0 - Live 0x0000000000000000**