lm-sensors/lm-sensors #287: No info about fans, liquid cooler connected directly to MB (Asus Prime Z370-p)

Sporknife, 2020-10-27:
Hello,
I'm trying to get some info about my cpu (radioator fans), liquidcooler and case fan or even control them if possible  but i noticed that i don't get any info about any of my hardware. Here is info that might help.
Motherboard: Asus Prime Z370-p
Nuvoton chip: 6793D
Dmidecode:[ ubuntu-paste](https://paste.ubuntu.com/p/2zYXMJbVYd/)
sensors: [ubuntu-paste](https://paste.ubuntu.com/p/mD6Q4y9v9n/)
sensors-detect: [ubuntu-paste](https://paste.ubuntu.com/p/VR8rFqFWZY/)
lsmod | grep "nct": [ubuntu-paste](https://paste.ubuntu.com/p/FnRCMV6g6y/)
I will try to reply asap. Thanks in advance !

RonnySvedman, 2020-11-14:
I've seen this problem with nct6775 in this forum somewhere. I think it was a conflict somewhere, have a look in dmesg ?

Sporknife, 2020-11-18:
@RonnySvedman hey. forgot to close this issue. For anyone else with the same problem check [this](https://wiki.archlinux.org/index.php/lm_sensors#Asus_Z97/Z170/X570_motherboards).

Sporknife, 2020-11-18:
It works now. Check my previous comment.

closed 2020-11-18 by Sporknife