lm-sensors/lm-sensors #289: sensors conflicting with itself can cause temp report of 511C, causing shutdown

MajorGonzo, 2020-11-14:
I've been using sensors to monitor my temps as my CPU and GPU temps on my new build were too high.  While doing this I have a cron job running which runs a script that calls sensors and ps and logs their output.  On at least four occasions, the cron script captured a 511C temp for the GPU immediately before the shutdown.  This is obviously an incorrect temp, but the computer shuts down anyway.  Since the script only runs once a minute, the chance of it capturing the temp while something else caused it is very slim.  Here's the catch.  This has ONLY happened while I was also running sensors on my second monitor (so I could also see the temps while gaming).  Somehow, the two instances of sensors must clash, with one of the instances getting a bad temp.

My cron script is:
```
date >> temps.log
sensors | grep e:.*+.*C | awk '{print $1 $2}' >> temps.log
ps c -eo cmd,%cpu,%mem --sort=-%cpu | head -5 | tail -4 >> temps.log
```
on my second monitor, I just run:
`watch -n 2 sensors`

Easy enough to avoid (don't do this), but I wanted to let you know of this problem.

MajorGonzo, 2020-11-16:
Update:  today the GPU crashed with me running watch -n 2 sensors on my second monitor, and the cron job turned off.  Seems getting temps on a second monitor while gaming is the problem.

RonnySvedman, 2020-11-16:
Are you gaming using proton/wine ?  is it possible that the game itself or steam is mucking about with the sensor chip and fan speeds causing a conflict? I have had sudden deaths too but didn't think it might be sensors related

MajorGonzo, 2020-11-16:
> Are you gaming using proton/wine ? is it possible that the game itself or steam is mucking about with the sensor chip and fan speeds causing a conflict? I have had sudden deaths too but didn't think it might be sensors related

I do use Proton and Wine to run Steam games (Warframe is probably the most graphics intensive), but it has not crashed while doing so.  Every crash has occurred while playing Dauntless using Lutris to run Epic Games Store.  I keep looking for answers, and it seems other folks have had problems monitoring temps on a second monitor as well.  However, other folks have had issues with crashes that seem related to the amdgpu driver not being able to properly throttle down the video card if it gets too hot (and my card runs about 85C, but sometimes creeps up a little higher).  Still, it has never crashed when I wasn't monitoring temps on a second monitor.  I stopped doing it for a while and no crashes, then just yesterday I was doing it again, and it crashed again.  Coincidence?  Perhaps.  I just turned down the graphics settings on Dauntless to see if it cools it a bit.  In addition, I have a 4th fan arriving today, and that should help keep it a bit cooler.

MajorGonzo, 2020-11-16:
> > Are you gaming using proton/wine ? is it possible that the game itself or steam is mucking about with the sensor chip and fan speeds causing a conflict? I have had sudden deaths too but didn't think it might be sensors related
> 
> I do use Proton and Wine to run Steam games (Warframe is probably the most graphics intensive), but it has not crashed while doing so. Every crash has occurred while playing Dauntless using Lutris to run Epic Games Store. I keep looking for answers, and it seems other folks have had problems monitoring temps on a second monitor as well. However, other folks have had issues with crashes that seem related to the amdgpu driver not being able to properly throttle down the video card if it gets too hot (and my card runs about 85C, but sometimes creeps up a little higher). Still, it has never crashed when I wasn't monitoring temps on a second monitor. I stopped doing it for a while and no crashes, then just yesterday I was doing it again, and it crashed again. Coincidence? Perhaps. I just turned down the graphics settings on Dauntless to see if it cools it a bit. In addition, I have a 4th fan arriving today, and that should help keep it a bit cooler.
> 
> From the logs before the crash, the first error was this:
> `[drm:amdgpu_job_timedout [amdgpu]] *ERROR* ring gfx timeout, signaled seq=2180988, emitted seq=2180990`
> grouped together with this:
> `[drm:amdgpu_job_timedout [amdgpu]] *ERROR* Process information: process Dauntless-Win64 pid 25154 thread Dauntless-Win64 pid 25236`
> which is followed by a cascade of errors, with series of these and others related to timing out:
> 
> ```
> [drm:amdgpu_atom_execute_table_locked [amdgpu]] *ERROR* atombios stuck executing D836 (len 824, WS 0, PS 0) @ 0xD9B6
> [drm:dce110_link_encoder_disable_output [amdgpu]] *ERROR* dce110_link_encoder_disable_output: Failed to execute VBIOS command table!
> [drm:atom_op_jump [amdgpu]] *ERROR* atombios stuck in loop for more than 5secs aborting
> ```

Update:  I modified my temperature monitoring analysis script to be able to filter by process, and found that, in fact, one time I WAS running Warframe when there was a crash with a temp of 511C.  Also, the GPU temp at the previous minute was only 72C.  Kinda eliminates the problem being related to one program category or the other (Proton vs Lutris), as well as the theory it's heat related.

MajorGonzo, 2020-11-25:
My bad.  It's not sensors.  sensors was in fact reading a bad value, but it was supplied by the system.  I did a clean install upgrading to 20.10, and the problem still exists, even without lm-sensors installed.  My apologies to the lm-sensors team.

closed 2020-11-25 by MajorGonzo